SmingHub/Sming #1607: SPIFFS w/ Serial race condition? (example provided)

kmihaylov, 2019-02-08:
Hello.

Recently I had nasty bug that blocked the execution of the program and deleted all my files. I tracked down it to be a combination of SPIFFS file write and some debug output.

I disabled the ```Serial.systemDebugOutput(true);``` and the problem went away. My devices now miss this bug (although now sometimes I'm getting all files deleted on blackout/power down, but don't have clue what it can be...:( 
It is good if you guys give your opinion.

In this example what happens is that the spiffs is initially formatted, a WDT follows, then formatted again (!?) and afterwards the file writing process works smoothly (at least in this little example).

This is the output from the example

```
Commandhandlercommand status registered
378840 Commandhandlercommand echo registered
379141 Commandhandlercommand help registered
379587 Commandhandlercommand debugon registered
383763 Commandhandlercommand debugoff registered
388031 Commandhandlercommand command registered
Welcome to the Stream Command executor
395687 Commandhandlercommand debug registered
Dbg    0.399 : Debug started
Dbg    0.402 : This is the debug output
405599 trying to mount spiffs at 0x00100000, length 65536
410649 fs.start: size:64 Kb, offset:0x100000

414642 First init file system
417226 sect_first: 100, sect_last: 110

470770 0%
521201 6%
571756 12%
617640 18%
664843 25%
714120 31%
761467 37%
809254 43%
856456 50%
903833 56%
951100 62%
995443 68%
1042691 75%
1089986 81%
1139010 87%
1185201 93%
1231265 100%
1231467 formated
1233044 mount res: 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 1344, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8 
tail 12
chksum 0xbd
csum 0xbd

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
��␂�p␃�n�{��o|�␌d␌l$l`␂��r�l��: 0071393 Commandhandlercommand status registered
71808 Commandhandlercommand echo registered
72107 Commandhandlercommand help registered
72409 Commandhandlercommand debugon registered
76382 Commandhandlercommand debugoff registered
80564 Commandhandlercommand command registered
Welcome to the Stream Command executor
88133 Commandhandlercommand debug registered
Dbg    0.091 : Debug started
Dbg    0.094 : This is the debug output
97959 trying to mount spiffs at 0x00100000, length 65536
102918 fs.start: size:64 Kb, offset:0x100000

106915 First init file system
109499 sect_first: 100, sect_last: 110

163051 0%
213350 6%
262456 12%
311352 18%
360194 25%
409344 31%
455243 37%
504199 43%
551403 50%
598715 56%
644611 62%
690326 68%
737552 75%
784844 81%
832483 87%
881859 93%
930961 100%
931161 formated
932736 mount res: 0

mode : softAP(62:01:94:15:fe:14)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
1944908 Writing data to spiffs

2944974 Writing data to spiffs

3944968 Writing data to spiffs

4944966 Writing data to spiffs

5944947 Writing data to spiffs
```

Here is the example: https://github.com/kmihaylov/Sming_fsTest

Not sure if is related to  #1313.
SDK# 2.0
Commit 0e2bf2a677efeb613d909656c4eafbe0fe319a21

mikee47, 2019-02-14:
@kmihaylov Would you check if #1610 fixes your problem?

kmihaylov, 2019-02-14:
With #1610 there is no WDT reset anymore!

closed 2019-02-14 by kmihaylov

mikee47, 2019-02-14:
> With #1610 there is no WDT reset anymore!

Music to my ears (?)!