SmingHub/Sming #1609: multiple definition of `malloc'

frankdownunder, 2019-02-09:
I get linker errors when using the SDK version 3.

With the default SDK version, everything is fine. I run this script:

``` 
    cd $SMING_HOME
    git checkout develop
    git pull
    make dist-clean
    cd ~/myApp
    make clean
    make V=1 | tee tmp.txt
```

and all is good. But if I include this line:
```   
export SDK_BASE="$SMING_HOME/third-party/ESP8266_NONOS_SDK"
```
at the start, I get 
opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -L/opt/Sming/Sming/compiler/lib/ -L/opt/Sming/Sming/third-party/ESP8266_NONOS_SDK/lib -Lout/build -L/opt/Sming/Sming/compiler/ld -Trboot.rom0.ld -nostdlib -u call_user_start -u Cache_Read_Enable_New -u spiffs_get_storage_config -u custom_crash_callback -Wl,-static -Wl,--gc-sections -Wl,-Map=out/build/app_0.map -Wl,-wrap,system_restart_local  -Wl,--start-group out/build/app_app.a -lmicroc -lmicrogcc -lhal -lphy -lpp -lnet80211 -llwip_open -lmqttc -lwpa -lsming -lmain2 -lcrypto -lpwm_open -lsmartconfig -Wl,--end-group -o out/build/app_0.out
out/build/libmain2.a(mem_manager.o): In function `malloc':
(.text+0x470): multiple definition of `malloc'
/opt/Sming/Sming/compiler/lib//libmicroc.a(malloc.o):C:\DEV\ESP8266\esp_microc/malloc/malloc.c:8: first defined here
out/build/libmain2.a(mem_manager.o): In function `free':
(.text+0x48c): multiple definition of `free'
/opt/Sming/Sming/compiler/lib//libmicroc.a(malloc.o):C:\DEV\ESP8266\esp_microc/malloc/malloc.c:8: first defined here
out/build/libmain2.a(mem_manager.o): In function `zalloc':
(.text+0x4a4): multiple definition of `zalloc'
/opt/Sming/Sming/compiler/lib//libmicroc.a(malloc.o):C:\DEV\ESP8266\esp_microc/malloc/malloc.c:18: first defined here
out/build/libmain2.a(mem_manager.o): In function `calloc':
(.text+0x4bc): multiple definition of `calloc'
/opt/Sming/Sming/compiler/lib//libmicroc.a(malloc.o):C:\DEV\ESP8266\esp_microc/malloc/malloc.c:23: first defined here
out/build/libmain2.a(mem_manager.o): In function `realloc':
(.text+0x4d4): multiple definition of `realloc'
/opt/Sming/Sming/compiler/lib//libmicroc.a(malloc.o):C:\DEV\ESP8266\esp_microc/malloc/malloc.c:13: first defined here
collect2: error: ld returned 1 exit status
make: *** [/opt/Sming/Sming/Makefile-rboot.mk:547: out/build/app_0.out] Error 1

frankdownunder, 2019-02-19:
Further information:  The code includes a custom crash handler with signature
extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end)
The linker errors go away if that code is removed.

slaff, 2019-02-20:
@frankdownunder It is highly recommended to NOT use any mallocs in your crash handler. Look at that directory https://github.com/attachix/ukit-esp8266-firmware/tree/master/lib/core and check the crash.h and crash.cpp files for ideas and implementation.

frankdownunder, 2019-02-21:
The code I was using did not call malloc, but it did call fileOpen and  fileWrite to write the crash file to spiffs, which must in turn do bad things.  This issue can be closed - except that I would still love some advice on how to find what was causing the problem - I had to resort to the slow process of removing code bit by bit until the problem went away.

closed 2019-02-26 by slaff