SmingHub/Sming #1615: axl_ssl_read hangs on memory allocation failure

mikee47, 2019-02-10:
See `third-party\axtls-8266\compat\lwipr_compat.c` line 161. If a memory allocation fails system just hangs and we get WDT.

I would suggest better behaviour would be to free any allocated memory and return error (-1) so at least the system keeps running.

@slaff I have a proposed patch for this, but this file belongs to a sub-module of axtls so not sure of the best way to approach this.

slaff, 2019-02-10:
@mikee47 Can make a pull request to this repo: https://github.com/attachix/lwirax and I will take care for the rest.

slaff, 2019-02-11:
The PR request (https://github.com/igrr/axtls-8266/pull/62) to axtls-8266 is created and waiting to be merged.

PR #1651 (2019-03-24): Added work-around patch for returning error instead of hanging.
Fixes #1615.
closed 2019-03-24 by slaff