SmingHub/Sming #1617: System remember SSID and PWD

Evgsurf, 2019-02-12:
This is not real issue. I just do not understand how it does work.
I build and flash some project which connect to local WiFi. -Ok , passed.
Later I build and flash Basic_Blink project which has not any! WiFi connections, but as I can see in console, board successfully connects to WiFi and get IP. 
Of couse, flashinit kills WiFi, but the question is, where board save WiFi and how to manage it?

mikee47, 2019-02-12:
These are stored in a reserved area of flash memory by the Espressif SDK.

Evgsurf, 2019-02-12:
I see, thank you. Any method to erase it (exclude flashinit) ?

mikee47, 2019-02-12:
`Any method to erase it (exclude flashinit) ?`

Try `WifiStation.config("", "", false, true)`

closed 2019-02-12 by Evgsurf