SmingHub/Sming #1620: Coding style changes: deprecated and override

mikee47, 2019-02-14:
I'm working on a PR to enforce use of `__ attribute__((deprecated))` throughout the framework; I'm proposing to use a macro `__deprecated` for convenience. The purpose is to ensure the framework builds without internally referencing any deprecated items. Travis enforces this. It will also assist users in migrating their code.

The second item is to apply `override` to every overridden virtual function. The purpose is to catch typing errors; using virtual with a mis-spelled method name or parameter type will simply create a new method.

One possible point of contention is whether to use `virtual` or `override` on inherited destructors. I'm thinking not, as it doesn't add anything and can actually confuse. virtual destructors do not behave like regular virtual methods, and typos are already flagged by the compiler. Thoughts?

slaff, 2019-02-14:
> One possible point of contention is whether to use `virtual` or `override` on inherited destructors. I'm thinking not, as it doesn't add anything and can actually confuse.

Correct.

mikee47, 2019-02-14:
What's the thinking on empty overridden destructors? Remove or keep?

slaff, 2019-02-15:
> What's the thinking on empty overridden destructors? Remove or keep?

Remove.

mikee47, 2019-02-15:
Is there any justification for using `#include <SmingCore/SmingCore.h>` where `<SmingCore>` is sufficient?

slaff, 2019-02-16:
> Is there any justification for using `#include <SmingCore/SmingCore.h>` in applications when `#include <SmingCore.h>` is sufficient?

I cannot think of any ... So yes, please, use SmingCore.h  instead of SmingCore/SmingCore.h .

PR #1629 (2019-02-19): Enforce code deprecation and virtual methods
* Use `SMING_DEPRECATED` attribute to ensure marked code not referenced within framework
* Apply `override` to all inherited virtual methods
* Ensure base virtual classes have virtual destructors
* Remove `virtual` from overriden destructors
* Remove empty overridden destructors and overriden methods that do not change base behaviour
* Ensure consistent ordering of constructors/destructor in header

Some samples use deprecated methods. If no suitable alternative has yet been provided, those methods have been un-deprecated and noted in the method com [... cut]
closed 2019-02-19 by slaff