SmingHub/Sming #1626: git submodules errors during first make after git clone

andr2000, 2019-02-16:
After I have cloned the repo at
`* develop 3a7fbb2d0b87 Fix `fromHttpDate()`, parameters from signed to unsigned and String optimisation. (#1625)`
I set up the environment and tried to make, but I see there are errors during git submodules fetch
[git_log.txt](https://github.com/SmingHub/Sming/files/2872063/git_log.txt)

mikee47, 2019-02-16:
You'll need to `make dist-clean` first.
Don't worry about the `Error 128 (ignored)` - that is because their is no patch file for those.

andr2000, 2019-02-16:
Well, this doesn't really help:
```
$:/tmp/Sming/Sming$ make dist-clean 
make[1]: Entering directory '/tmp/Sming/Sming/third-party/esp-open-lwip'
make[1]: Makefile.open: No such file or directory
make[1]: *** No rule to make target 'Makefile.open'.  Stop.
make[1]: Leaving directory '/tmp/Sming/Sming/third-party/esp-open-lwip'
Makefile:483: recipe for target 'clean' failed
make: [clean] Error 2 (ignored)
make[1]: Entering directory '/tmp/Sming/samples/Accelerometer_MMA7455'
```
and then
```
$:/tmp/Sming/Sming$ make
Fetching Arduino Library Libraries/Adafruit_BME280_Library/ ...
Submodule 'Sming/Libraries/Adafruit_BME280_Library' (https://github.com/adafruit/Adafruit_BME280_Library.git) registered for path 'Libraries/Adafruit_BME280_Library'
Cloning into '/tmp/Sming/Sming/Libraries/Adafruit_BME280_Library'...
Submodule path 'Libraries/Adafruit_BME280_Library': checked out '321186220e1e04080bb70fa45aae6e6a5820899f'
error: unrecognized input
Makefile:533: recipe for target 'Libraries/Adafruit_BME280_Library/library.properties' failed
make: [Libraries/Adafruit_BME280_Library/library.properties] Error 128 (ignored)
```

andr2000, 2019-02-16:
The reason I am worried about those "ignored" errors is that I build Sming with Yocto [1], [2] and those make the build fail. This doesn't happen with master branch though... and because of rboot fixes I decided to temporarily switch to develop branch.
[1] https://www.yoctoproject.org/
[2] https://github.com/andr2000/meta-prod-smarthome/blob/master/recipes-esp8266/sming/sming.bb

mikee47, 2019-02-16:
Have you done `env SMING_HOME=/tmp/Sming/Sming' ?

andr2000, 2019-02-16:
`declare -x SMING_HOME="/tmp/Sming/Sming"
`

mikee47, 2019-02-16:
`make V=1` give any clues as to what's happening?

andr2000, 2019-02-16:
$:/tmp/Sming/Sming$ make V=1
git submodule update --init --recursive Libraries/Adafruit_BME280_Library/
Submodule 'Sming/Libraries/Adafruit_BME280_Library' (https://github.com/adafruit/Adafruit_BME280_Library.git) registered for path 'Libraries/Adafruit_BME280_Library'
Cloning into '/tmp/Sming/Sming/Libraries/Adafruit_BME280_Library'...
Submodule path 'Libraries/Adafruit_BME280_Library': checked out '321186220e1e04080bb70fa45aae6e6a5820899f'
touch Libraries/.patches/Adafruit_BME280_Library.patch
cd Libraries/Adafruit_BME280_Library/; git apply -v ```
/tmp/Sming/Sming/Libraries/.patches/Adafruit_BME280_Library.patch --ignore-whitespace --whitespace=nowarn
error: unrecognized input
Makefile:533: recipe for target 'Libraries/Adafruit_BME280_Library/library.properties' failed
make: [Libraries/Adafruit_BME280_Library/library.properties] Error 128 (ignored)
```
```
ls -la Libraries/.patches/Adafruit_BME280_Library.patch
-rw-r--r-- 1 a2k a2k 0 Feb 16 17:44 Libraries/.patches/Adafruit_BME280_Library.patch
```
So, the patch is empty and this is why git rejects to apply

andr2000, 2019-02-16:
it seems that
`touch Libraries/.patches/Adafruit_BME280_Library.patch`
created that

mikee47, 2019-02-16:
I get those as standard when I build, but git just ignores it, doesn't crash. Guess there's a setting in git for that?

andr2000, 2019-02-16:
```
--- a/Sming/Makefile
+++ b/Sming/Makefile
@@ -523,7 +523,7 @@ third-party/%:
        $(vecho) "Fetching $(dir $@) ..."
        $(Q) $(GIT) submodule update --init --recursive $(dir $@)
        $(Q) touch $(patsubst third-party/%/,third-party/.patches/%.patch, $(dir $@))
-       $(Q) -cd $(dir $@); $(GIT) apply -v $(patsubst third-party/%/,$(SMING_HOME)/third-party/.patches/%.patch, $(dir $@)) --ignore-whitespace --whitespace=nowarn
+       $(Q) -cd $(dir $@); $(GIT) apply -v $(patsubst third-party/%/,$(SMING_HOME)/third-party/.patches/%.patch, $(dir $@)) --ignore-whitespace --whitespace=nowarn || true
 # if the new submodule brings source code files that need to be compiled inside Sming
 # then we need somehow to be able to "see" these new files.
 # For now we solve this by "reloading" the makefile after fetching a module.
@@ -533,7 +533,7 @@ Libraries/%:
        $(vecho) "Fetching Arduino Library $(dir $@) ..."
        $(Q) $(GIT) submodule update --init --recursive $(dir $@)
        $(Q) touch $(patsubst Libraries/%/,Libraries/.patches/%.patch, $(dir $@))
-       $(Q) -cd $(dir $@); $(GIT) apply -v $(patsubst Libraries/%/,$(SMING_HOME)/Libraries/.patches/%.patch, $(dir $@)) --ignore-whitespace --whitespace=nowarn
+       $(Q) -cd $(dir $@); $(GIT) apply -v $(patsubst Libraries/%/,$(SMING_HOME)/Libraries/.patches/%.patch, $(dir $@)) --ignore-whitespace --whitespace=nowarn || true
```
Normally we can't ignore those errors, but it seems to be ok for "just build and that's it" :)

mikee47, 2019-02-16:
I guess the makefile could be improved by checking for the existence of the patch file first :-)

andr2000, 2019-02-16:
Yeah, at least. That was just a quick and dirty hack ;)

andr2000, 2019-02-16:
BTW, do you by chance know what was the reason to touch the patch at all?

andr2000, 2019-02-16:
@mikee47 @slaff could you please review?

closed 2019-02-18 by andr2000