SmingHub/Sming #1635: debugf not printing

icodk, 2019-02-20:
Can't get debugf to print anything. Compile OK. Serial.printf works fine
Tried with both  DEBUG_VERBOSE_LEVEL = 0 and then DEBUG_VERBOSE_LEVEL = 3
in Makefile-user.mk
Serial.printf prints the DEBUG_VERBOSE_LEVEL 
using SDK: v1.5.4(baaeaebb) on linux
SMING_VERSION "3.7.1"

slaff, 2019-02-26:
@icodk If you mean the debug statements from the library then you should compile the library itself with high verbosity. Here is what you can do:

```bash
# Go to Sming home directory
cd $SMING_HOME  

# make sure to use the develop branch
git checkout develop

# get the latest code
git pull

# make sure that all third party libraries are clean and no old-libraries will be used
make dist-clean

# Now go to a sample. In our case this is the `HttpServer_ConfigNetwork` sample 
cd ../samples/HttpServer_ConfigNetwork/

# If you want to see all debug messages you MUST set this variable before the compilation of the library
export DEBUG_VERBOSE_LEVEL=3

# Now compile the sample. It will take care also to compile the library
make

# Then make sure to flash your application
make flash

...
```

I have double-checked with the latest `develop` version (as of minutes ago) and debugf seems to work as expected.

icodk, 2019-02-26:
@slaff Thanks

closed 2019-02-28 by slaff