SmingHub/Sming #1637: Web server RE initialization

icodk, 2019-02-20:
Is (how) it possible to restructure the paths a web server is responding to later in the life time of the application ?
in old version of sming (about 2.0) I just added a path.clear() function that cleared the path map just before I populated it with a subset of the original pages (saving space and access to some functionality not needed in the long run). 
I could do it again of course but it is not version portable. Could be nice to have it build in.
The simplest way would probably be to expose the resource tree as public.

mikee47, 2019-02-21:
> The simplest way would probably be to expose the resource tree as public.

That's a good idea. At present, it's unsafe because `ResourceTree` inherits from `HashMap` which doesn't manage objects very well - it has the potential to leak like a sieve and objects have to be manually destroyed. I'm currently working up a PR to rectify this situation, which I think will make your suggestion practical and simplify things considerably.

icodk, 2019-02-21:
Looking forward...
Thanks

mikee47, 2019-02-28:
@icodk You can now use `HttpServer::paths.clear()`

slaff, 2019-02-28:
> Is (how) it possible to restructure the paths a web server is responding to later in the life time of the application ?

@icodk That should be possible with the latest version of the code from the `develop` branch. Give it a try and write back if you need some help.

icodk, 2019-02-28:
@mikee47 & @slaff 
That's great Thanks

closed 2019-02-28 by icodk