SmingHub/Sming #1640: Reason for irom offset

r44083, 2019-02-25:
Hi, I have noticed that linker script "compiler/ld/rboot.rom0.ld" defines irom section with some offset (0x2010) from irom start:
```
  irom0_0_seg :                         org = 0x40202010, len = (1M - 0x2010)
```
**Why is 0x2010, what is located at start address of irom?
For what reason you shift irom start address?**

slaff, 2019-02-25:
Because we use bootloader. The 0x2010 offset is the space needed for the bootloader and the bootloader config at the beginning of the FLASH. After that offset starts our application.

r44083, 2019-02-25:
Thnk you slaff.

closed 2019-02-25 by r44083

r44083, 2019-02-25:
Could you please answer, for what reason you use bootloader? Is it for FOTA?

slaff, 2019-02-25:
> Is it for FOTA?

Yep.