SmingHub/Sming #1643: promiscuous mode in Sming

icodk, 2019-03-06:
Promiscuous mode works fine in Arduino/esp8266 with both  sdk 1.5.3 [https://www.esp8266.com/viewtopic.php?f=29&t=12127 ](https://www.esp8266.com/viewtopic.php?f=29&t=12127)  and 3.0.0 but I could not make it work with Sming (tried only with 1.5.3). Everything compiles and uploads but the call back function that should be called upon reception of wifi activity is never called. Could it be an issue with  C/C++ calling calling convention ?

mikee47, 2019-03-13:
I've just pushed #1647 which is a first attempt to provide a WifiSniffer class based on this code. The sniffer initialisation (`WifiSniffer::begin`) cannot be called in `init()`, has to be done after network stack's been initialised (i.e. onReady).

slaff, 2019-03-15:
@icodk Take a look at the new Wifi_Sniffer sample and give it a try.

PR #1647 (2019-03-15): Add `WifiSniffer` class and sample application
Uses promiscuous mode to provide generic mechanism for Wifi sniffing.

Closes #1643.
closed 2019-03-15 by slaff