SmingHub/Sming #1654: paradox of Sming

kwis2, 2019-03-28:
In my opinion there is paradox in Sming; 
Sming is very good framework for intermediate user. In meantime user should be rally  advanced to be able to run Sming especially on Windows machines. 
Note Windows user often mean not enought advanced to migrate to linux....

I have returned to Sming after a few month and found it is much more advanced than before, in meantime there are still problems with make it to run. 
Maybe solution it would be to ask Sming to work with any environment like for instance PlatformIO?

Vinimuller, 2019-03-28:
I dont know what to say about PlatformIO, but I can't agree more with you about your paradox ahahahh

kwis2, 2019-03-28:
@Vinimuller maybe some examples:
I have tried to show Sming to students of electornic engineering (first years). After a few hours they were able to run quite advanced projects, but they failed with instaling Sming on their computers. Simple reason - how could they follow Windows-Quickstart with `git config --global core.autocrlf input`, while default there is no git on windows machines? 
More interesting were linux users - follow Linux-Quickstart `export SMING_HOME=/opt/Sming/Sming`  then `cd $SMING_HOME/../..`  and what happened when there was no Sming dir??
So for me it is paradox - first year students are able to use Sming, but they do not use it because of ... (I really don't know how to name it).
Advantage of environments - PlatformIO it is only as example - is it's way of how to work with it. 
For instance ota - all user should do, it is to set board and it works. Sumarizing immediate knowledge to make run, immediate to  write code.
With Sming rBoot (Basic_rBoot example) there is readme which say:

> - Copy ....
> - Adjust ...
> - Uncomment ...
> - Ensure ....
> - If ...
> - If ...
> - After ....  

there is even no information how to setup different board to make example work (try run ESP8266-01 with this example). But if by any way user make it to work, he could simply write code. Summarizing extremly advance knowledge to run, immediate to write code.
And this is paradox in my oppinion which rejects most of beginner users, so they will not return when will be more advanced

closed 2019-05-24 by kwis2

slaff, 2019-06-04:
@kwis2 If you haven't heard yet we have a Host Emulator that allows application code to be compiled and run natively before flashing it on a microcontroller. Here is a short interactive tutorial: https://www.katacoda.com/slaff/scenarios/sming-host-emulator. The tutorial should be easy to follow and understand even from non-core Sming developers.

icodk, 2019-06-05:
Wao ! katacoda is great and host emulater even more.
Regarding  host emulator, I have the follwoing questions:
1. there are no output function calls (debugf etc.) still there is output for digitalWrite and pinMode. Where is it comes from and how can it be controled?
2. I inserted a call to debugf but nothing came out. 
3.  Where can we see documentation of using host emulator in our own environment ?
 The only visible change was running the setup-network-linux.sh and definition of the SMING_ARCH
Thanks for a great framework  Sming that you  keep evolving  and improving.

mikee47, 2019-06-05:
@slaff That is awesome!

slaff, 2019-06-05:
> there are no output function calls (debugf etc.) still there is output for digitalWrite and pinMode. Where is it comes from and how can it be controled?
I inserted a call to debugf but nothing came out.

The tutorial has four steps and you have to read them carefully :). In step 3 you will find "In separate command windows, start two telnet sessions (a terminal for each serial port):

To view the output from UART0 open a new terminal and type: telnet localhost 10000 To view the output from UART1 open a new terminal and type:telnet localhost 10001". 

There is your serial output. 

> Where can we see documentation of using host emulator in our own environment ?

https://github.com/SmingHub/Sming/blob/develop/Sming/Arch/Host/readme.md and 
https://github.com/SmingHub/Sming/wiki/Host-Emulator.  I guess I should add them to the interactive tutorial. 

> Thanks for a great framework Sming that you keep evolving and improving.

Thanks to all the contributors and supporters we evolve further. Especially to our new super-coding-hero @mikee47 :)

slaff, 2019-06-05:
> Where can we see documentation of using host emulator in our own environment ?

The katacoda tutorial itself is a "fancy" documentation and live demonstration in one.

slaff, 2019-06-05:
By the way the current version of the tutorial is in github: https://github.com/slaff/katacoda-scenarios/tree/master/sming-host-emulator and everyone can improve by sending a PR.

icodk, 2019-06-05:
Thanks again for help and patient

frankdownunder, 2019-06-11:
@slaff  I love this host emulator, your efforts are much appreciated.