SmingHub/Sming #1656: ESP8266-01 module problem with Sming 3.7.1

kwis2, 2019-03-30:
I have problems to run programs on ESP8266-01 modules. None of my programs, which worked on Sming 3.4 not run with version 3.7.1. Of course I have modified it and succesfully compile, but module do not start to run. The only response I got it was after manual reset

> ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> 
> ets_main.c 

manual reset after some time 

>  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> 
> ets_main.c 

I have compiled Basic_Blink example and got infinite loop of wdt resets. 
My module ESP8266-01 with 1M flash, Windows 10 with sming installed by choco. I have tried different settings - below is 512k and GPIO skip. None of flashinit, Sming make didn't help. 
I think it is rBoot ESP8266-01 setting problem.

>  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
> 
> wdt reset
> load 0x40100000, len 1540, room 16 
> tail 4
> chksum 0x18
> load 0x3ffe8000, len 748, room 4 
> tail 8
> chksum 0x0c
> csum 0x0c
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   4 Mbit
> Flash Mode:   DIO
> Flash Speed:  40 MHz
> rBoot Option: GPIO skip mode (16)
> 
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> 
> ================================================================
> 3ffffa60:  3d3d3d0a 3d3d3d3d 3d3d3d3d 3d3d3d3d  
> 3ffffa70:  3d3d3d3d 3d3d3d3d 3d3d3d3d 3d3d3d3d  
> ..........
> 3fffff90:  ffff0000 00006d70 006d7066 00007070  
> 3fffffa0:  000202e9 00766564 ffffff00 55aa55aa  
> 
> ================================================================
>

verybadsoldier, 2019-04-01:
Which SDK are you using? For me only SDK 3.0 worked in the end.

Maybe this can help:
https://github.com/SmingHub/Sming/issues/1570

kwis2, 2019-04-01:
With SDK 2.0.0 with Basic_Blink I got:
when I set 1M flash as my ESP8266-01 module

> chksum 0x38
> load 0x3ffe8000, len 632, room 8 
> tail 0
> chksum 0x5f
> csum 0x5f
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> ota1 not set
> ota2 not set
> p 101 and p 4 covered: addr:0x2000 + len:0xfe000 > 0xfb000
> system_partition_table_regist: failed
> size_map = 2
> partition[0]: 1, 0x00000000, 0x00001000
> partition[1]: 100, 0x00001000, 0x00001000
> partition[2]: 101, 0x00002000, 0x000fe000
> partition[3]: 4, 0x000fb000, 0x00001000
> partition[4]: 5, 0x000fc000, 0x00001000
> partition[5]: 6, 0x000fd000, 0x00003000
> 
so I decided to set 512k

> chksum 0x38
> load 0x3ffe8000, len 632, room 8 
> tail 0
> chksum 0x5f
> csum 0x5f
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   4 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> emapg
> map 0 err
> system_partition_table_regist: failed
> size_map = 0
> partition[0]: 1, 0x00000000, 0x00001000
> partition[1]: 100, 0x00001000, 0x00001000
> partition[2]: 101, 0x00002000, 0x000fe000
> partition[3]: 4, 0x0007b000, 0x00001000
> partition[4]: 5, 0x0007c000, 0x00001000
> partition[5]: 6, 0x0007d000, 0x00003000
>

kwis2, 2019-04-01:
There is no difference between SDK 2.0.0 and one included with Sming. I have checked 3.7.1 and develop branches. There were no difference.
For now I am not able to compile SDK 3.0.0.
In my opinion it is another settings problem

> tail 0
> chksum 0x38
> load 0x3ffe8000, len 632, room 8 
> tail 0
> chksum 0x5f
> csum 0x5f
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> ota1 not set
> ota2 not set
> p 101 and p 4 covered: addr:0x2000 + len:0xfe000 > 0xfb000
> system_partition_table_regist: failed
> size_map = 2
> partition[0]: 1, 0x00000000, 0x00001000
> partition[1]: 100, 0x00001000, 0x00001000
> partition[2]: 101, 0x00002000, 0x000fe000
> partition[3]: 4, 0x000fb000, 0x00001000
> partition[4]: 5, 0x000fc000, 0x00001000
> partition[5]: 6, 0x000fd000, 0x00003000
>

mikee47, 2019-04-01:
@kwis2 Looks like overlapping sectors in the partition table! Needs a fix - for now, try editing `$(SMING_HOME)/appinit/user_main.cpp` line 108 and change `MAX_PROGRAM_SECTORS - 2` to `MAX_PROGRAM_SECTORS - 8`, then `make` for Sming and app.

kwis2, 2019-04-02:
I am trying to install esp-open-sdk with #1570 method.
after `make VENDOR_SDK=3.0.0 STANDALONE=y` I got 

> cp esptool/esptool.py /opt/esp-open-sdk/xtensa-lx106-elf/bin/
> Creating irom version of libc...
> /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-objcopy --rename-section .text=.irom0.text \
>         --rename-section .literal=.irom0.literal /opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/lib/libc.a /opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/lib/libcirom.a;
> unzip -q -o 
> UnZip 6.00 of 20 April 2009, by Debian. Original by Info-ZIP.
> 
> Usage: unzip [-Z] [-opts[modifiers]] file[.zip] [list] [-x xlist] [-d exdir]
>   Default action is to extract files in list, except those in xlist, to exdir;
>   file[.zip] may be a wildcard.  -Z => ZipInfo mode ("unzip -Z" for usage).
> 
>   -p  extract files to pipe, no messages     -l  list files (short format)
>   -f  freshen existing files, create none    -t  test compressed archive data
>   -u  update files, create if necessary      -z  display archive comment only
>   -v  list verbosely/show version info       -T  timestamp archive to latest
>   -x  exclude files that follow (in xlist)   -d  extract files into exdir
> modifiers:
>   -n  never overwrite existing files         -q  quiet mode (-qq => quieter)
>   -o  overwrite files WITHOUT prompting      -a  auto-convert any text files
>   -j  junk paths (do not make directories)   -aa treat ALL files as text
>   -U  use escapes for all non-ASCII Unicode  -UU ignore any Unicode fields
>   -C  match filenames case-insensitively     -L  make (some) names lowercase
>   -X  restore UID/GID info                   -V  retain VMS version numbers
>   -K  keep setuid/setgid/tacky permissions   -M  pipe through "more" pager
> See "unzip -hh" or unzip.txt for more help.  Examples:
>   unzip data1 -x joe   => extract all files except joe from zipfile data1.zip
>   unzip -p foo | more  => send contents of foo.zip via pipe into program more
>   unzip -fo foo ReadMe => quietly replace existing ReadMe if archive file newer
> Makefile:181: polecenia dla obiektu '/.dir' nie powiodły się
> make: *** [/.dir] Error 10

kwis2, 2019-04-02:
The only way I found to make Sming to work it is to compile `esp-open-sdk` with `make VENDOR_SDK=2.0.0 STANDALONE=y` then `Sming` `develop` compile with `make ENABLE_CMD_EXECUTOR=0 VENDOR_SDK=3.0.0` what in fact mean `SDK 2.0.0`.
When I add `export SDK_BASE=/opt/Sming/Sming/third-party/ESP8266_NONOS_SDK/` I got:

> /opt/Sming/samples/Basic_Blink$ make all
> make -C /opt/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0 RBOOT_SILENT=0
> make[1]: Wejście do katalogu '/opt/Sming/Sming/third-party/rboot'
> CC rboot-stage2a.c
> LD /opt/Sming/samples/Basic_Blink/out/build/rboot-stage2a.elf
> E2 /opt/Sming/samples/Basic_Blink/out/build/rboot-hex2a.h
> CC rboot.c
> LD /opt/Sming/samples/Basic_Blink/out/build/rboot.elf
> E2 /opt/Sming/samples/Basic_Blink/out/firmware/rboot.bin
> make[1]: Opuszczenie katalogu '/opt/Sming/Sming/third-party/rboot'
> CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
> CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c
> /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c: In function 'rboot_set_config':
> /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c:57:2: error: implicit declaration of function 'pvPortMalloc' [-Werror=implicit-function-declaration]
>   buffer = (uint8*)os_malloc(SECTOR_SIZE);
>   ^
> /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c:73:2: error: implicit declaration of function 'vPortFree' [-Werror=implicit-function-declaration]
>   os_free(buffer);
>   ^
> cc1: all warnings being treated as errors
> /opt/Sming/Sming/Makefile-rboot.mk:668: polecenia dla obiektu 'out/build//opt/Sming/Sming/third-party/rboot/appcode/rboot-api.o' nie powiodły się
> make: *** [out/build//opt/Sming/Sming/third-party/rboot/appcode/rboot-api.o] Error 1

kwis2, 2019-04-02:
Another issue is that I have ESP-01 module with 1M flash  (p25q80 chip) and only way to run code is to set flash to 512. When I set 1M I got as below. What in my opinion mean programming OK, but something wrong with memory setup for this module.

>  Running Cesanta flasher stub...
> Flash params set to 0x0020
> Writing 4096 @ 0x0... 4096 (100 %)
> Wrote 4096 bytes at 0x0 in 0.4 seconds (77.9 kbit/s)...
> Writing 221184 @ 0x2000... 221184 (100 %)
> Wrote 221184 bytes at 0x2000 in 19.2 seconds (92.0 kbit/s)...
> Leaving...
> python -m serial.tools.miniterm /dev/ttyUSB0 115200 
> --- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> �C�tail 0
> chksum 0x38
> load 0x3ffe8000, len 632, room 8 
> tail 0
> chksum 0x5f
> csum 0x5f
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> ����p␃�go�l␛$l␄␌␃$�b�␃␄␌��`␃��␂�␛�o�;��g|�␌�$�xsd��
and much more garbage

kwis2, 2019-04-03:
I have tried Basic_Webskeleton example with ESP8266-01 module (1M flash not working - set 512k)
program compiled succesfully but problem with programming. I don't exactly know what could it mean `WARNING: Unlikely to work as data goes beyond end of flash. Hint: Use --flash_size` but in my opinion it is some problems with memory settings ?

> /opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin
> esptool.py v1.2
> Connecting...
> Running Cesanta flasher stub...
> Flash params set to 0x0000
> Writing 4096 @ 0x0... 4096 (100 %)
> Wrote 4096 bytes at 0x0 in 0.4 seconds (77.7 kbit/s)...
> Writing 327680 @ 0x2000... 327680 (100 %)
> Wrote 327680 bytes at 0x2000 in 28.5 seconds (92.1 kbit/s)...
> WARNING: Unlikely to work as data goes beyond end of flash. Hint: Use --flash_size
> Writing 196608 @ 0x100000... 196608 (100 %)
> Wrote 196608 bytes at 0x100000 in 17.1 seconds (91.9 kbit/s)...
> 
and finally got infinite loop of

> python -m serial.tools.miniterm /dev/ttyUSB0 115200 
> --- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ��0000000, excvaddr=0x000000b5, depc=0x00000000
> Fatal exception (28): 
> epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
> Fatal exception (28): 
> epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
> Fatal exception (28): 
> epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
> Fatal exception (28): 
> ......

slaff, 2019-04-04:
> Wrote 327680 bytes at 0x2000 in 28.5 seconds (92.1 kbit/s)...
WARNING: Unlikely to work as data goes beyond end of flash. 

1MB is not enough to hold the bootloader, its config, the actual ROM, the system config and the minimal file system.  

In our code we are quite `optimistic` and assume that 512K for Spiffs should fit. 
```
else ifeq ($(SPI_SIZE), 1M)
        flashimageoptions += -fs 8m
        SPIFF_SIZE ?= 524288  #512K

``` 

What you can do is choose a smaller size for spiffs. Try recompiling the application using the following:

```
cd $SMING_HOME/../samples/HttpServer_WebSockets
export SPIFF_SIZE=131072
make clean
make 

; The command below should be called once and will set the system config data
make flashinit 

make flash
```

Give it a try and paste the output.

> and finally got infinite loop of

The infinite loop is happening IMHO because  the spiffs are not fully written and in addition overlap the system config.

slaff, 2021-03-04:
Should be fixed in the latest `develop` with the introduction of partition tables #2171.

closed 2021-03-04 by slaff