SmingHub/Sming #1658: Incompatibility between String class and Mosquitto broker

MayaPosch, 2019-04-06:
After a freshly built & flashed ESP8266 node failed to connect to a Mosquitto MQTT broker with nothing but a brief `'socket error on client <unknown>'`, I had a look at why this (previously working) code failed to work, with the MQTTClient_Hello sample working.

In the end I traced it down to a null-byte in the client ID that gets sent to the Mosquitto broker. When a null byte is added to the client ID string, Mosquitto's connect routine pushes it through its `mosquitto_validate_utf8()` routine in `utf8_mosq.c`, which returns an error when it finds a null byte, which consequently aborts the connection attempt.

As to why the sample code worked, that is because it provided a const string to the String constructor of the first parameter, which invokes the `String::copy()` routine, which sets the length of the String instance to be the number of characters, minus the null byte that gets added as well.

However, since I used the MAC address of the ESP8266 as the client ID (a practice I have used with Sming since early 2017), it gets a String instance from `StationClass::getMAC()`, which in turn invokes `makeHexString()` in Data/HexString.cpp.

It is this function which sets the String length to:

`if(!result.setLength(charsPerByte * length + 1))`

This means the characters plus the null byte as the length of the String. Consequently, when the MQTT client then copies this client ID from the String instance, it copies it without null byte in the case of the const character string, and with a null byte in the case of the MAC address String.

I hacked around this issue for now by having `makeHexString()` set the String length to just the characters (removing the + 1), but this is obviously far from ideal.

Ideally I guess String would be like std::string, with an internally managed buffer and an API that prevents such direct manipulations. Right now I'm not sure what the right course of action here is, beyond maybe slotting in that quick hack for now.

mikee47, 2019-04-06:
@MayaPosch You have it exactly right, `makeHexString()` is just plain wrong. It's miscalculating the string length and doesn't need to set the nul terminator as `setLength()` does that. I'll have a PR shortly. Thank you for taking the time to track this bug down.

slaff, 2019-04-06:
@MayaPosch @mikee47 Thank you both!

PR #1659 (2019-04-06): makeHexString() incorrectly appends nul character to result
Fixes #1658
commit 5300c0bc (2019-04-06): makeHexString() incorrectly appends nul character to result (#1659)

Fixes #1658
closed 2019-04-06 by slaff