SmingHub/Sming #1660: SPIFFS crash on api_spiffs_erase (example provided)

kmihaylov, 2019-04-07:
Hello. Finally I found when my FS is crashing. It it some time after data has been written to it, that's why it was a little bit hard to figure it out when exactly the bug appears.

I found that while there is no need for api_spiffs_erase, the consistency of the FS is OK, the device can be rebooted w/o problems (data loss, broken FS).

On the first appearance of "api_spiffs_erase", however, the FS is immediately broken.

I made a simple program where one can write some data to the FS.
Issue command ```save 10``` where the integer is how many times the fileSetContent will be called. The sample text is in the cpp file and can be changes. I set it to 10 bytes for convenience (idk how SPIFFS reserves bytes for the filename).

You have to try with some random repetition values, 100 is good to begin with. When you figure out where "api_spiffs_erase" is shown you can decrease the argument of ```save```to this position -1.

I got "api_spiffs_erase" on the 63rd write (630 bytes of data excluding what bytes were reserved for file name, spiffs, etc).

If I try to repeat the file save 62 times, the FS works OK, the device can be safely rebooted w/o data loss.
After the reboot only one ```save``` operation is needed to reach "api_spiffs_erase" and the FS gets immediately broken. On the next reboot it is formatted  by the firmware.

Example code is provided here: https://github.com/kmihaylov/Sming_fsTest.git

Tested with commit  5300c0b (current develop)

```
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.6-beta1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:91:cf:b1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 2032 bytes to 1429...
Wrote 2032 bytes (1429 compressed) at 0x00000000 in 0.1 seconds (effective 123.9 kbit/s)...
Hash of data verified.
Compressed 275280 bytes to 196722...
Wrote 275280 bytes (196722 compressed) at 0x00002000 in 17.5 seconds (effective 126.1 kbit/s)...
Hash of data verified.
Compressed 65536 bytes to 107...
Wrote 65536 bytes (107 compressed) at 0x00100000 in 0.0 seconds (effective 38119.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
--- DTR inactive ---
--- RTS inactive ---
r$␀$ܟ|␀�$�|␂␌␌␄�␄d�␄b|��␃�2�r�b�␌#��on�lNo���␌"␜p��l{l{lp�n�␐␃␌␄�␄l␄␄␌␄␌␌#␌o�|␃$�␄��␌#��on�␂l��␀$`␂�␛␒on␄l ␃␏␃os���n␄␌␒� ␂p�o�␐␂␌␄{�ܜ��␌␄␌␄c␄o�|␂����␌#��nN�␂␌␄l ␃�␒2nol�l ␃␏␃os���n␄␌b�� ␂␏rےo␌␄#��`␃rd�#�on���p|␄�b␜␀lB�␓op`␒#�2c�#��N␀��n�b�N#p␡�n␏l␜�o�␂n����␟␀␌␄�␌l$pc�� ␃␜�o�␂ol$n��#␄␌␌d�$l␏l␜�o�␂o�␒���b␌␌␄��␄lld~sl�b�no��d�␒non>␀�␓o܄�N�␃o␎$r$␄�on���N��|~�o␌�b␏l��␃�␛�o��r��o|�␌l␌ldl ␂��{�l�l␒�␄␌␄l ␃��r�d�␜�␛�␌␄␌$`␂��r�$␌��␀�␟␀␌dl ␂��␛�n�␃N␎$rdN␑-�ver275956 Commandhandlercommand status registered
276383 Commandhandlercommand echo registered
276693 Commandhandlercommand help registered
277128 Commandhandlercommand debugon registered
281304 Commandhandlercommand debugoff registered
285572 Commandhandlercommand command registered
Welcome to the Stream Command executor
292990 trying to mount spiffs at 0x00100000, length 65536
298026 fs.start: size:64 Kb, offset:0x100000

302024 First init file system
304608 sect_first: 100, sect_last: 10f

333003 6%
360833 12%
385806 18%
408859 25%
433852 31%
458776 37%
483740 43%
508669 50%
531721 56%
556627 62%
579633 68%
602679 75%
627674 81%
652644 87%
676649 93%
699670 100%
699874 formatted
701453 mount res: 0

706169 hwsDelegateDemo instantiated, waiting for data
mode : softAP(be:dd:c2:91:cf:b1)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
save 62

Full cmd is: save 62
Cmd1: save, cmd2: 62
6782446 Bytes written to fs: 10
6790006 Bytes written to fs: 20
6797567 Bytes written to fs: 30
6805116 Bytes written to fs: 40
6812667 Bytes written to fs: 50
6820200 Bytes written to fs: 60
6827732 Bytes written to fs: 70
6835249 Bytes written to fs: 80
6842771 Bytes written to fs: 90
6850274 Bytes written to fs: 100
6857774 Bytes written to fs: 110
6865261 Bytes written to fs: 120
6872752 Bytes written to fs: 130
6880222 Bytes written to fs: 140
6887684 Bytes written to fs: 150
6895146 Bytes written to fs: 160
6902660 Bytes written to fs: 170
6910205 Bytes written to fs: 180
6917749 Bytes written to fs: 190
6925280 Bytes written to fs: 200
6932808 Bytes written to fs: 210
6940322 Bytes written to fs: 220
6947828 Bytes written to fs: 230
6955321 Bytes written to fs: 240
6962820 Bytes written to fs: 250
6970297 Bytes written to fs: 260
6977769 Bytes written to fs: 270
6985232 Bytes written to fs: 280
6992692 Bytes written to fs: 290
7000132 Bytes written to fs: 300
7007628 Bytes written to fs: 310
7015012 Bytes written to fs: 320
7022632 Bytes written to fs: 330
7030144 Bytes written to fs: 340
7037635 Bytes written to fs: 350
7045125 Bytes written to fs: 360
7052602 Bytes written to fs: 370
7060087 Bytes written to fs: 380
7067544 Bytes written to fs: 390
7075001 Bytes written to fs: 400
7082442 Bytes written to fs: 410
7089877 Bytes written to fs: 420
7097296 Bytes written to fs: 430
7104715 Bytes written to fs: 440
7112122 Bytes written to fs: 450
7119526 Bytes written to fs: 460
7126913 Bytes written to fs: 470
7134340 Bytes written to fs: 480
7141796 Bytes written to fs: 490
7149252 Bytes written to fs: 500
7156688 Bytes written to fs: 510
7164122 Bytes written to fs: 520
7171544 Bytes written to fs: 530
7178966 Bytes written to fs: 540
7186368 Bytes written to fs: 550
7193766 Bytes written to fs: 560
7201155 Bytes written to fs: 570
7208537 Bytes written to fs: 580
7215917 Bytes written to fs: 590
7223281 Bytes written to fs: 600
7230636 Bytes written to fs: 610
7238041 Bytes written to fs: 620
--- RTS active ---
--- RTS inactive ---
{l␀$��|␀�$�|␃␄␌␄�␄d�␄b|����␛�{�"�␄b��no�$on��#␄c␜p�c$`␂r$p�o�␐␂␌␌�␌$␌␄␌␌␄␌B␌n�|␃$�␄��p␄��on�␂l��␀$ ␂�␛␒nN␌l ␂␏␂nr����␂␌␄␛�`␃p�o�␐␂␌␄r�����␌␌␄␌#␌N�|␃���␄␌"��on�␂␌␄l`␃�␒2nol�l`␃␎␃or���N␌␌"��`␂␎{��n␄␌b�� ␂sl�b�no߄���c�|�␌$b�2np ␛b�␒"�b��o␀���␂␛Nobp~�o␎$␜�n�␃o�o~␒"␌␄␌�␌$lpb��`␂␜�n�␃n$lo���␌␌␄l�ll␎$␜�n�␃n�␛�ۂB␌␄␌�␌$dl␡rd�#�on��$�␒oNo␞␀�␛o܄�n�␃o␎$r$␌�oo��|�␂��o�␂␄�#␎$��␒�N�{��o|�␄l␄ldl ␂�o�$��: 0079599 Commandhandlercommand status registered
80022 Commandhandlercommand echo registered
80328 Commandhandlercommand help registered
80640 Commandhandlercommand debugon registered
84597 Commandhandlercommand debugoff registered
88778 Commandhandlercommand command registered
Welcome to the Stream Command executor
96109 trying to mount spiffs at 0x00100000, length 65536
101058 fs.start: size:64 Kb, offset:0x100000

106535 mount res: 0

106895 hwsDelegateDemo instantiated, waiting for data
mode : softAP(be:dd:c2:91:cf:b1)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
save 1

Full cmd is: save 1
Cmd1: save, cmd2: 1
4205519 api_spiffs_erase
4230420 api_spiffs_erase
4261372 Bytes written to fs: 10
--- RTS active ---
--- RTS inactive ---
{l␀l���␃�l�|␂␌␄␌�␄l�␌b|��␂�␛�r�b�␄c��on�lNo���␌"␜p�l{lrdp�o�␐␃␄␌�␌l␄␌␄␄␌␄c␌o��Nl�␄��␄c��nN�␂l��␀l ␂�2␛on␄l`␃␎␃N{���n␄␌␛�� ␃p�o�␐␃␄␌r�ܜ��␌␄␌␄c␌o�|␃���␄␌b��No�␃␄␌$ ␂�␒␛nol�l`␃␎␃N{���n␄␄#��`␃␎{��n␄␌b��␎l{l�#�oN���p|␌�b␜␀lb�␛Np ␛b�␒"�b��o␀��n�"�Ncp~�n␏l␜�o�␂o�o~␒b␌␄␌�␜�␌$dpc�� ␃␜�o�␂ol$n���␌␄␌$�l$␎$␜�n�␃N�␛�ۂ"␌␄␌��␄l$l~r$�b�No��l�␒noo␞␀�␛o܌�n�␂n␎l{l␌�no�#�o܄�n�␂␌�#␏d��␃�␛�n��r��n|�␌$␌$l$`␂�o�$��: 0079745 Commandhandlercommand status registered
80169 Commandhandlercommand echo registered
80475 Commandhandlercommand help registered
80786 Commandhandlercommand debugon registered
84743 Commandhandlercommand debugoff registered
88924 Commandhandlercommand command registered
Welcome to the Stream Command executor
96256 trying to mount spiffs at 0x00100000, length 65536
101204 fs.start: size:64 Kb, offset:0x100000

105203 First init file system
107787 sect_first: 100, sect_last: 10f

136173 6%
163947 12%
188700 18%
211540 25%
236320 31%
261114 37%
285814 43%
310556 50%
333560 56%
358432 62%
381405 68%
404403 75%
428366 81%
453333 87%
476369 93%
501263 100%
501466 formatted
503041 mount res: 0

507755 hwsDelegateDemo instantiated, waiting for data
mode : softAP(be:dd:c2:91:cf:b1)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
```

mikee47, 2019-04-07:
@kmihaylov You've highlighted the problem very accurately! The logic in `spiffs_mount_internal()` (`spiffs_sming.c` line 114) is too simplistic -

```
  uint32_t dat;
  bool writeFirst = false;
  flashmem_read(&dat, cfg->phys_addr, 4);
  //debugf("%X", dat);

  if (dat == UINT32_MAX)
  {
	  debugf("First init file system");
	  spiffs_format_internal(cfg);
	  writeFirst = true;
  }
```

The volume gets wiped if SPIFFS sector #0 is in the erased state when the device reboots, which is a perfectly legitimate state for SPIFFS to be in.

Here's a better way:

```
// Call SPIFFS_mount() with appropriate parameters
int res = do_mount();
// Check result
if(res < 0) {
	// Mount failed, format the volume then retry
	format();
	res = do_mount();
}
// Final result of mount operation
return res;
```

There's also a volume checking function in SPIFFS, which is time-consuming but could be integrated like this:

```
int res = do_mount();
if(res < 0) {
	// Mount failed, perform a volume check
	check_fs();
	res = do_mount();
	if(res < 0) {
		format();
		res = do_mount();
	}
}
return res; // Final result of mount operation
```

You can find some example code for the above [here](https://github.com/mikee47/IFS/blob/master/src/IFS/SPIFlashFileSystem.cpp)

kmihaylov, 2019-04-08:
I did as you suggested:

https://github.com/kmihaylov/Sming/blob/fix/spiffs_mount_prevent_format/Sming/Services/SpifFS/spiffs_sming.c

Now it successfully mounts the fs every time (even if I flashed the device with DISABLE_SPIFFS = 1). The files are written to the FS, but I'm curious how i works if it wasn't inited. Also the problem of unnecessary inited FS disappeared and the files seem to contain the data they should contain.

SPIFFS magically inits the FS on the fly or something else happens that I don't understand yet.

mikee47, 2019-04-08:
Happy that's working for you! The `DISABLE_SPIFFS` only controls whether an image is built and uploaded during flash.

>SPIFFS magically inits the FS on the fly

Pretty much! It doesn't erase sectors unless it needs the space. Have you tested the `check_fs()` part yet? If you wanted to do some stress testing, try filling the flash with random data first. That should fail the first mount operation and force it to do a `check_fs()`.

kmihaylov, 2019-04-08:
It mounts the FS despite it was prefilled with random data.

```
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m \
0x00000 out/firmware/rboot.bin \
0x02000 out/firmware/rom0.bin \
0x100000 random.bin && python -m serial.tools.miniterm /dev/ttyUSB0 115200
```
The mount() function is called 10 seconds after boot to ensure that the debug messages refer to the first mount of the random-data space (I had fear that it formats the FS too quickly).

fileSetContent however now returns ```open errno -10001```.

FS is unusable.

--------------------------

Flashing zeros to the SPIFFS area enables successful fileSetContent() operation, but the FS check returns "-10024"

```
$ /opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x100000 zero.bin 
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.6-beta1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:91:cf:b1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 65536 bytes to 84...
Wrote 65536 bytes (84 compressed) at 0x00100000 in 0.0 seconds (effective 44188.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
$ python -m serial.tools.miniterm /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
--- DTR inactive ---
--- RTS inactive ---
{l␀$ܞ|␀�l�|␂␌␄␌�␄l�␌#|����␒�r�B�␌#�p~�o�lon���␌B␜p��cl ␃{lp�N�␐␂␌␄�␌$␌␄␌␄␌␌B␌o��ol�␄��␄b��oN�␃d��␀$`␂�2␒oN␌$`␃␎␃or�ےn␄␌2�� ␂p�n�␐␂␌␄{�ܜ��␌␄␌␄b␄N�|␂����␌"��no�␃␌␄l ␂�␒␛ond�␜{l␏␂n{���o␌␌b��`␃␎rےn␄␌B�� ␂r$�#�no���p|␄�|�␄lB�|Br␃N��␓B�#��o␀��␛␒no"p␡�n␎$␜�N�␂n�o~␒#␄␌␄�␄l$pc��`␂␜�o�␂nl$n��c␄␌␌$�$l␎$␜�o�␂N�␒����>␀␌␄��␌$l$~r$�#�no��$�␛Nno␞␀�␒n���N�␂n␎$rd␌�on�#�n��|␡�o␄�#␏l��␃�␓�o�r��o|�␌l␄$ll ␂��r�l��: 0079839 Commandhandlercommand status registered
80249 Commandhandlercommand echo registered
80548 Commandhandlercommand help registered
80850 Commandhandlercommand debugon registered
84834 Commandhandlercommand debugoff registered
89011 Commandhandlercommand command registered
Welcome to the Stream Command executor
96380 hwsDelegateDemo instantiated, waiting for data
mode : softAP(be:dd:c2:91:cf:b1)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
10101001 FS check res: -10024
10101225 trying to mount spiffs at 0x00100000, length 65536
10101420 fs.start: size:64 Kb, offset:0x100000

10103543 mount res: 0

save 1

Full cmd is: save 1
Cmd1: save, cmd2: 1
27644332 api_spiffs_erase
27688299 api_spiffs_erase
27712270 api_spiffs_erase
[... 174 lines cut ...]
33531747 Bytes written to fs: 960
33539296 Bytes written to fs: 970
33546829 Bytes written to fs: 980
33554356 Bytes written to fs: 990
33561877 Bytes written to fs: 1000
--- RTS active ---
--- RTS inactive ---
rd␀lܟ|␀�$�|␂␄␌␄�␌d�␌#|��␂�2�{��␟�␄b��no�loN���␌c␜p��$r$rlp�n�␐␂␌␄�␌$␌␄␄␌␄␌c␄n�|␂l�␌��␄b��no�␃l��␀$`␃�␒␛on␌$`␃␏␂n{���o␌␌␒� ␃p�n�␐␂␌␌r�����␄␄␌␄b␌N��o���␄␄b��oN�␃␌␄l ␂�2␒ool�$`␃␏␂n{���o␌␌b�� ␃␏r��o␌␌b�� ␃{$�#�no�����b�"␜␀$b�␒np ␒c�␒#��|��n␀���2␒no"p␡�N␎d␜�o�␃o���B␄␌␄�␄ldpb��`␃␜�n�␂o$lo���␄␄␌$�d$␎l␜�n�␃o�␛���␐B␄␌␄��␌$ld~rl�b�oN�␀�l����oN␟␀�␓o���N�␂N␎drl␌�Nn��|�␂�|␡�n␌�b␎$��␃�␒�n�s��N|�␌d␄lld ␃�n�l��: 0080305 Commandhandlercommand status registered
80716 Commandhandlercommand echo registered
81015 Commandhandlercommand help registered
81317 Commandhandlercommand debugon registered
85301 Commandhandlercommand debugoff registered
89477 Commandhandlercommand command registered
Welcome to the Stream Command executor
96846 hwsDelegateDemo instantiated, waiting for data
mode : softAP(be:dd:c2:91:cf:b1)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
10101468 FS check res: -10024
10101691 trying to mount spiffs at 0x00100000, length 65536
10101886 fs.start: size:64 Kb, offset:0x100000

10103999 mount res: 0
```

mikee47, 2019-04-09:
This is probably an unrealistic test, for the flash to get in that state would probably indicate other problems so perhaps not realistic. For more robust operation, try setting `SPIFFS_USE_MAGIC=1` in `spiffs_config.h` - further details [here](https://github.com/pellepl/spiffs/wiki/Using-spiffs).

kmihaylov, 2019-04-10:
@mikee47 respect for your guidance. It's very helpful.

I put
```#define SPIFFS_USE_MAGIC                (1)```
and 
```#define SPIFFS_USE_MAGIC_LENGTH   1``` 

And now after ```make flashinit``` it works almost as expected:
1. FS not found;
2. Format initiated
3. Unsuccessful mount.

I did a dump of the formatted partition and it is 0xFF everywhere, obviously there are no magic numbers.

So far, so good.

I thought to put some data (in order for the FS to put some magic numbers inside), but since the FS couldn't be mount this option is ignored.

I read ```spiffs_format_internal``` in spiffs_sming.c and I don't see the function to write some magic numbers during format.

So I guess it only wipes the flash
```c
while( sect_first <= sect_last )
  {
	if(flashmem_erase_sector( sect_first++ ))
	{
		int percent = ++cur * 100 / total;
		if (percent > last)
			debugf("%d%%", percent);
		last = percent;
	}
	else
	{
		ETS_INTR_UNLOCK();
		return false;
	}
  }
```

UPDATE:
I just found the correct procedure for initializing SPIFFS with magics: https://github.com/pellepl/spiffs/wiki/Using-spiffs

```
If SPIFFS_USE_MAGIC is enabled, mounting procedure will look for magic. Recommended formatting procedure is as point two above. Recapped:

    Call SPIFFS_mount

    If SPIFFS_mount fails with SPIFFS_ERR_NOT_A_FS, keep going. Otherwise, call SPIFFS_unmount

    Call SPIFFS_format

    Call SPIFFS_mount again.
```

Well, I tried to modify the code as pointed by the author but with no luck (FS just coudn't be mount the second time.

Currently I plan to neglect USE_MAGIC and to replace this code:
```c
  if (dat == UINT32_MAX)
  {
	  debugf("First init file system");
	  spiffs_format_internal(cfg);
	  writeFirst = true;
  }

  int res = SPIFFS_mount(&_filesystemStorageHandle,
    cfg,
    spiffs_work_buf,
    spiffs_fds,
    sizeof(spiffs_fds),
    spiffs_cache,
    sizeof(spiffs_cache),
    NULL);
  debugf("mount res: %d\n", res);

  if (writeFirst)
  {
	  file_t fd = SPIFFS_open(&_filesystemStorageHandle, "initialize_fs_header.dat", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
	  SPIFFS_write(&_filesystemStorageHandle, fd, (u8_t *)"1", 1);
	  SPIFFS_fremove(&_filesystemStorageHandle, fd);
	  SPIFFS_close(&_filesystemStorageHandle, fd);
  }
```

with this one:
```c
file_t fd = SPIFFS_open(&_filesystemStorageHandle,"initialize_fs_header.dat",SPIFFS_RDONLY,0);
  SPIFFS_close(&_filesystemStorageHandle, fd);
	if(fd < 0) {
		debugf("No initialized SPIFFS volume found. Initializing FS now...");
		file_t fd = SPIFFS_open(&_filesystemStorageHandle, "initialize_fs_header.dat", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
		SPIFFS_write(&_filesystemStorageHandle, fd, (u8_t *)"1", 1);
		//SPIFFS_fremove(&_filesystemStorageHandle, fd);
		SPIFFS_close(&_filesystemStorageHandle, fd);
	} else {
		debugf("It seems that the mounted SPIFFS is properly inited.");
	}
```

However as indicated in my previous post, if there is some previous data on the flash (wiped flash is all 0xFF as I undestand?) it may lead to unusuable filesystem.

So it is pretty good to erase the flash before use.
- One way is to initiate make flashinit or just to flash the spiffs rom image with some files.
- SPIFFS_mount() could not be trust for recognizing partition.
- I didn't succeed with MAGICs, so this is no option too.
- I'm thinking for manually setting the file "initialize_fs_header.dat" w/o deleting it. One can try to mount the FS, check if file exists and if not - first erase the flash, then init the FS (which is just writing a byte of information as I see?).

slaff, 2020-02-04:
@kmihaylov Can you fetch the latest `develop` version and check if you still have problems with spiffs?