SmingHub/Sming #1670: Debugger crashes after continuation

slaff, 2019-04-13:
Multiple continuation commands can break a debugging session. 

A shorter version of the commands below can be used to reproduce the issue:
```
make gdb
(gdb) c
(Attached) break
(gdb) br blink
(gdb) c
(gdb) [ENTER]
(gdb) [ENTER]
(Attached) [ENTER]
```

A longer version with recording from a real debugging session.

```
1. New LiveDebugg application is flashed on the NodeMCU device.
2. make gdb is executed.
---
For help, type "help".
Type "apropos word" to search for commands related to "word".
0xffffffff in ?? ()
overlapping memory region
overlapping memory region
overlapping memory region
overlapping memory region
overlapping memory region
overlapping memory region
add symbol table from file "out/build/bootrom.elf" at
	.text_addr = 0x40000000

Welcome to SMING!
Type 'c' (continue) to run application

A program is being debugged already.  Kill it? (y or n) y
Remote debugging using /dev/ttyUSB0
gdbstub_init () at /home/slavey/dev/esp8266.dev.box/dev/Sming/Sming//gdb/gdbstub.cpp:908
908			gdbstub_do_break();
(gdb) 
(gdb) c
Continuing.
LiveDebug sample
Explore some capabilities of the GDB debugger.

[OS] mode : sta(18:fe:34:f3:74:e7)
[OS] add if0
252348026 GdbAttach(1)
252391206 open log 0
(Attached) break
255528829 gdb_read_console() returned 6
Calling gdb_do_break()

Program received signal SIGTRAP, Trace/breakpoint trap.
gdb_do_break () at /home/slavey/dev/esp8266.dev.box/dev/Sming/Sming//gdb/gdbstub.cpp:937
937		gdbstub_do_break();
(gdb) br blink
Breakpoint 1 at 0x40105d4c: file app/application.cpp, line 53.
(gdb) c
Continuing.

Breakpoint 1, blink () at app/application.cpp:53
53	{
(gdb) 
Continuing.

Breakpoint 1, blink () at app/application.cpp:53
53	{
(gdb) 
Continuing.
(Attached) 

***** Fatal exception 20 (INSTR_PROHIBITED)
pc=0xffffffff sp=0x3ffffec0 excvaddr=0x00000000
ps=0x00000030 sar=0x0000001e vpri=0x402072cc
r00: 0xffffffff=        -1 r01: 0x3ffffec0=1073741504 r02: 0x00000000=         0 
r03: 0x00000001=         1 r04: 0x00000000=         0 r05: 0x3ffeebf0=1073671152 
r06: 0x40224020=1075986464 r07: 0x3ffe8ac0=1073646272 r08: 0x00000008=         8 
r09: 0x00000000=         0 r10: 0x3ffe8f64=1073647460 r11: 0x3ffe8a9c=1073646236 
r12: 0x3fffc230=1073726000 r13: 0x401062fc=1074815740 r14: 0x3fffc200=1073725952 
r15: 0x00000022=        34 

Program received signal SIGSEGV, Segmentation fault.
0xffffffff in ?? ()
```

slaff, 2019-04-13:
@mikee47 Can you check if you can reproduce this issue on your device(s)?

mikee47, 2019-04-13:
@slaff Yes, I get the same. It happens because GDB is in the middle of a syscall and ignores anything sent to it. I need to investigate further to understand exactly why it's producing this particular behaviour.

If you set `GDB_ENABLE_SYSCALL=0` in `gdbstub-cfg.h` and rebuild the problem goes away, although we can't interact with GDB in the sample.

mikee47, 2019-04-13:
OK, so the `blink` debug break takes us into the debugger command loop, but GDB doesn't know we've stopped because it's still processing the syscall. When the 'F' response (which completes the syscall) is received, it takes us back out of the debugger loop immediately, which is wrong; at that point we should tell GDB again why we've stopped so things can proceed as for a regular break handling.

When I change the code to fix this, it starts working again although I still need to hit enter to complete the console read before we know there's been a break. I can continue and single step without issue.

However, when I test the code with a regular (non-hardware) timer, it breaks but then hangs at the `gdbstub_icount_ena_single_step()` call when I attempt to continue: GDB sends a 's' (step) instruction at this point - no idea why. Performing an `si` command has the same effect, it does the same. Investigation continues.

I'll open a PR with the changes so far...

mikee47, 2019-04-13:
@slaff OK, I think that's fixed. Let me know how you get on.

slaff, 2019-04-13:
> OK, I think that's fixed.

Let me check...

slaff, 2019-04-13:
It looks better from the gdb cli with manually typing commands. But I would suggest to disable by default the GDB_ENABLE_SYSCALL. At the moment the output looks a bit confusing to me:

```
(gdb) br blink
Breakpoint 2 at 0x40105d4c: file app/application.cpp, line 53.
(gdb) c
Continuing.

Breakpoint 2, blink () at app/application.cpp:53
53	{
(gdb) 
Continuing.
//  Repeated 5 or 6 times...
// Then comes
(Attached) 
// I type help but  get thrown back to gdb

Breakpoint 2, blink () at app/application.cpp:53
53	{
(gdb) 
Continuing.
(Attached) 

Breakpoint 2, blink () at app/application.cpp:53
53	{
(gdb) 
Continuing.
// Repeated multiple types

[OS] chg_B1:-40
// And then finally comes the help output

LiveDebug interactive debugger sample. Available commands:
  readfile1 : Use syscall file I/O functions to read and display a host file
```

Additionally I tried to use Eclipse as a visual debugger. Setting breakpoints and stepping works nicely. But if I remove a breakpoint and set again a new one on the same line the debugger seems to freeze. I guess Eclipse does not send another `br blink` but stops the execution before setting the breakpoint because I can see the following:
```
Program received signal SIGINT, Interrupt.
0x4010612e in gdb_uart_callback (uart=<optimized out>, status=<optimized out>) at /path/to/Sming/Sming//gdb/gdbuart.cpp:342
342	
```

slaff, 2019-04-14:
Fixed in ##1671.

closed 2019-04-14 by slaff