SmingHub/Sming #1683: Migrate Sming documentation to Sphinx/readthedocs

mikee47, 2019-05-04:
The quality of the documentation available for the ESP-32 IDF is very good, easy to navigate and well organised. For example, take https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/vfs.html. We get a good introduction, some examples and details of the API all on the same page.

The Wiki is easy update and to submit ideas to, but it's not particularly easy to organise and hard to find what we need. Moving the Sming documentation to the same build system would offer some advantages:

* Better quality of documentation
* Managed within GIT so we can manage it like source code, and keep it in sync with the source tree(s)
* Automated documentation builds (via readthedocs)
* Documentation in a variety of formats (PDF, HTML)
* Style consistent with ESP-IDF (also Esp8266 and Esp32 Arduino) so linking to documentation in those projects would probably work better
* Ability to combine existing Wiki documentation with source code (doxygen) documentation

I've opened this issue to hopefully start a discussion!

slaff, 2019-05-10:
@mikee47 Do you know if there is a good `Sphinx/readthedocs` tutorial that can teach not only us, but also all developers that are contributing or plan to contribute to Sming?

mikee47, 2019-05-10:
I haven't delved too deeply into this yet, but [Getting Started with Sphinx](https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html) is a good starting point, especially the three references at the foot of the page. If I find anything more succinct I'll post it up.

mikee47, 2019-05-10:
I also came across https://pandoc.org/ which may be useful converting existing markdown docs into reStructuredText. There's a cut'n'paste version here which I tried briefly https://pandoc.org/try/

slaff, 2019-05-10:
So we have the Wiki documentation using markdown and we have the doxygen API documentation. Shall we convert both or only the markdown one?

mikee47, 2019-05-10:
> 
> 
> So we have the Wiki documentation using markdown and we have the doxygen API documentation. Shall we convert both or only the markdown one?

Definitely both, but I think the important thing is to get the basic structure in place so all the related stuff is together, even if it's just a load of placeholders. We can then work on it section-by-section, importing from the Wiki/doxygen output as required. The API documentation can be in its own section, but restructured so, for example, all the Streams are in one section with a related overview.

The IDF docs build system uses [Breathe](https://breathe.readthedocs.io/en/latest/) to 'provide a bridge between the Sphinx and Doxygen documentation systems' and uses doxygen's XML output. That's more involved but we can look at that perhaps once we've decided on the basic structure.

mikee47, 2019-07-08:
I think we're now starting to have difficulty with documentation, trying to keep the Sming 3.9+ and 3.8 LTS separate and yet in sync with the code.

With major stuff I've documented it separately:

* Building is now handled differently - documented in `Sming/building.md`
* Host arch. (emulator) documented in `Arch/Host/readme.md`

Some of this has made it into the Wiki, but that's only a brief intro. and certainly don't want to have to maintain in two separate places. Many Components already have their own documentation (including Arduino Libraries) so the job of the documentation system should be to pull that together systematically.

If we can get this sorted for 3.9 release then it's just a matter of diverting users to the new docs, leaving the Wiki in place for LTS. Unless anyone else has started work on this issue I intend to start working up a PR (@slaff, I've started work on porting SSL for the emulator, but I think this takes priority.)

slaff, 2019-07-08:
> (@slaff, I've started work on porting SSL for the emulator, but I think this takes priority.)

I am waiting anxiously for this :) 

> Unless anyone else has started work on this issue I intend to start working up a PR 

Ok, go for it.

mikee47, 2019-07-09:
This is going a lot faster than I'd envisaged so should have something shortly. Just getting the structure right, then I'll go through everything to ensure it's consistent for 3.9.

@slaff To host online documentation I guess we'll need an account on readthedocs, setting up webhooks for building etc. https://docs.readthedocs.io/en/stable/config-file/index.html.

slaff, 2019-07-09:
>  To host online documentation I guess we'll need an account on readthedocs, setting up webhooks for building etc. https://docs.readthedocs.io/en/stable/config-file/index.html.

Working on it... we are waiting for the owner of SmingHub to grant access to read-the-docs.

slaff, 2019-07-09:
Meanwhile I can manually generate the documentation. For the moment we have the following page:
https://sming-framework.readthedocs.io/en/latest/

mikee47, 2019-07-09:
Any ideas for what to put in the 'copyright' text in the page footers?

mikee47, 2019-07-09:
OK, first rough draft in my repo https://github.com/mikee47/Sming.git branch is feature/sphinx-docs. Should have everything from the Wiki, converted into reStructuredText.

I have not made any changes to the content yet, this is just looking at the mechanics and pulling in the reference sources. The makefile is separate at the moment, so go into `docs/sphinx` and do `make html`.

All the docs. have to be under 'source', which means we can't directly pull anything in from the source trees. So, in the makefile I have a list of files to be copied into the source folder for building. At the moment it's just two files, `Arch/Host/readme.md` and `Sming/building.md`.

This demonstrates that we can have a mix of reStructuredText and markdown, so we don't necessarily need to go converting everything.

slaff, 2019-07-09:
> Any ideas for what to put in the 'copyright' text in the page footers?

Alexey Skurydin (a.k.a anakod) and [contributors](https://github.com/SmingHub/Sming/graphs/contributors) or 
[Sming Developer Team](https://github.com/SmingHub/Sming/graphs/contributors) .

slaff, 2019-07-09:
So, on my system also the following python package was required: `recommonmark`. Maybe that should be added to the requirements file.

Running `make html` ends up with the following error:
```
[slavey@/media/sla....dev.box/dev/Sming/docs/sphinx] make html
Running Sphinx v1.7.9
None
WARNING: The config value `source_suffix' has type `dict', defaults to `list'.
loading pickled environment... not yet created
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 32 source files that are out of date
updating environment: 32 added, 0 changed, 0 removed
/usr/local/lib/python2.7/dist-packages/recommonmark/parser.py:65: UserWarning: Container node skipped: type=document                                                                    
  warn("Container node skipped: type={0}".format(mdnode.t))
reading sources... [100%] troubleshooting/windows                                                                                                                                       
/media/....dev.box/dev/Sming/docs/sphinx/source/experimental/index.rst:5: WARNING: toctree contains reference to nonexisting document u'experimental/signed-ota'

Sphinx error:
master file /media.../dev/Sming/docs/sphinx/source/contents.rst not found
Makefile:19: recipe for target 'html' failed
```

What am I doing wrong?

mikee47, 2019-07-09:
I've got sphinx version 2.1.2, maybe try a later version?

mikee47, 2019-07-09:
The missing file is a capitalisation issue, will fix. I've stuck to a lower-case-only convention for filenames.

slaff, 2019-07-10:
> The missing file is a capitalisation issue, will fix. I've stuck to a lower-case-only convention for filenames.

The file `source/contents.rst` is not added to the repository. Check if you have it locally only.

mikee47, 2019-07-10:
I don't have `source/contents.rst`, where's it referenced from?

slaff, 2019-07-10:
> I don't have source/contents.rst, where's it referenced from?

Good question. Either the version that I use requires it by default (Sphinx v1.7.9) or some of the markup tags as require it. The error that I get is 

```
Running Sphinx v1.7.9
None
WARNING: The config value `source_suffix' has type `dict', defaults to `list'.
loading pickled environment... not yet created
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 32 source files that are out of date
updating environment: 32 added, 0 changed, 0 removed
/usr/local/lib/python2.7/dist-packages/recommonmark/parser.py:65: UserWarning: Container node skipped: type=document                                                                    
  warn("Container node skipped: type={0}".format(mdnode.t))
reading sources... [100%] troubleshooting/windows                                                                                                                                       

Sphinx error:
master file /media/slavey/416431e8-fd46-4982-896d-5025a72dd361/slavey/dev/esp8266.dev.box/dev/Sming/docs/sphinx/source/contents.rst not found
Makefile:19: recipe for target 'html' failed
```

slaff, 2019-07-10:
Ok,after looking at that page: https://github.com/readthedocs/readthedocs.org/issues/2569 I guess the adding `master_doc = 'index'` in conf.py is what is needed. At least it works for me now.

mikee47, 2019-07-10:
http://www.sphinx-doc.org/en/master/usage/configuration.html#confval-master_doc

`Changed in version 2.0: The default is changed to 'index' from 'contents'.`

I'll add it explicitly. Should probably decide which version to standardise on, http://www.sphinx-doc.org/en/master/changes.html.

mikee47, 2019-07-10:
I've downgraded to sphinx 1.8.5 for better compatibility with breathe (for API docs).

slaff, 2019-07-11:
> Should probably decide which version to standardise on, http://www.sphinx-doc.org/en/master/changes.html.

Let's go for 1.8 but with compatibility for 2.x whenever possible. 

@mikee47 Event at that early stage the documentation looks quite promising. 

[image]

My thoughts on organizational changes related to the documentation

# Samples
We should try to exforce documentation style for the samples. For example there should be a Readme.md file that can be used directly in the Sphinx documentation. That readme file should describe
- what is the purpose of the sample
- what is demonstrated
- useful information for beginner developer, related to the sample
- (optional) screenshot
- etc.

And that should replace the Wiki page with the outdated samples and their information.

# Arduino Libraries
It would be great if we can get documentation about the Arduino Libraries directly from their projects. Or just provide a list with the supported Arduino libraries and URL links of the original project(if available)

# Introduction
There should be a short introduction page that is the first to be opened. Best would be if we use directly the Readme.md file in the root of the repository. 

# Multi-arch support
Should be documented and present on the first page(s).

# Documentation Per Version
Last, but not least, in readthedocs we should have separate documentation per version. For example https://sming-framework.readthedocs.io/en/latest/ should lead to the latest `develop` version. https://sming-framework.readthedocs.io/en/stable/ - should be latest `master` version. And 
https://sming-framework.readthedocs.io/en/x.y.z/ should be for version  x.y.z. Example 3.8.0.

mikee47, 2019-07-11:
For samples, I've used reStructuredText so we can integrate it better. Documenting the libraries is tricker because it's a mixture of various file types. I've also started migrating the API documentation, way to go though. The ESP-IDF incorporates custom roles (in `docs/link-roles.py`) so links to specific examples can be added. That is something I'm considering here because we can then break everything down logically and refer to relevant samples.

mikee47, 2019-07-11:
I've plenty to do getting the structure and API sorted, so don't intend to do anything with the sample documentation for now. If anyone would like to lend a hand with that please do! Just check out my working branch and submit a PR with any changes you'd like. Thanks!

mikee47, 2019-07-12:
@slaff Are we still considering these elements experimental?

* LWIP2 #1289 22/11/2017
* SDK 3 #1470 12/10/2018

slaff, 2019-07-12:
> @slaff Are we still considering these elements experimental?
> 
> * LWIP2 #1289 22/11/2017
> * SDK 3 #1470 12/10/2018

* SDK 3 is no longer experimental. 
* LWIP 2 should also be considered stable. Apart from the fact that espconn_* functions will not work with it. Meaning that also SmartConfig will not work. Unless we use the original source code from the espconn functions and make them compatible with LWIP 2.

mikee47, 2019-07-12:
Just reviewing docs so they're consistent with current state.

slaff, 2019-07-12:
> Just reviewing docs so they're consistent with current state.

Once you are ready with the structure me and the some of the other contributors will help you revise the documentation.

mikee47, 2019-07-12:
I've pushed some commits, getting there I think. I need to add some notes on how to edit documentation but it's a bit less top-heavy now with the documentation sitting in each Component, Library or sample. I've added some 'magic roles' so we can just add stuff like

```
:sample:`Basic_Blink`
```

to insert a reference to a specific sample, using the title in the document by default. I'd like to automate some other stuff, such as including a link on every sample readme page to the source code on github, for the appropriate version, but that can be done as an improvement when we've worked out how to do it!

We can add doxygen information using directives from this page `https://breathe.readthedocs.io/en/latest/directives.html`. There is a sphinx extension called [exhale](https://exhale.readthedocs.io/en/latest/) which itself uses `breathe`, but I think we can achieve what we need using `breathe` directly so probably no need. In fact, it's probably better since it gives us precise control over what to add on each page. I think any additional work will be in improving the javadoc comments in the source code.

mikee47, 2019-07-12:
The list of Arduino Libraries is coming up short at the moment as I'm only showing those with readme files available. I'll take another look at that to see how it can be improved.

mikee47, 2019-07-12:
As for building the docs, you need to do it from a project directory (using `make docs`) because we need to pull in (and patch) all submodules so the documentation is available. Note there are some warnings which I need to address.

slaff, 2019-07-13:
@mikee47 Why don't you submit your current work as PR with  "[WIP] title"? We can start putting our comments there and also be able to suggest changes by sending PR to your repo.

slaff, 2019-07-28:
@frankdownunder @riban-bw Can you take a look at the latest sphinx documentation in the `develop` branch and help us improve it?

mikee47, 2019-07-28:
Just for info, I'm working on some documentation additions under the general topic `Sming Multitasking` as it's crucial to understanding how Sming works, so will include:

* Background on pre-emptive vs. co-operative multitasking, with relative advantages/disadvantages
* Callbacks / Delegates
* Task queues
* Interrupts
* Dealing with time-consuming operations
* Common problems

I hope that the new doc. system will encourage a new way of working, in that submission of samples, Components, Libraries, etc. will include better documentation which will get automatically pulled into the build - all it needs is a `README.rst` file.

slaff, 2019-09-27:
@mikee47 Is there anything left to do related to the documentation or I can close this discussion?

mikee47, 2019-09-27:
@slaff I've nothing further planned for documentation structure, so yes think we can close this thread.

closed 2019-09-28 by slaff

mikee47, 2019-09-29:
@slaff Just to make you aware I'm still having to update the docs. manually via https://readthedocs.org/projects/sming/builds/

slaff, 2019-09-29:
> Just to make you aware I'm still having to update the docs. manually via

I know :(.
@alonewolfx2 @anakod @hreintke Can someone from you guys grant access to read-the-docs so that we can push generated documentation automatically?

mikee47, 2019-10-07:
@slaff For info got some documentation tidying up to submit (get rid of 'to be completed' in sample README files, few fixes, github links and so on). Probably have it in a day or two.