SmingHub/Sming #1707: What is the proper way to run valgrind with the host emulator?

slaff, 2019-05-31:
What is the proper way to run valgrind with the host emulator on 64 bit operating system? Running compiled and working samples under valgrind report false-positive issues.

# Initial Setup:
- 64 bit Ubuntu 16.04 OS.
- GCC C/C++ compiler with support for 32 bit architecture. Installed gcc-multilib, g++-multilib.
- The compilation process for the Host architecture produces 32 bit code. 
- The compiled application runs without issues.
- Installed 64 bit valgrind.
- installed `libc6-dbg:i386` to allow valgrind to be able to analyze 32 bit applications.

# Expected result
When running valgrind on 64 bit operating system and trying to analyze the compiled program to be able to catch real memory issues.

# Actual result
As soon as there is a memory operation application segfaults and valgrind reports memory issues. As can be seen below.

Valgrind is started using:

```
valgrind  --track-origins=yes --leak-check=full out/firmware/app --flashfile=flash.bin --uart=0 --uart=1 --pause=5 --flashsize=4M
```

And soon after the start the following error shows up.
```
main: If required, you may start terminal application(s) now
pause: Waiting for 5 seconds...
main: >> Starting Sming <<

==29482== Invalid read of size 2
==29482==    at 0x8064F5A: operator+(StringSumHelper const&, String const&) (WString.cpp:358)
==29482==    by 0x806A51B: HttpClientConnection::multipartProducer() (HttpClientConnection.cpp:99)
==29482==    by 0x8069725: MethodCaller<HttpPartResult (HttpClientConnection::*)()>::invoke() (Delegate.h:83)
==29482==    by 0x806F777: operator() (Delegate.h:172)
==29482==    by 0x806F777: MultipartStream::getNextStream() (MultipartStream.h:50)
==29482==    by 0x806FD2E: MultiStream::readMemoryBlock(char*, int) (MultiStream.cpp:29)
==29482==    by 0x80751C3: StreamTransformer::readMemoryBlock(char*, int) (StreamTransformer.cpp:36)
==29482==    by 0x8067A8C: TcpConnection::write(IDataSourceStream*) (TcpConnection.cpp:264)
==29482==    by 0x80681D9: TcpClient::pushAsyncPart() (TcpClient.cpp:157)
==29482==    by 0x806A3A2: HttpClientConnection::onReadyToSendData(TcpConnectionEvent) (HttpClientConnection.cpp:295)
==29482==    by 0x8067766: TcpConnection::onReceive(pbuf*) (TcpConnection.cpp:110)
==29482==    by 0x806838A: TcpClient::onReceive(pbuf*) (TcpClient.cpp:118)
==29482==    by 0x8067DEE: TcpConnection::internalOnReceive(pbuf*, signed char) (TcpConnection.cpp:572)
==29482==  Address 0x4436e46 is 6 bytes inside a block of size 12 free'd
==29482==    at 0x402D7B8: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==29482==    by 0x806A46C: removeElementAt (WVector.h:371)
==29482==    by 0x806A46C: remove (WVector.h:338)
==29482==    by 0x806A46C: extractAt (ObjectMap.h:305)
==29482==    by 0x806A46C: HttpClientConnection::multipartProducer() (HttpClientConnection.cpp:94)
==29482==    by 0x8069725: MethodCaller<HttpPartResult (HttpClientConnection::*)()>::invoke() (Delegate.h:83)
==29482==    by 0x806F777: operator() (Delegate.h:172)
==29482==    by 0x806F777: MultipartStream::getNextStream() (MultipartStream.h:50)
==29482==    by 0x806FD2E: MultiStream::readMemoryBlock(char*, int) (MultiStream.cpp:29)
==29482==    by 0x80751C3: StreamTransformer::readMemoryBlock(char*, int) (StreamTransformer.cpp:36)
==29482==    by 0x8067A8C: TcpConnection::write(IDataSourceStream*) (TcpConnection.cpp:264)
==29482==    by 0x80681D9: TcpClient::pushAsyncPart() (TcpClient.cpp:157)
==29482==    by 0x806A3A2: HttpClientConnection::onReadyToSendData(TcpConnectionEvent) (HttpClientConnection.cpp:295)
==29482==    by 0x8067766: TcpConnection::onReceive(pbuf*) (TcpConnection.cpp:110)
==29482==    by 0x806838A: TcpClient::onReceive(pbuf*) (TcpClient.cpp:118)
==29482==    by 0x8067DEE: TcpConnection::internalOnReceive(pbuf*, signed char) (TcpConnection.cpp:572)
==29482==  Block was alloc'd at
==29482==    at 0x402C6BC: operator new(unsigned int) (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==29482==    by 0x8063796: ObjectMap<String, IDataSourceStream>::set(String const&, IDataSourceStream*) (ObjectMap.h:215)
==29482==    by 0x806403B: operator= (ObjectMap.h:82)
==29482==    by 0x806403B: setFile (HttpRequest.h:125)
==29482==    by 0x806403B: connectOk(IPAddress, IPAddress, IPAddress) (application.cpp:176)
==29482==    by 0x8062B96: FunctionCaller<void (*)(IPAddress, IPAddress, IPAddress), void, IPAddress, IPAddress, IPAddress>::invoke(IPAddress, IPAddress, IPAddress) (Delegate.h:110)
==29482==    by 0x80659F1: operator() (Delegate.h:172)
==29482==    by 0x80659F1: WifiEventsClass::WifiEventHandler(_esp_event*) (WifiEvents.cpp:60)
==29482==    by 0x8065B66: operator() (WifiEvents.cpp:22)
==29482==    by 0x8065B66: WifiEventsClass::WifiEventsClass()::{lambda(_esp_event*)#1}::_FUN(_esp_event*) (WifiEvents.cpp:22)
[... 130 lines cut ...]
==29482==    by 0x80681D9: TcpClient::pushAsyncPart() (TcpClient.cpp:157)
==29482==    by 0x806A3A2: HttpClientConnection::onReadyToSendData(TcpConnectionEvent) (HttpClientConnection.cpp:295)
==29482==    by 0x8067766: TcpConnection::onReceive(pbuf*) (TcpConnection.cpp:110)
==29482==    by 0x806838A: TcpClient::onReceive(pbuf*) (TcpClient.cpp:118)
==29482==    by 0x8067DEE: TcpConnection::internalOnReceive(pbuf*, signed char) (TcpConnection.cpp:572)
==29482==  Block was alloc'd at
==29482==    at 0x402C17C: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==29482==    by 0x402E370: realloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==29482==    by 0x8064967: String::changeBuffer(unsigned int) (WString.cpp:177)
==29482==    by 0x80649A6: String::reserve(unsigned int) (WString.cpp:155)
==29482==    by 0x8064A43: String::copy(char const*, unsigned int) (WString.cpp:193)
==29482==    by 0x8064C47: String::operator=(String const&) (WString.cpp:239)
==29482==    by 0x80637A2: Entry (ObjectMap.h:326)
==29482==    by 0x80637A2: ObjectMap<String, IDataSourceStream>::set(String const&, IDataSourceStream*) (ObjectMap.h:215)
==29482==    by 0x806403B: operator= (ObjectMap.h:82)
==29482==    by 0x806403B: setFile (HttpRequest.h:125)
==29482==    by 0x806403B: connectOk(IPAddress, IPAddress, IPAddress) (application.cpp:176)
==29482==    by 0x8062B96: FunctionCaller<void (*)(IPAddress, IPAddress, IPAddress), void, IPAddress, IPAddress, IPAddress>::invoke(IPAddress, IPAddress, IPAddress) (Delegate.h:110)
==29482==    by 0x80659F1: operator() (Delegate.h:172)
==29482==    by 0x80659F1: WifiEventsClass::WifiEventHandler(_esp_event*) (WifiEvents.cpp:60)
==29482==    by 0x8065B66: operator() (WifiEvents.cpp:22)
==29482==    by 0x8065B66: WifiEventsClass::WifiEventsClass()::{lambda(_esp_event*)#1}::_FUN(_esp_event*) (WifiEvents.cpp:22)
==29482==    by 0x804B8C8: wifi_status_callback (esp_wifi.c:60)
==29482== 
```

slaff, 2019-05-31:
@mikee47 Any idea what might cause `valgrind` to act strangely?

mikee47, 2019-05-31:
I don't have ValGrind installed ATM, having issues with Linux VM so can't check. Could you try with a simpler example? e.g. HostTests.

slaff, 2019-05-31:
> Could you try with a simpler example? e.g. HostTests.

This one seems to work nicely. And small issues reported. But at least no segfault(s). So valgrind is doing fine. Most probably the issue comes when valgrind replaces our str* functions and mem* functions with its own.

```
==2198== Syscall param write(buf) points to uninitialised byte(s)
==2198==    at 0x407D00F: ??? (syscall-template.S:84)
==2198==    by 0x804A223: writeFlashFile(unsigned int, void const*, unsigned int) (flashmem.cpp:92)
==2198==    by 0x804A40B: flashmem_write (flashmem.cpp:133)
==2198==    by 0x8067B72: api_spiffs_write (spiffs_sming.c:29)
==2198==    by 0x8069F3F: spiffs_phys_wr (spiffs_cache.c:221)
==2198==    by 0x806C98F: spiffs_object_create (spiffs_nucleus.c:959)
==2198==    by 0x8068510: SPIFFS_open (spiffs_hydrogen.c:255)
==2198==    by 0x80665F4: fileOpen(String const&, FileOpenFlags) (FileSystem.cpp:24)
==2198==    by 0x8066695: fileSetContent(String const&, char const*, int) (FileSystem.cpp:121)
==2198==    by 0x806673B: fileSetContent(String const&, String const&) (FileSystem.cpp:116)
==2198==    by 0x8063F6E: test_json() (test-json.cpp:88)
==2198==    by 0x805D2AA: init() (application.cpp:12)
==2198==  Address 0xfeb76ca3 is on thread 1's stack
==2198==  in frame #5, created by spiffs_object_create (spiffs_nucleus.c:918)
==2198==  Uninitialised value was created by a stack allocation
==2198==    at 0x806C859: spiffs_object_create (spiffs_nucleus.c:918)
```

closed 2019-05-31 by slaff