SmingHub/Sming #1720: Host architecture - BAD PIN

frankdownunder, 2019-06-11:
In /opt/Sming/Sming/Arch/Host/Core/Digital.cpp, we have

static inline bool checkPin(uint16_t pin)
{
	if(pin < 16) {
		return true;
	} else {
		hostmsg("BAD PIN %u", pin);
		return false;
	}
}

But a Wemos D1 Mini has pin D0 defined as GPIO 16
https://www.google.com/search?q=wemos+d1+mini+pinout&client=ubuntu&hs=q8D&channel=fs&tbm=isch&source=iu&ictx=1&fir=HI8V5x7XNyIhnM%253A%252CSnlcq4QotyYk3M%252C_&vet=1&usg=AI4_-kQGWGLdSbahE5I7UK8Rp8jy684mOQ&sa=X&ved=2ahUKEwjIh8zlzOLiAhXSvp4KHbCTBwwQ9QEwAHoECAUQBA#imgrc=HI8V5x7XNyIhnM:

Should it instead be
	if(pin <= 16)

mikee47, 2019-06-12:
Comparing with `Arch/Esp8266/Core/Digital.cpp`, yes you are correct.

slaff, 2019-06-12:
@frankdownunder Do you want to create a PR which fixes the issue? I will take care to review it and merge it.

frankdownunder, 2019-06-18:
OK I will

closed 2019-06-23 by slaff