SmingHub/Sming #1725: base64 decoding problem padding bytes

verybadsoldier, 2019-06-20:
I have a problem regarding the base64 API. This base64 encoded string:
`YWRtaW46dGVzdA==`

gets decoded by this code:
`userPass = base64_decode(userPass)`

to this
`admin:test␏�`

So there are one or two extra bytes appended (depending on encoding).

When I remove the padding bytes manually before decoding:
`YWRtaW46dGVzdA`
then I get
`admin:test`
(which is what I want)

Is it me failing to understand base64 or is this a bug?

mikee47, 2019-06-20:
This code:

```
	String s = base64_decode("YWRtaW46dGVzdA==");
	m_printHex("base64_decode", s.c_str(), s.length());
```

Produces this:

```
base64_decode: 61 64 6d 69 6e 3a 74 65 73 74                    admin:test
```

As expected. If you could post the problem code snippet that would help.

verybadsoldier, 2019-06-20:
Sorry, but that one line basically was the code snippet actually. But I tested again using your code:
```
    String s = base64_decode("YWRtaW46dGVzdA==");
    m_printHex("base64_decode", s.c_str(), s.length());
```
and I get this:
```
base64_decode: 61 64 6d 69 6e 3a 74 65 73 74 0f be              admin:test..
```

But sorry, I should have mentioned that I am on 3.7.0.1 (still fighting with API changes of 3.8.0). But I could not find anything about base64 mentioned in the bugtracker.

verybadsoldier, 2019-06-20:
Well, I managed to port my stuff to Sming 3.8.0 but the problem still persists. I don't understand why it is happening for me but not for you. Code seems not very complex.

mikee47, 2019-06-20:
@verybadsoldier OK, so I'm getting the same result as you running code on the Esp8266; the result I posted above was running on the emulator. Clearly something strange is going on so needs further investigation...

mikee47, 2019-06-21:
I've traced the problem to signed/unsigned chars. The base64 code assumes that `char` is signed, but it isn't (https://stackoverflow.com/questions/2054939/is-char-signed-or-unsigned-by-default). Perhaps the safest fix is to add `-fsigned-char` to compiler flags as there may be other code which makes this presumption:

```
diff --git a/Sming/Makefile b/Sming/Makefile
index 602b617e..d17a3e3e 100644
--- a/Sming/Makefile
+++ b/Sming/Makefile
@@ -255 +255 @@ MFORCE32 := $(shell $(CC) --help=target | grep mforce-l32)
-CFLAGS_COMMON = -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections
+CFLAGS_COMMON = -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -fsigned-char
```

verybadsoldier, 2019-06-21:
Nice catch!

PR #1726 (2019-06-22): Use explicit signed types in base64_decode
Code breaks if compiler doesn't default to signed chars, therefore explicit types must be used. In this case, the `int8_t` type is more appropriate than char. Fixes #1725.
closed 2019-06-22 by verybadsoldier