SmingHub/Sming #1746: Migrating to 3.9

verybadsoldier, 2019-07-06:
Thanks to @mikee47  for porting the esp_rgbww_firmeware (https://github.com/verybadsoldier/esp_rgbww_firmware) to the current Sming version. I am in the process of getting into it myself and would like to document here the issues I have encountered. Maybe things can be improved for a smoother transition but maybe it's also just based on my stupidity.

After I set `SDK_BASE` correctly it did compile. When issuing `make flash` I got this:
```
vbs@ubuntu:~/Projects/esp_rgbww_firmware$ make flash

esp_rgbww_firmware: Invoking 'flash' for Esp8266 (debug) architecture

Building /home/vbs/Projects/esp_rgbww_firmware/out/Esp8266/debug/lib/clib-App-cbd0ae68dc06b8f1ac2abd0dd8821d24.a
make[1]: Nothing to be done for 'build'.
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/component.mk:43: ! SSL support is not enabled. To enable it type: 'make clean; make ENABLE_SSL=1'
esp_rgbww_firmware: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8624|    1572
    rodata|           ReadOnly Data (RAM)|    3FFE8630|    3FFEA3C4|    7572
       bss|      Uninitialized Data (RAM)|    3FFEA3C8|    3FFF18F8|   30000
      text|            Cached Code (IRAM)|    40100000|    401075EB|   30187
irom0_text|           Uncached Code (SPI)|    40202010|    4026C710|  435968
Total Used RAM : 39144
Free RAM : 42776
Free IRam : 2599
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
Killing Terminal to free /dev/ttyUSB0
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Components/terminal/component.mk:29: recipe for target 'kill_term' failed
make: [kill_term] Killed (ignored)
WriteFlash  0x002000=out/Esp8266/debug/firmware/rom0.bin 0x100000=out/Esp8266/debug/firmware/spiff_rom.bin
/home/vbs/Projects/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 400000 write_flash -ff 40m -fm qio -fs 4MB  0x002000 out/Esp8266/debug/firmware/rom0.bin 0x100000 out/Esp8266/debug/firmware/spiff_rom.bin
usage: esptool write_flash [-h] [--flash_freq {40m,26m,20m,80m}]
                           [--flash_mode {qio,qout,dio,dout}]
                           [--flash_size {detect,4m,2m,8m,16m,32m,16m-c1,32m-c1,32m-c2}]
                           [--no-progress] [--verify]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument --flash_size/-fs: invalid choice: '4mb' (choose from 'detect', '4m', '2m', '8m', '16m', '32m', '16m-c1', '32m-c1', '32m-c2')
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Arch/Esp8266/app.mk:100: recipe for target 'flash' failed
make: *** [flash] Error 2
```

So it seems the `esptool.py` doesn't like the parameter `-fs 4MB` and prefers `-fs 4m`.

So in `Sming/Sming/Arch/Esp8266/Components/esptool/component.mk` I put this:
```
ifeq ($(SPI_SIZE), 256K)
	flashimageoptions	+= -fs 256K
else ifeq ($(SPI_SIZE), 1M)
	flashimageoptions	+= -fs 1M
	INIT_BIN_ADDR		:= 0x0fc000
	BLANK_BIN_ADDR		:= 0x0fe000
else ifeq ($(SPI_SIZE), 2M)
	flashimageoptions	+= -fs 2M
	INIT_BIN_ADDR		:= 0x1fc000
	BLANK_BIN_ADDR		:= 0x1fe000
else ifeq ($(SPI_SIZE), 4M)
	flashimageoptions	+= -fs 4M
	INIT_BIN_ADDR		:= 0x3fc000
	BLANK_BIN_ADDR		:= 0x3fe000
else
	flashimageoptions	+= -fs 512K
	INIT_BIN_ADDR		:= 0x07c000
	BLANK_BIN_ADDR		:= 0x04b000
endif
```
Basically removing the "B"s.

Then it did flash
```
vbs@ubuntu:~/Projects/esp_rgbww_firmware$ COM_PORT_ESPTOOL=/dev/ttyUSB0 make flash

esp_rgbww_firmware: Invoking 'flash' for Esp8266 (debug) architecture

Building /home/vbs/Projects/esp_rgbww_firmware/out/Esp8266/debug/lib/clib-App-cbd0ae68dc06b8f1ac2abd0dd8821d24.a
make[1]: Nothing to be done for 'build'.
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/component.mk:43: ! SSL support is not enabled. To enable it type: 'make clean; make ENABLE_SSL=1'
esp_rgbww_firmware: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8624|    1572
    rodata|           ReadOnly Data (RAM)|    3FFE8630|    3FFEA3C4|    7572
       bss|      Uninitialized Data (RAM)|    3FFEA3C8|    3FFF18F8|   30000
      text|            Cached Code (IRAM)|    40100000|    401075EB|   30187
irom0_text|           Uncached Code (SPI)|    40202010|    4026C710|  435968
Total Used RAM : 39144
Free RAM : 42776
Free IRam : 2599
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
Killing Terminal to free /dev/ttyUSB0
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Components/terminal/component.mk:29: recipe for target 'kill_term' failed
make: [kill_term] Killed (ignored)
WriteFlash  0x002000=out/Esp8266/debug/firmware/rom0.bin 0x100000=out/Esp8266/debug/firmware/spiff_rom.bin
/home/vbs/Projects/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 500000 write_flash -ff 40m -fm qio -fs 4M  0x002000 out/Esp8266/debug/firmware/rom0.bin 0x100000 out/Esp8266/debug/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Writing 479232 @ 0x2000... 479232 (100 %)
Wrote 479232 bytes at 0x2000 in 10.1 seconds (379.1 kbit/s)...
WARNING: Unlikely to work as data goes beyond end of flash. Hint: Use --flash_size
Writing 786432 @ 0x100000... 786432 (100 %)
Wrote 786432 bytes at 0x100000 in 16.7 seconds (377.8 kbit/s)...
Leaving...
```

Note the message 
`WARNING: Unlikely to work as data goes beyond end of flash. Hint: Use --flash_size`

Surprisingly it did still boot. Stay tuned...

mikee47, 2019-07-06:
Thanks for the update. That's an old version of esptool, maybe we should just include the current one in the repo.

mikee47, 2019-07-06:
Try building with `SPI_SIZE=4M`

verybadsoldier, 2019-07-06:
> Thanks for the update. That's an old version of esptool, maybe we should just include the current one in the repo.

Hm, but Sming explictly uses the esptool from the ESP SDK here `$(ESP_HOME)/esptool/esptool.py`, no? And I think I am using the latest ESP SDK. So why is my esptool others than yours?
```
vbs@ubuntu:~/Projects/esp-open-sdk$ git remote -v
origin	git@github.com:pfalcon/esp-open-sdk.git (fetch)
origin	git@github.com:pfalcon/esp-open-sdk.git (push)

commit c70543e57fb18e5be0315aa217bca27d0e26d23d (HEAD -> master, origin/master, origin/HEAD)
Author: Max Filippov <jcmvbkbc@gmail.com>
Date:   Wed Nov 21 01:16:43 2018 -0800

    Makefile: Add explicit rule for building libc.a
    
    Introduce intermediate file goal crosstool-NG/.built that does actual
    toolchain build to make sure that multiple 'ct-ng build' commands are
    not run in parallel. This allows to use make -jN to build esp-open-sdk.
    
    Signed-off-by: Max Filippov <jcmvbkbc@gmail.com>
```
> Try building with SPI_SIZE=4M

That is already set in the `component.mk`:
```
# SPI EEPROM SIZE
SPI_SIZE = 4M
```

Currently I have problems with the webserver not replying JSON data. Maybe something regarding the migration to the JSON v6 library. The JSON string seems just empty. Still investigating...

mikee47, 2019-07-06:
I run Linux in a VM so find it more convenience to use Windows for flashing. Tried it in Linux and same result as you. Fix on its way.

slaff, 2019-07-06:
Fixed with the merge of PR #1748

closed 2019-07-06 by slaff