SmingHub/Sming #1749: Host compile fails on Ubuntu 18 with preprocessor error

verybadsoldier, 2019-07-06:
I tried to get the host compile running on my Ubuntu 18 VM. I could not find documentation regarding host compile but as far as I could find out I just have to specify `SMING_ARCH=HOST`?

Trying to compile a simple sample I get this:
```
vbs@ubuntu:~/Projects/Sming/samples/Websocket_Client$ V=1 make SMING_ARCH=Host

Websocket_Client: Invoking 'all' for Host (debug) architecture
-- Parsing App
-- Parsing Sming
-- Parsing sming-arch
-- Parsing driver
-- Parsing esp_hal
-- Parsing esp_wifi
-- Parsing gdbstub
-- Parsing heap
-- Parsing hostlib
-- Parsing libc
-- Parsing lwip
-- Parsing spi_flash
-- Parsing vflash
-- Parsing spiffs
-- Parsing http-parser
-- Parsing libb64
-- Parsing ws_parser
-- Parsing mqtt-codec
-- Parsing libyuarel
-- Parsing ArduinoJson
-- Parsing terminal

Building /home/vbs/Projects/Sming/Sming/out/Host/Linux/debug/lib/clib-driver.a
make -r -R --no-print-directory -C /home/vbs/Projects/Sming/Sming/out/Host/Linux/debug/build/driver -f /home/vbs/Projects/Sming/Sming/component-wrapper.mk COMPONENT_NAME=driver COMPONENT_PATH=/home/vbs/Projects/Sming/Sming/Arch/Host/Components/driver COMPONENT_LIBDIR=/home/vbs/Projects/Sming/Sming/out/Host/Linux/debug/lib COMPONENT_LIBNAME=driver COMPONENT_LIBHASH=  build
g++ -I/home/vbs/Projects/Sming/Sming/Components -I/home/vbs/Projects/Sming/Sming/System/include -I/home/vbs/Projects/Sming/Sming/Wiring -I/home/vbs/Projects/Sming/Sming/Core -I/home/vbs/Projects/Sming/Sming -I/home/vbs/Projects/Sming/Sming/Arch/Host -I/home/vbs/Projects/Sming/Sming/Arch/Host/Core -I/home/vbs/Projects/Sming/Sming/Arch/Host/System/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/driver/include -I/home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver -I/home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include -I/home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/esp8266/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_wifi/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/gdbstub/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/heap/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/hostlib/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/libc/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/lwip -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/lwip/lwip/src/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/lwip/lwip/contrib/ports/unix/port/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/spi_flash/include -I/home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/spi_flash/include -I/home/vbs/Projects/Sming/Sming/Arch/Host/Components/vflash/include -I/home/vbs/Projects/Sming/Sming/Components/spiffs -I/home/vbs/Projects/Sming/Sming/Components/spiffs/spiffs/src -I/home/vbs/Projects/Sming/Sming/Components/http-parser/include -I/home/vbs/Projects/Sming/Sming/Components/libb64/include -I/home/vbs/Projects/Sming/Sming/Components/ws_parser/include -I/home/vbs/Projects/Sming/Sming/Components/mqtt-codec/include -I/home/vbs/Projects/Sming/Sming/Components/libyuarel/include -I/home/vbs/Projects/Sming/Sming/Libraries/ArduinoJson/include -I/home/vbs/Projects/Sming/Sming/Components/terminal/include -Wall -Wundef -Wpointer-arith -Wno-comment -Wl,-EL -finline-functions -fdata-sections -ffunction-sections -DARDUINO=106 -Werror -Wno-parentheses -Wno-unused-variable -Wno-unused-but-set-variable -Os -g -DARCH_HOST -m32 -Wno-deprecated-declarations -DPROJECT_DIR=\"/home/vbs/Projects/Sming/samples/Websocket_Client\" -DSMING_HOME=\"/home/vbs/Projects/Sming/Sming\" -DENABLE_SSL=1 -DENABLE_CMD_EXECUTOR=1 -DDEBUG_PRINT_FILENAME_AND_LINE=0 -DCUST_FILE_BASE= -DDEBUG_VERBOSE_LEVEL=2 -Werror -std=c++11 -felide-constructors  -MM -MT Arch/Esp8266/Components/driver/SerialBuffer.o /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp -MF Arch/Esp8266/Components/driver/SerialBuffer.cpp.d
In file included from /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed/limits.h:194:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed/limits.h:34,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/esp_systemapi.h:19,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.h:15,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp:13:
/usr/include/limits.h:145:17: error: missing binary operator before token "("
 #if __GLIBC_USE (IEC_60559_BFP_EXT)
                 ^
In file included from /usr/lib/gcc/x86_64-linux-gnu/7/include/stdint.h:9:0,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/c_types.h:3,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/esp_systemapi.h:20,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.h:15,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp:13:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
 #if __GLIBC_USE (IEC_60559_BFP_EXT)
                 ^
In file included from /home/vbs/Projects/Sming/Sming/System/include/m_printf.h:12:0,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/esp_systemapi.h:42,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.h:15,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp:13:
/usr/include/string.h:36:42: error: missing binary operator before token "("
 #if defined __cplusplus && __GNUC_PREREQ (4, 4)
                                          ^
/usr/include/string.h:164:21: error: missing binary operator before token "("
      || __GLIBC_USE (LIB_EXT2))
                     ^
/usr/include/string.h:173:43: error: missing binary operator before token "("
 #if defined __USE_XOPEN2K8 || __GLIBC_USE (LIB_EXT2)
                                           ^
/usr/include/string.h:491:19: error: missing binary operator before token "("
 #if __GNUC_PREREQ (3,4)
                   ^
/home/vbs/Projects/Sming/Sming/component-wrapper.mk:152: recipe for target 'Arch/Esp8266/Components/driver/SerialBuffer.cpp.d' failed
make[1]: *** [Arch/Esp8266/Components/driver/SerialBuffer.cpp.d] Error 1
/home/vbs/Projects/Sming/Sming//project.mk:375: recipe for target 'driver-build' failed
make: *** [driver-build] Error 2
```

Seems like quite simple preprocessor errors but I frankly have no clue currently what is causing it. Maybe I am doing something wrong in general regarding host compile?

mikee47, 2019-07-06:
This particular sample uses SSL, which isn't yet supported on Host (work in progress). Doesn't look like that's causing this particular problem though, could you try something simpler like `Basic_Blink`?

mikee47, 2019-07-06:
https://github.com/SmingHub/Sming/wiki/Host-Emulator

verybadsoldier, 2019-07-07:
Happens also for `Basic_Blink`:

```
vbs@ubuntu:~/Projects/Sming/samples/Basic_Blink$ SMING_ARCH=Host make

Basic_Blink: Invoking 'all' for Host (debug) architecture

Building /home/vbs/Projects/Sming/Sming/out/Host/Linux/debug/lib/clib-driver.a
In file included from /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed/limits.h:194:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed/limits.h:34,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/esp_systemapi.h:19,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.h:15,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp:13:
/usr/include/limits.h:145:17: error: missing binary operator before token "("
 #if __GLIBC_USE (IEC_60559_BFP_EXT)
                 ^
In file included from /usr/lib/gcc/x86_64-linux-gnu/7/include/stdint.h:9:0,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/c_types.h:3,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/esp_systemapi.h:20,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.h:15,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp:13:
/usr/include/stdint.h:286:17: error: missing binary operator before token "("
 #if __GLIBC_USE (IEC_60559_BFP_EXT)
                 ^
In file included from /home/vbs/Projects/Sming/Sming/System/include/m_printf.h:12:0,
                 from /home/vbs/Projects/Sming/Sming/Arch/Host/Components/esp_hal/include/esp_systemapi.h:42,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.h:15,
                 from /home/vbs/Projects/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp:13:
/usr/include/string.h:36:42: error: missing binary operator before token "("
 #if defined __cplusplus && __GNUC_PREREQ (4, 4)
                                          ^
/usr/include/string.h:164:21: error: missing binary operator before token "("
      || __GLIBC_USE (LIB_EXT2))
                     ^
/usr/include/string.h:173:43: error: missing binary operator before token "("
 #if defined __USE_XOPEN2K8 || __GLIBC_USE (LIB_EXT2)
                                           ^
/usr/include/string.h:491:19: error: missing binary operator before token "("
 #if __GNUC_PREREQ (3,4)
                   ^
/home/vbs/Projects/Sming/Sming/component-wrapper.mk:152: recipe for target 'Arch/Esp8266/Components/driver/SerialBuffer.cpp.d' failed
make[1]: *** [Arch/Esp8266/Components/driver/SerialBuffer.cpp.d] Error 1
/home/vbs/Projects/Sming/Sming//project.mk:375: recipe for target 'driver-build' failed
make: *** [driver-build] Error 2
```

Looks like something is screwing with the preprocessor but no good idea yet... So you guys are on Windows with MinGW and that works?

mikee47, 2019-07-07:
Host compiles as 32-bit application so on 64-bit systems you'll need additional libraries which should fix your issues. Further details are here https://github.com/SmingHub/Sming/wiki/Host-Emulator.

slaff, 2019-07-07:
See also: https://www.katacoda.com/slaff/scenarios/sming-host-emulator

verybadsoldier, 2019-07-07:
Thanks, it works after `sudo apt-get install gcc-multilib g++-multilib`.

Sorry, I didn't read it well enough I guess.

closed 2019-07-07 by verybadsoldier