SmingHub/Sming #1753: HttpServer_AJAX empty JSON responses

verybadsoldier, 2019-07-07:
Since the webserver in my own application only responds with empty pages since I updated to latest Sming I took a look into `HttpServer_AJAX` to see if it is working there.

As far as I understand it the app registers e.g. the URL `http://<IP>/ajax/input`. From the code I think it should deliver some sort of result body when issuing a `GET` request (for example in a browser).

But I get only 0 bytes back:
```
HTTP/1.1 200 OK
Content-Type: application/json
Content-Length: 0
Connection: keep-alive
Server: HttpServer/Sming
```

I debugged it a bit and I think it is related to the JsonDocument. Before sending the answer even the JsonDocument returns a `size` of 0 so I think there is something wrong.

Tested on latest `develop`.

mikee47, 2019-07-07:
Definitely a bug! Try this patch:

```
diff --git a/Sming/Libraries/ArduinoJson6/include/JsonObjectStream.h b/Sming/Libraries/ArduinoJson6/include/JsonObjectStream.h
index d3411c3d2..f8c7d72b8 100644
--- a/Sming/Libraries/ArduinoJson6/include/JsonObjectStream.h
+++ b/Sming/Libraries/ArduinoJson6/include/JsonObjectStream.h
@@ -28,6 +28,7 @@ public:
     */
 	JsonObjectStream(Json::SerializationFormat format, size_t capacity = 1024) : doc(capacity), format(format)
 	{
+		doc.to<JsonObject>();
 	}
 
 	/** @brief Create a JSON object stream using default (Compact) format
```

verybadsoldier, 2019-07-07:
Thanks, that fixed it! `GET` requests seem ok now. But in `POST` requests the payload doesn't make it into the callback, always empty String... Investigating...

PR #1756 (2019-07-08): Bugfix: `JsonObjectStream.h` (version 6), returns empty document
Constructor needs to call `to<JsonObject>` to initialise the JsonDocument ready for writing, otherwise all writes will fail. Fixes #1753.
commit fcf44cf5 (2019-07-08): bugfix: `JsonObjectStream.h` (version 6), returns empty document (#1756)

Fixes #1753
closed 2019-07-08 by slaff