SmingHub/Sming #1755: Basic_WebSkeletonApp JSON POST body always empty

verybadsoldier, 2019-07-07:
When testing `Basic_WebSkeletonApp` and sending a `POST` request with a JSON body to URL of the app:

`http://<IP>/config`

like this:
```
POST /config HTTP/1.1
Host: 192.168.2.28
Content-Type: application/json
Cache-Control: no-cache
Postman-Token: 5d4c1d47-1b7c-0d0d-1cab-b887b39f531d

{
    "StaSSID": "PleaseEnterSSID",
    "StaPassword": "PleaseEnterPass",
    "StaEnable": 1
}
```

Then the body data does not make it into the URI handler callback. The output of the app is:
```
147026296 The headers are complete
147028443 Update config
147029159 NULL bodyBuf
147031444 Storing 17 bytes in stream
147034753 realloc 17 -> 165
147037041 Storing 19 bytes in stream
147040354 Storing 24 bytes in stream
147043650 Storing 26 bytes in stream
147046934 Storing 2 bytes in stream
147050165 TCP onReadyToSendData: 1
147053436 TCP Written: 88, Available: 88, isFinished: 1, PushCount: 1 [TcpBuf: 2692]
147060722 TcpClient stream finished
147064028 TCP received: 582 bytes
147067058 TCP onReadyToSendData: 1
147117072 TCP sent: 88
147117211 TCP onReadyToSendData: 2
```

Not the line `147029159 NULL bodyBuf` which is indicating that the callback didn't receive the data:
```
void onConfiguration(HttpRequest& request, HttpResponse& response)
{
	if(request.method == HTTP_POST) {
		debugf("Update config");
		// Update config
		if(request.getBody() == nullptr) {
			debugf("NULL bodyBuf");
			return;
		} else {
```

mikee47, 2019-07-07:
Got this running in emulator.

* `HttpServerConnection.cpp` line 107. `HttpServerConnection::onHeadersComplete()`.
Request doesn't have `HTTP_HEADER_CONTENT_TYPE` set, so `bodyParser` doesn't get set.

* `HttpServerConnection::onBody()`. `HttpServerConnection.cpp` line 142.
No `bodyParser`, and no `onBody` handler. Body gets discarded.

So either the requestor is wrong because they've missed out the content type, or the server is wrong because it should assume a default. I'm not sure which is appropriate?

verybadsoldier, 2019-07-07:
I think the bodyParser got set in the code to accept `application/json`:

```
void startWebServer()
{
	if(serverStarted)
		return;

	server.listen(80);
	server.paths.set("/", onIndex);
	server.paths.set("/config", onConfiguration);
	server.paths.set("/config.json", onConfiguration_json);
	server.paths.set("/state", onAJAXGetState);
	server.paths.setDefault(onFile);
	server.setBodyParser("application/json", bodyToStringParser);
```

mikee47, 2019-07-07:
Found the problem, `Content-Type` is getting interpreted as a custom HTTP header type. Now looking for a fix.

mikee47, 2019-07-07:
@verybadsoldier Thanks for these bug reports. Keep 'em coming!

PR #1757 (2019-07-08): Bugfix: `LOAD_FSTR()` doesn't NUL-terminate buffered string
Bug introduced in #1709. When a `FlashString` object is loaded into RAM, buffered contents are not NUL-terminated so produces random and unpredictable results, such as failed comparisons. This only happens where the string length is a multiple 4 characters, such as "Content-Type" (12).
Fixes #1755.
commit b54f1137 (2019-07-08): Bugfix: `LOAD_FSTR()` doesn't NUL-terminate buffered string (#1757)

Bug introduced in #1709. Fixes #1755.
closed 2019-07-08 by slaff