SmingHub/Sming #1764: Controller crash loop after flashinit due to missing Wifi config (probably)

verybadsoldier, 2019-07-13:
With the RGBWW firmware I am experiencing a problem that after using `make flashinit` the controller hangs in an infinite crash loop. The problem seems reproducible for me.

What happened:
* I have the RGBWW firmware running on the controller (Wiif config is present)
* I do `make flashinit`
* I do `make flash` to again flash the same firmware as before
* Crash loop

I think the key is the Wifi configuration. If it is not present then the problem appears. I can fix the problem by flashing an old firmware and setting up Wifi there. Then I can again flash the current firmware and it will run stable.

So begins with flashin the current firmware while a Wifi config is present:
```
vbs@ubuntu:~/Projects/esp_rgbww_firmware$ make flash

esp_rgbww_firmware: Invoking 'flash' for Esp8266 (debug) architecture

Building /home/vbs/Projects/esp_rgbww_firmware/out/Esp8266/debug/lib/clib-App-07df350943f744e11e1aada7269d20e0.a
C+ /home/vbs/Projects/esp_rgbww_firmware/app/networking.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/webserver.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/eventserver.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/otaupdate.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/mqtt.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/stepsync.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/ledctrl.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/application.cpp
C+ /home/vbs/Projects/esp_rgbww_firmware/app/jsonprocessor.cpp
AR /home/vbs/Projects/esp_rgbww_firmware/out/Esp8266/debug/lib/clib-App-07df350943f744e11e1aada7269d20e0.a
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/component.mk:42: ! SSL support is not enabled. To enable it type: 'make clean; make ENABLE_SSL=1'
esp_rgbww_firmware: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8604|    1540
    rodata|           ReadOnly Data (RAM)|    3FFE8610|    3FFEA384|    7540
       bss|      Uninitialized Data (RAM)|    3FFEA388|    3FFF18A8|   29984
      text|            Cached Code (IRAM)|    40100000|    40107593|   30099
irom0_text|           Uncached Code (SPI)|    40202010|    4026C984|  436596
Total Used RAM : 39064
Free RAM : 42856
Free IRam : 2687
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
Killing Terminal to free /dev/ttyUSB0
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Components/terminal/component.mk:29: recipe for target 'kill_term' failed
make: [kill_term] Killed (ignored)
WriteFlash 0x00000=out/Esp8266/debug/firmware/rboot.bin 0x002000=out/Esp8266/debug/firmware/rom0.bin 0x100000=out/Esp8266/debug/firmware/spiff_rom.bin
/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Arch/Esp8266/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 -b 500000 write_flash -ff 40m -fm qio -fs 4MB 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x002000 out/Esp8266/debug/firmware/rom0.bin 0x100000 out/Esp8266/debug/firmware/spiff_rom.bin
esptool.py v2.7-dev
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:0d:8e:a7:84:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 500000
Changed.
Configuring flash size...
Compressed 2336 bytes to 1695...
Wrote 2336 bytes (1695 compressed) at 0x00000000 in 0.0 seconds (effective 387.8 kbit/s)...
Hash of data verified.
Compressed 475840 bytes to 311908...
Wrote 475840 bytes (311908 compressed) at 0x00002000 in 6.9 seconds (effective 548.2 kbit/s)...
Hash of data verified.
Compressed 786432 bytes to 244416...
Wrote 786432 bytes (244416 compressed) at 0x00100000 in 5.2 seconds (effective 1198.9 kbit/s)...
Hash of data verified.

Leaving...
[... 343 lines cut ...]
0x40211dc8: wpa_sm_disassociate at ??:?
0x40211dc8: wpa_sm_disassociate at ??:?
0x402566a0: String::operator=(String const&) at /home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Wiring/WString.cpp:750
0x4023bc94: operator new(unsigned int) at /home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Arch/Esp8266/Components/esp8266/esp_cplusplus.cpp:53
0x40241ddc: std::_Function_handler<void (bool, Vector<BssInfo>), Delegate<void (bool, Vector<BssInfo>)>::Delegate<AppWIFI>(void (AppWIFI::*)(bool, Vector<BssInfo>), AppWIFI*)::{lambda(bool, Vector<BssInfo>)#1}>::_M_invoke(std::_Any_data const&, bool, Vector<BssInfo>) at /home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Libraries/ArduinoJson6/ArduinoJson/src/ArduinoJson/Numbers/FloatParts.hpp:55
0x4025deb1: std::function<void (bool, Vector<BssInfo>)>::operator()(bool, Vector<BssInfo>) const at /home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Arch/Esp8266/Platform/Station.cpp:328
0x40211dc8: wpa_sm_disassociate at ??:?
0x40211dc8: wpa_sm_disassociate at ??:?
0x4025df4d: StationClass::staticScanCompleted(void*, STATUS) at /home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Arch/Esp8266/Platform/Station.cpp:328
0x40211dc8: wpa_sm_disassociate at ??:?
0x40211dc8: wpa_sm_disassociate at ??:?
0x40237b50: chip_v6_unset_chanfreq at ??:?
0x4025ded0: StationClass::staticScanCompleted(void*, STATUS) at /home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/Arch/Esp8266/Platform/Station.cpp:328
0x40208dfa: scan_cancel at ??:?
0x40208af5: scan_start at ??:?
0x40209065: scan_remove_probe_ssid at ??:?
0x4022ff4f: pp_tx_idle_timeout at ??:?
0x4021388d: ets_timer_handler_isr at ??:?
0x402138d2: ets_timer_handler_isr at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
```

It does not happend with `3.8.0` but with current develop.

verybadsoldier, 2019-07-14:
I tested a bit more to pin-point this and it really happens as soon as I delete the Wifi config in a running controller.

So I have a controller running nicely and then I call
```
void AppWIFI::forgetWifi() {
    debug_i("AppWIFI::forget_wifi");
    WifiStation.config("", "");
    WifiStation.disconnect();
    _client_status = CONNECTION_STATUS::IDLE;
}
```
then I get into this crash loop.

mikee47, 2019-07-14:
I've come across this one before but seems a little inconsistent. Running under the debugger shows corruption of one of the BssInfo values.

mikee47, 2019-08-01:
@verybadsoldier Fixed in #1791, loss of config triggers an AP scan which crashes when it tries to sort the list.

slaff, 2019-08-02:
Feel free to re-open if the issue is still present in the latest `develop` version.

closed 2019-08-02 by slaff