SmingHub/Sming #1765: Crash after OTA update from 3.8.0 to current develop

verybadsoldier, 2019-07-14:
Well, OTA updates from a 3.8.0-fw to the current version fails. The update works but when the controller boots with the new firmware then it instantly crashes.

I can reproduce like this:
* Running a 3.8.0-FW
* Starting OTA-update to a new FW

The OTA-update from new-FW to another new-FW works fine.

The problem looks like this (log is quite large so I split it up on pastebin and give comments what happened):

Part 1- 3.8.0-FW is booting:
https://pastebin.com/pNkRUqcv

Part 2 - OTA update is started and updates fine:
https://pastebin.com/aGrbLE7k

Part 3 - Then the new firmware boots but crashes:
```

state: 5 -> 0 (0)
rm 0
pm close 7
58418809 TCP connection error: -8
del if0
usl
58419532 disconnect from ssid hsync, reason 8

58419783 AppWIFI::_STADisconnect reason - 8 - counter 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0xb7
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0xe9
csum 0xe9

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting temp rom.
Booting rom 1.
Fatal exception (28): 
epc1=0x402137bc, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x00000300, depc=0x00000000
Fatal exception (28): 
epc1=0x402137bc, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x00000300, depc=0x00000000
Fatal exception (28): 
epc1=0x402137bc, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x00000300, depc=0x00000000
Fatal exception (28): 
epc1=0x402137bc, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x00000300, depc=0x00000000
Fatal exception (28): 
epc1=0x402137bc, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x00000300, depc=0x00000000
Fatal exception (28): 
epc1=0x402137bc, epc2=0x00000000, epc3=0x40000f68, excvaddr=0x00000300, depc=0x00000000
Fatal exception (28): 
```

When doing the same but starting already from current-FW then the OTA works and the controller runs correctly:
https://pastebin.com/uhbpUYA4

slaff, 2019-07-16:
Did you create both firmwares using the same SDK version?

verybadsoldier, 2019-07-16:
I am always using the SDK bundled with Sming. So yes, afaik it didn't change in a while.

verybadsoldier, 2019-10-16:
So, I am back to this as I am upgrading the RGBWW project to Sming `4.0.0`. And I am again encountering this error.

My latest stable firmware was based on Sming `3.8.0` which I think uses SDK `3.0.0`:
https://github.com/SmingHub/Sming/blob/3.8.0/Sming/third-party/ESP8266_NONOS_SDK/include/version.h

Now current Sming uses SDK `3.0.1`.

So is this error expected and is there a solution? I am afraid not being able to OTA-update this is a showstopper for me :(

mikee47, 2019-10-16:
The dump tells you where it's crashing and, if it's the same code as above (28 LOAD_PROHIBITED) then the bad code is at 0x402137bc in your new image. The easiest way to diagnose that is to build it for GDB - assuming it's getting as far as init() - then you'll get a nice backtrace.

verybadsoldier, 2019-10-18:
Ok, I tried that (well, I tried alot actually) but also the version compiled with `ENABLE_GDB` also crashes in the same way. If I understand correctly than that version was supposed to wait for a `gdb` to connect before starting the firmware. But it seems to crash for that wait happens.

But, I tried alot of other things to isolate this problem:
It seems that this problem only happens with the sming4-firmware being in slot 1 while having the sming3-firmware being in slot 0. If I replace the slot 0 firmware then also with the sming4-firmware then suddenly the sming4-firmware in slot 1 boots fine.
For me this is quite surprising but maybe you have an idea to push me in the right direction?

I will describe in detail one sequence of flashing/testing:

1. `make flashinit`
2. Flash rboot: 
`esptool.py -p /dev/ttyUSB0 -b 500000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/rboot.bin`
3. Verify the expectations: `rboot` in place but no valid ROM in any slot:
```
vbs@ubuntu:~/Projects/esp_rgbww_firmware$ python2 -m serial.tools.miniterm /dev/ttyUSB0 38400
--- Miniterm on /dev/ttyUSB0  38400,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0xb7
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0xe9
csum 0xe9

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done
```
4. Flash the sming4-firmware to slot *1

verybadsoldier, 2019-10-18:
mistakenly closed...

verybadsoldier, 2019-10-18:
Sorry for confusion: I just wanted to write a comment but accidently closed the issue. Finished my comment now...

verybadsoldier, 2019-10-18:
The exact same thing can be reproduced with e.g. `Basic_Blink`. Just doing `flashinit` and then flashing the `rom0.bin` to `0x202000`.

verybadsoldier, 2019-10-20:
@mikee47 
Does anything I said any sense to you? :)

mikee47, 2019-10-20:
@verybadsoldier Thank you for taking the time to break down this problem, I found it fairly quickly using `Basic_Blink`. To verify:

```
make flashinit flashboot
```

produces

```
Rom 0 at 2000 is bad.
Rom 1 at 202000 is bad.
No good rom available.
user code done
```

I then do:

```
esptool.py -p COM4 -b 921600 write_flash -ff 40m -fm dio -fs 4MB 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x202000 out/Esp8266/debug/firmware/rom0.bin
```

and get normal operation:

```
Booting rom 1 at 202000, load addr 230990.
rf cal sector: 1019
```

verybadsoldier, 2019-10-20:
Hm sorry, not sure if I understand it: so the problem did not occur on your end?

mikee47, 2019-10-20:
Yes, I got the same problem. Try this patch, see if it fixes it for you:

```
diff --git a/Sming/Components/rboot/component.mk b/Sming/Components/rboot/component.mk
index d96533cff..c31350118 100644
--- a/Sming/Components/rboot/component.mk
+++ b/Sming/Components/rboot/component.mk
@@ -117 +117 @@ CUSTOM_TARGETS			+= $(LIBMAIN_RBOOT_DST)
-COMPONENT_LDFLAGS		:= -u Cache_Read_Enable_New
+EXTRA_LDFLAGS			:= -u Cache_Read_Enable_New
```

verybadsoldier, 2019-10-21:
Fixed by #1900

PR #1900 (2019-10-21): Fix bug where ROM2 fails to boot
rBoot makefile should be using `EXTRA_LDFLAGS`, not `COMPONENT_LDFLAGS` (which doesn't exist). Fixes #1765 .

The `Cache_Read_Enable_New()` library function is replaced to set the memory mapped address region,
so as this didn't happen it attempted to execute invalid code from the normal address space at 0x2000 (instead of0x202000).

Also corrected in the esp8266 component, which affects the crash handler.

Bug introduced in #1744.
closed 2019-10-21 by verybadsoldier