SmingHub/Sming #1771: Broken COMPONENT_SRCDIRS in component.mk

lucas-bremond, 2019-07-21:
Take the `Basic_Blink` example, and uncomment this line:

```
COMPONENT_SRCDIRS := app
```

then build:

```
make
```

It will fail like this:

```
Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture

Building /opt/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-ef56d5c2b5d9b2d843ac14b0b0917be5.a
C+ /opt/Sming/samples/Basic_Blink/app/application.cpp
AR /opt/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-ef56d5c2b5d9b2d843ac14b0b0917be5.a
/opt/Sming/Sming/component.mk:42: ! SSL support is not enabled. To enable it type: 'make clean; make ENABLE_SSL=1'
make[1]: Entering directory '/opt/Sming/Sming/Arch/Esp8266/Components/rboot/rboot'
E2 /opt/Sming/samples/Basic_Blink/out/Esp8266/debug/firmware/rboot.bin
make[1]: Leaving directory '/opt/Sming/Sming/Arch/Esp8266/Components/rboot/rboot'
OC out/Esp8266/debug/lib/libmain-rboot.a
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
/opt/Sming/Sming/out/Esp8266/debug/lib/clib-esp8266-7f470de3585a2594a9dc3b91a9e15bdc.a(startup.o):(.text.user_init+0x0): undefined reference to `gdb_init'
/opt/Sming/Sming/out/Esp8266/debug/lib/clib-esp8266-7f470de3585a2594a9dc3b91a9e15bdc.a(startup.o):(.text.user_init+0x15): undefined reference to `gdb_init'
collect2: error: ld returned 1 exit status
/opt/Sming/Sming/Arch/Esp8266/app.mk:34: recipe for target 'out/Esp8266/debug/build/app_0.out' failed
make: *** [out/Esp8266/debug/build/app_0.out] Error 1
```

Seems that there is a path conflict/error somewhere in the build system, which make it incompatible with `COMPONENT_SRCDIRS`.

mikee47, 2019-07-21:
@lucas-bremond Thanks for reporting this. Found the problem, fix on its way.

slaff, 2019-07-22:
@lucas-bremond the problem should be fixed in the latest develop version. Feel free to reopen the issue if it still persists. And make sure to provide additional details.

closed 2019-07-22 by slaff