SmingHub/Sming #1773: Fix make flash terminal

lucas-bremond, 2019-07-21:
After successfully flashing, using:

```
make flash
```

a new terminal opens, according to:

```
flash: all kill_term ##Write the rBoot boot sector, application image and (if enabled) SPIFFS image
	$(call WriteFlash,$(FLASH_RBOOT_BOOT_CHUNKS) $(FLASH_RBOOT_APP_CHUNKS) $(FLASH_SPIFFS_CHUNKS))
ifeq ($(ENABLE_GDB), 1)
	$(GDB_CMDLINE)
else
	$(TERMINAL)
endif
```

https://github.com/SmingHub/Sming/blob/efc4d8bb9fbea782e164489067b0c2740451a910/Sming/Arch/Esp8266/app.mk#L94

`TERMINAL` being defined here:

```
TERMINAL		?= python -m serial.tools.miniterm $(COM_PORT) $(COM_SPEED_SERIAL) $(COM_OPTS)
```

https://github.com/SmingHub/Sming/blob/fdee78c26299bec99df65f873d95882d37668b24/Sming/Components/terminal/component.mk#L18

Problem is: if the chip return non-ASCII chars, the miniterm console will crash (with the default settings)

```
...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/dist-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/usr/lib/python2.7/dist-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u240c' in position 0: ordinal not in range(128)
```

Adding the following args to the command fixes it:

```
python -m serial.tools.miniterm --raw --encoding ascii /dev/ttyUSB0 115200
```

but to my knowledge, there is no current way to specify this in the Makefile (using `COM_OPTS` does not work, as it is set after `COM_PORT` and `COM_SPEED_SERIAL`).

Any idea?

mikee47, 2019-07-22:
I agree having `COM_OPTS` at the end is silly! I've got a PR to address that, but need to decide what the appropriate default setting is for it. Could be `--raw -encoding ascii` as you suggest, however I can't get it to crash with any settings, so wonder if it's a bug in pyserial somewhere? I've got `pip show pyserial` showing version 3.4...

mikee47, 2019-07-22:
OK, it crashes nicely (!) with --encoding UTF-16. I'll go with your recommendation for the default.

mikee47, 2019-07-22:
Incidentally, you could always override the entire `TERMINAL` variable in your project's `component.mk` file (or via `export`)...

slaff, 2019-08-02:
Fixed with the merge of PR #1777

closed 2019-08-02 by slaff