SmingHub/Sming #1774: UdpServer_Echo sample code: doesn't sent string

Oshio09, 2019-07-22:
When I execute the "UdpServer_Echo" example, I can successfully receive the packages that are coming from my computer and they are printed at the serial terminal.

Although the function `udp.sendStringTo()` returns `true`, I used **netcat**, **tcpdump**, **wireshark** and a custom software I made, none of those reported any udp package activity in the port I specified.

Can someone reproduce the behavior?

I'm using the develop branch.
commit number: 414bf05ba2d35e3dd8ad2a30fdbb4e909f554896

slaff, 2019-07-22:
@Oshio09 please, check PR #1775.

Oshio09, 2019-07-22:
@slaff yes its working.
But when using the makefile in the "UdpServer_Echo" folder what its been compiled its the blink example. This also happens with the original commit (414bf05).
I copied the application.cpp to a folder with the `Makefile` and `component.mk` files copied from the blink example for it to work.
Can you confirm if this is a problem or a mistake that I'm making?

closed 2019-07-22 by Oshio09

slaff, 2019-07-22:
> Can you confirm if this is a problem or a mistake that I'm making?

I haven't experienced such an issue. I usually compile examples by typing:

```
cd $SMING_HOME/../samples/UdpServer_Echo
make
```

And all is good. For this issue I actually used the Host emulator to test it on my local system. And I did this to make it run locally (on the Host)

```
# To setup the virtual network I use
$SMING_HOME/Arch/Host/Tools/setup-network-linux.sh wlan0 
# Then compile as usual
cd $SMING_HOME/../samples/UdpServer_Echo
# with SMING_ARCH=Host
make SMING_ARCH=Host
# and then test it using
make run
```
```

Oshio09, 2019-07-22:
@slaff  I followed your steps and everything works perfectly. Thank you for your assistance.
I didn't knew about the Host functionality, I believe it will be worth some mention to this feature at the `Readme.md` for newbies like me to discover about this in a easier way.

slaff, 2019-07-22:
@Oshio09 Check our documentation: https://sming.readthedocs.io/en/latest/

Oshio09, 2019-07-22:
Thanks, amazing work!

PR #1775 (2019-07-22): Fix UpdServer_Echo.
Fixes #1774