SmingHub/Sming #1778: HW Timer: Minimum Time

Oshio09, 2019-07-23:
I need to generate a very short pulse, ideally the on cycle should be ~500ns. 
After reading the API documentation I have some questions:

1) There is a way to set the hw timer interrupt interval to less than 1us? If not, this limitation is due NONOS SDK?
2) The API documents the macro `MIN_HW_TIMER_INTERVAL_US 0x32`, but its not used in Timer.cpp. Intuitively the name implies that an interval smaller than 50us isn't achievable, this is true?

barbiani, 2019-07-23:
Can you spare the cycles in a busy wait?

mikee47, 2019-07-23:
(2) `MIN_HW_TIMER_INTERVAL_US` is used in `HardwareTimer.cpp`, `Timer.cpp` concernes low-accuracy software timers. @barbiani has the right idea I think, 500ns is only 40 CPU cycles (or 80 if you bump the clock).

Oshio09, 2019-07-23:
> Can you spare the cycles in a busy wait?
> […](#)

Unfortunately not, I'm using the esp8266 to do a lot of adc reads that need to be transmitted in "real-time". 
Here is a resume of my project:
The pulse is used as the clock of a 4-bit counter ic, the 4 bit output is then used as the channel selection for a lot of multiplexers. This multiplexers are meant to connect the cells of a resistive matrix to the ADC input. In the firmware perspective the code needs to generate a pulse, perform an ADC read, do some trivial math to check the reading and then repeat the procedure until all cells are read, this data is sent through UDP protocol to some external device (computer/phone) and the cycle repeats until the external device send a stop message.
Now I'm using 700 cells as prototype, but the idea is to support as many cells as possible, and have at least 20 full matrix reading per second, because of this requirement I'm trying to make the code very efficient and a busy wait wouldn't be a good alternative. My circuit can handle one ADC read each ~2us, my goal is to make the ESP perform the cycle I described as close to 2us as possible.

Oshio09, 2019-07-23:
@barbiani Unfortunately not, I'm using the esp8266 to do a lot of adc reads that need to be transmitted in "real-time".
Here is a resume of my project:
The pulse is used as the clock of a 4-bit counter ic, the 4 bit output is then used as the channel selection for a lot of multiplexers. This multiplexers are meant to connect the cells of a resistive matrix to the ADC input. In the firmware perspective the code needs to generate a pulse, perform an ADC read, do some trivial math to check the reading and then repeat the procedure until all cells are read, this data is sent through UDP protocol to some external device (computer/phone) and the cycle repeats until the external device send a stop message.
Now I'm using 700 cells as prototype, but the idea is to support as many cells as possible, and have at least 20 full matrix reading per second, because of this requirement I'm trying to make the code very efficient and a busy wait wouldn't be a good alternative. My circuit can handle one ADC read each ~2us, my goal is to make the ESP perform the cycle I described as close to 2us as possible.

Oshio09, 2019-07-23:
@mikee47 Thanks for pointing this out my IDE intellisense pointed me to the wrong file.
Yes the macro is been used in the `HardwareTimer.cpp`, but do you see a motive why the minimum time couldn't not be 1us?

mikee47, 2019-07-24:
The hardware timer generates an interrupt, so introduces a load of code to save/restore context which adds significantly to the latency. This is why short intervals (< 100us) are a bad idea because the code will spend all its time saving context to the stack, executing your code, then restoring context.

I don't think you need to use the hardware timer at all, you just need the code to run as fast as possible with a minimum of wasted CPU cycles.

I'm going to assume that your counter IC increments on the rising edge of its CLK input, so you can drop it LOW at any time, then set it HIGH when you're ready to switch inputs, like this:

1. Set the MUX output HIGH to change the input.
2. Call `system_adc_read()`.
3. Set the MUX output LOW.
4. Perform your calculations. Provided it takes > 500ns to do this step, no additional delays necessary.

At 20 matrix reads/sec one iteration would take 50ms which is too long to hog the CPU for, so you'll need to break it up into chunks.

Oshio09, 2019-07-24:
Thanks for sharing your thoughts. I wasn't aware of this 100us interrupt overhead, I'll keep that in mind.

I like the simplicity of step 4, using the calculation's as a delay to generate the pulse seems a good idea, if its to fast I can introduce some dummy calculations to keep the CPU busy for a few cycles. 

Regarding the CPU hog, I was thinking to do 1 matrix read send over UDP then call the `WDTClass::alive ()` before repeating the cycle, do you think this will work?

closed 2019-07-24 by Oshio09

mikee47, 2019-07-24:
The interrupt overhead isn't peculiar to the Esp8266, it's same with any processor.

If I understand correctly, once matrix reading has started it continues indefinitely until explicitly stopped. However, you must release control fully back to the SDK in a timely manner. Bear in mind that the Esp8266 is just a single CPU, without any operating system, and also has to keep the WiFi connection active and serviced at a low level. It can only do this if you give control back to the SDK. Internally, the SDK has a main control loop which will look something like this:

	for(;;) {
		ServiceWifi();
		ServiceTimers();
		ServiceTasks();
		FeedWatchdog();
	}

When your code gets called in response to a network request, or a timer that you've set up, then you must deal with the request then return as promptly as possible to allow things to continue.

To get your code to be called as fast as possible, use the task queue. I'll illustrate with some code, and as we're in C++ land I'll use a class:

	class MatrixReader
	{
	public:
		void start()
		{
			if(!state.active) {
				// Do whatever setup you need
				state.position = 0;
				state.active = true;
				service();
			}
		}

		void stop()
		{
			state.active = false;
		}

	private:
		void service()
		{
			// Do some reading of the matrix
			...
	
			// If we want to continue reading, schedule a callback
			if(state.active) {
				System.queueCallback([](uint32_t Param) {
					serviceMatrix();
				});
			}
		}
	
	private:
		// Our state information
		bool active; ///< Set to true when reading is being performed
		unsigned position; ///< For example, keep track of read position
	};

And the setup code:

	static MatrixReader reader;
	
	// This might be your callback to handle a network request 
	void requestToStartReadingMatrix()
	{
		reader.start();
	}

This is just a rough-up of how I'd do this, don't be surprised if it doesn't compile!

icodk, 2019-07-24:
https://www.instructables.com/id/625-Nanosecond-Resolution-Timer-for-Any-Microcontr/
https://sub.nanona.fi/esp8266/timing-and-ticks.html

mikee47, 2019-07-24:
I got curious about loop times, so wrote a quick sample to check it out. On my nodeMCU this is what I'm getting:

`Elapsed count = 249267, total = 1000002, min = 0, max = 135, average = 4`

Times in microseconds. It's not doing anything else, a real example would deviate significantly under network load.

```
#include <SmingCore.h>
#include <Services/Profiling/ElapseTimer.h>

class MatrixReader
{
public:
	void start()
	{
		if(!active) {
			// Do whatever setup you need
			position = 0;
			times.reset();
			loopTimer.start();
			active = true;
			service();
		}
	}

	void stop()
	{
		active = false;
	}

private:
	void service()
	{
		auto elapsed = loopTimer.elapsed();

		++times.count;
		times.total += elapsed;
		times.min = std::min(times.min, elapsed);
		times.max = std::max(times.max, elapsed);
		loopTimer.start();

		if(times.total >= 1000000) {
			Serial.printf("Elapsed count = %u, total = %u, min = %u, max = %u, average = %u\r\n", times.count,
						  times.total, times.min, times.max, times.total / times.count);
			times.reset();
		}

		// Do some reading of the matrix
		// ...

		// If we want to continue reading, schedule a callback
		if(active) {
			System.queueCallback([](uint32_t param) { reinterpret_cast<MatrixReader*>(param)->service(); },
								 reinterpret_cast<uint32_t>(this));
		}
	}

private:
	// Our state information
	bool active = false;   ///< Set to true when reading is being performed
	unsigned position = 0; ///< For example, keep track of read position

	struct {
		unsigned count;
		uint32_t total;
		uint32_t min;
		uint32_t max;

		void reset()
		{
			count = 0;
			total = min = max = 0;
		}
	} times;
	ElapseTimer loopTimer;
};

static MatrixReader reader;

void init()
{
	Serial.begin(SERIAL_BAUD_RATE);
	reader.start();
}
```