SmingHub/Sming #1781: Proposal: (Signed) OTA firmware updates using HTTP POST file upload

aemseemann, 2019-07-26:
This proposal is intended to evaluate interest in an alternative OTA update mechanism for Sming and discuss potential realization aspects.

As far as I'm aware, Sming currently only provides the OTA update method demonstrated in samples/Basic_rBoot.
In this approach, the ESP downloads the firmware image from a server provided by the developer. The update is initiated using a secondary channel, e. g. by a command received from the serial console. While the second channel can be considered an additional security measure, it is sometimes inconvenient, especially if you do not have access to the serial console, e. g. for mains-powered devices like Sonoff switches.
 
As an alternative, I propose an update mechanism where the new firmware is pushed to the device via an HTTP POST file upload request. This is the type of request generated by a web form containing an `<input type="file"...>` control and thus would allow to easily integrate a convenient firmware upgrade page to an existing web interface. While this is probably the preferred method employed by end users, the same interface can be used during development by emulating the upload request from a Makefile target (e. g. using `curl -F ...`) to enable a completely wireless development workflow.

I implemented an initial draft [here](https://github.com/aemseemann/Sming/tree/feature/push-ota-signed)  (branch feature/push-ota-signed => [samples/Signed_OTA](https://github.com/aemseemann/Sming/tree/feature/push-ota-signed/samples/Signed_OTA)) and would really like to known your opinion in general and if it is worth pursuing an integration into Sming somehow.

**Some technical details:**

File uploads generate an HTTP request with content type `multipart/form-data`. Since Sming does not support this MIME type yet (I guess [this](https://github.com/SmingHub/Sming/issues/552) feature somehow didn't survive the HTTP code restructuring), I implemented a MultiPart parser ([MultipartResource.h](https://github.com/aemseemann/Sming/blob/feature/push-ota-signed/samples/Signed_OTA/app/MultipartResource.h)) and on top of it a form data parser ([FormDataRequest.h](https://github.com/aemseemann/Sming/blob/feature/push-ota-signed/samples/Signed_OTA/app/FormDataRequest.h)). While the layering approach seems viable and appropriate from a reusability perspective, there is a potential memory leak if the request is not completed successfully. This is because an instance of the actual Multipart/FormData parser is allocated in the `onHeaderComplete` callback of the HTTP request and its pointer is stored in the `HttpRequest::args` member. The deallocation is done in the `onMessageComplete` callback, which is not invoked in case of parser errors or if the connection is interrupted in between. Note that the built-in BodyParsers currently suffer from the same potential memory leak, because they use the same member to store heap-allocated per-request state. Fixing this issue would require some changes to Sming's HTTP server code.

To handle file uploads, the application must provide a writable stream that receives the file content. The actual firmware update is done in [FirmwareUpdateStream.h](https://github.com/aemseemann/Sming/blob/feature/push-ota-signed/samples/Signed_OTA/app/FirmwareUpdateStream.h)/[cpp](https://github.com/aemseemann/Sming/blob/feature/push-ota-signed/samples/Signed_OTA/app/FirmwareUpdateStream.cpp), which implements the stream interface and parses the uploaded ota file.

The firmware update file is generated by a small Python script (otagen.py) and can contain multiple Rom images, which are all uploaded at once. The `FirmwareUpdateStream` automatically selects the matching ROM for the next unused slot and ignores the other ROM(s) contained in the ota file. (From my experience, the transfer time overhead for additional ROMs is negligible, since the overall OTA update duration is dominated by flash memory write speed.)

The whole firmware file is secured by a signature. While the received data is always written to the flash immediately, the updated ROM slot is only activated after successful signature verification using the verification key embedded in the currently running firmware. If the verification fails, the header of the just updated slot is invalidated to prevent accidentally booting into an unsanctioned firmware, e. g. via rboot's GPIO_SKIP feature.
The signature algorithm is 'EdDSAph', as implemented by the crypt_sign_... family of methods from [libsodium](https://github.com/jedisct1/libsodium). I did a quick and dirty integration of libsodium as a component by hacking its configure script to cross-compile for the ESP8266. However, for a sustainable multi-platform solution, I would replace the autotools-based build process. After a quick check, it appears nothing fancy is done during the build, so it might be as easy as listing source directories and letting Sming's default component build process handle the rest.
Alternatively, if the license permits, one could extract the relevant functions out of libsodium and directly integrate them in the OTA update code. However, I would consider the full libsodium a valuable addition to Sming, regardless of the OTA functionality. It's always nice to have access to strong crypto in a network-centered framework.

**Limitations:**

The example currently only supports updating the "main" firmware, neither the bootloader nor the file system. 
If desired, adding support for OTA bootloader updates should be quite easy, as the bootloader is usually small enough to fit into RAM and thus could be received completely and verified before actually being written to flash memory.
For file system upgrades one has to distinguish between separate SPIFFS for each ROM slot or a single shared SPIFFS. In the first case, the update could be done in exactly the same way as for the firmware ROM. A shared SPIFFS is more difficult, because this usually implies that some settings or other content should be preserved throughout firmware updates. Maybe updating files one by one, each carrying its own signature, is a viable approach in this case.

So how do you feel about this? Any ideas or suggestions on how to proceed?

icodk, 2019-07-26:
If I understand your proposal, it is about two different aspects of OTA.  A signed OTA image is agreat  ide and I think it should be an option to be added to the existing  OTA system. I think it could be easily implemented by adding a secret key to an MD5 hash of the image to be sent. 
Regarding OTA invocation, you can build any invocation system you need on top of what already exist. In one of my projects, the device periodically contact a server  to find out if a new image is available. This of course has the theoretically disadvantage of a delay between the availability of a new image and the next communication cycle initiated by the device, which in my case has no practical implications. "pushing" anything to the device will require (in most cases?) port forwarding in the firewall with security implications I would like to avoid.
Am I missing something ?

slaff, 2019-07-27:
> I would consider the full libsodium a valuable addition to Sming, regardless of the OTA functionality. It's always nice to have access to strong crypto in a network-centered framework.

I like the idea. @aemseemann can you create a PR with libsodium as a component so that we can test it. From my humble experience libraries that are not written for embedded systems are rarely usable due to their memory requirements, but I would be more than happy if libsodium proves me wrong.

slaff, 2019-07-27:
Concerning Http Server with support for multipart: I had experimented with streamable multipart-parser a long ago. The code is still not finished but you can see its current state at: https://github.com/slaff/Sming/tree/feature/httpserver-multipart-bodyparser

aemseemann, 2019-07-27:
@slaff: sure, I'll prepare a PR for libsodium. Memory requirements should not be an issue. At least for the features I am familiar with, it operates on user-provided buffers and does no allocations on its own. 

@icodk: I agree that the signing aspect is independent of the update method and maybe `rBootItemOutputStream` could be extended accordingly and shared between both ota mechanisms. However, the MD5 idea seem disputable. I am no crypto expert, but from what I've heard MD5 is more or less broken. Also, 'salting' it with a secret key implies that the same key must be available on the device for verification purposes, maybe even contained (in plain text!) in the update image. Using an SSL encrypted transport would certainly avoid this issue, but then you probably have pinned or otherwise verified your server certificate already, which establishes the necessary level of trust and obviates the need for signing the update image.
Bottom line: When adding the complexity of an ota signing mechanism, just do it properly and use well-established state-of-the-art crypto instead of a home-grown solution.

Regarding the existing ota mechanism: I agree that it serves many use cases, but IMO not all of them. Basically, what I aim for is an equivalent to the 'make flash' command in a local purely wireless development setup. No port forwarding required, because the development computer and the device are on the same network (or at least bridged together).
In addition, the proposed update method allows to provide a privacy-friendly (no internet access required) OTA method for unexperienced users via a web configuration GUI, akin to the firmware update method used by many routers and other commercial devices.

slaff, 2019-08-01:
@aemseemann File upload support in the HTTP server is on the way. Try PR #1792. The code needs to be organized better but it should be working and relatively easy to understand and extend and/or improve.

slaff, 2019-08-03:
> can you create a PR with libsodium as a component so that we can test it.

@aemseemann You can use that code as inspiration: https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/libsodium. We are still waiting for your PR :-)

slaff, 2019-08-03:
PR #1795 is opening the way for signed OTA updates.

aemseemann, 2019-08-03:
@slaff: The [PR](https://github.com/SmingHub/Sming/pull/1797) is ready now. 
Sorry that I kept you waiting. There wasn't any time last week to work on the issue.

I've seen that file upload support is now on develop. Will have a look at https://github.com/SmingHub/Sming/pull/1795 tomorrow.

closed 2021-11-16 by slaff