SmingHub/Sming #1782: Ideas to improve system_adc_read_fast() sample rate.

Oshio09, 2019-07-27:
From the ideas from issue #1778 I did some test to see how much time the ESP8266 takes to do one ADC reading. I wonder if there is some way to decrease the time it takes to perform the reading and by consequence improve the sampling rate? 

I tried to see the source code from `system_adc_read_fast()` but it seems that is closed source.

I leave here the code I'm using and the output:

**CODE**
```
#include <SmingCore.h>

#define PIN_HIGH GPIO_REG_WRITE(GPIO_OUT_W1TS_ADDRESS, (1 << 4)) // D2 HIGH
#define PIN_LOW  GPIO_REG_WRITE(GPIO_OUT_W1TC_ADDRESS, (1 << 4)) // D2 LOW

void inline __measure()
{
	ets_intr_lock(); //close interrupt
	uint16 _adc_num = 1;
	uint8 _adc_clk_div = 8;
	uint16 *_adc_addr = new uint16_t[_adc_num](); // Allocate space and init all values with 0

	volatile int32_t t_before, t_after;

	PIN_HIGH;
	for(uint8_t i=0; i<10; i++)
	{
		asm volatile("rsr %0, ccount" : "=r"(t_before));
		system_adc_read_fast(_adc_addr, _adc_num, _adc_clk_div);
		asm volatile("rsr %0, ccount" : "=r"(t_after));

		Serial.printf("\n\nTime = %u \tt_after = %u \t t_before = %u\n\n", (uint32_t)(t_after - t_before), t_after, t_before);
		Serial.printf("ADC = %d\n", _adc_addr[0]);
	}
	ets_intr_unlock(); //open interrupt
}

void init()
{
	Serial.begin(460800);
	Serial.systemDebugOutput(true);
	Serial.printf("\n\nSming app skeleton ready\n");

	System.setCpuFrequency(CpuFrequency::eCF_160MHz);
	if (System.getCpuFrequency() == CpuFrequency::eCF_80MHz)
		Serial.printf("Running at 80MHz\n");
	else
		Serial.printf("Running at 160MHz\n");

	WifiStation.enable(false);
	pinMode(4,OUTPUT);
	System.onReady(__measure);
}

```
**OUTPUT**
Time is measured in clock cycles. CPU running at 160MHz.
```
Time = 4829     t_after = 5514330        t_before = 5509501
ADC = 629

Time = 2667     t_after = 5541868        t_before = 5539201
ADC = 646

Time = 2666     t_after = 5647999        t_before = 5645333
ADC = 643

Time = 2667     t_after = 5872974        t_before = 5870307
ADC = 644

Time = 2666     t_after = 6097769        t_before = 6095103
ADC = 645

Time = 2666     t_after = 6322645        t_before = 6319979
ADC = 645

Time = 2666     t_after = 6547589        t_before = 6544923
ADC = 645

Time = 2667     t_after = 6772586        t_before = 6769919
ADC = 644

Time = 2667     t_after = 6997472        t_before = 6994805
ADC = 644

Time = 2667     t_after = 7222346        t_before = 7219679
ADC = 644
```
**Circuit**
The circuit is a nodemcu that has its pin D2 (GPIO4) wired to a resistor in series with a led. The adc input is connected in the resistor-led node.

mikee47, 2019-07-27:
This is interesting as an academic exercise, but in practice if you have to disable WiFi and interrupts to sample like this anyway isn't that killing the whole point of the Esp8266 in the first place? So you'd end up having to use `system_adc_read()`. Also, if you're multiplexing the ADC input what about settling time?

If this doesn't work well enough then I'd probably look for a solution using an external ADC, perhaps as part of a co-processor.

Still, curiosity peaked so here's a program to exercise various test cases, including the regular system_adc_read().

Note that this isn't really a question for Sming, you may find better answers elsewhere, like Espressif's BBS?

Results below are on nodeMCU. Whacking the CPU speed up does make a little difference, but the biggest improvement is from batching the reads (AdcNum > 1).

Note: Nothing attached to the ADC input.

```
#0: SLOW 80MHz  AdcNum = 1      Iterations = 100        Cycles = 773262         Av. cycles = 7732       Elapsed = 9697  Average = 96
ADC = [5]
#0: SLOW 160MHz AdcNum = 1      Iterations = 100        Cycles = 1311991        Av. cycles = 13119      Elapsed = 8212  Average = 82
ADC = [6]
#1: SLOW 80MHz  AdcNum = 8      Iterations = 100        Cycles = 6105249        Av. cycles = 7631       Elapsed = 76339         Average = 95
ADC = [7, 7, 5, 8, 7, 6, 6, 7]
#1: SLOW 160MHz AdcNum = 8      Iterations = 100        Cycles = 10446291       Av. cycles = 13057      Elapsed = 65301         Average = 81
ADC = [7, 6, 6, 5, 6, 7, 6, 6]
#2: FAST 80MHz  AdcNum = 1      Iterations = 100        Cycles = 168952         Av. cycles = 1689       Elapsed = 2136  Average = 21
ADC = [5]
#2: FAST 160MHz AdcNum = 1      Iterations = 100        Cycles = 261915         Av. cycles = 2619       Elapsed = 1649  Average = 16
ADC = [5]
#3: FAST 80MHz  AdcNum = 8      Iterations = 100        Cycles = 472585         Av. cycles = 590        Elapsed = 5931  Average = 7
ADC = [5, 7, 5, 5, 8, 7, 9, 5]
#3: FAST 160MHz AdcNum = 8      Iterations = 100        Cycles = 776380         Av. cycles = 970        Elapsed = 4864  Average = 6
ADC = [7, 7, 8, 5, 8, 2, 5, 7]
#4: FAST 80MHz  AdcNum = 32     Iterations = 100        Cycles = 1521377        Av. cycles = 475        Elapsed = 19041         Average = 5
ADC = [7, 7, 4, 5, 0, 7, 5, 7, 8, 7, 5, 7, 8, 0, 5, 7, 7, 18, 7, 7, 7, 7, 7, 5, 7, 8, 7, 5, 8, 7, 7, 7]
#4: FAST 160MHz AdcNum = 32     Iterations = 100        Cycles = 2533444        Av. cycles = 791        Elapsed = 15846         Average = 4
ADC = [7, 12, 5, 3, 5, 8, 7, 7, 5, 5, 7, 7, 5, 5, 7, 5, 5, 7, 7, 5, 7, 7, 5, 5, 4, 7, 7, 5, 7, 7, 7, 5]
```

Here's the code:

```
#include <SmingCore.h>
#include <Services/Profiling/ElapseTimer.h>

SimpleTimer measurementTimer;

struct TestCase {
	bool fast;
	unsigned sampleCount;
	unsigned adcClockDiv;
	unsigned iterations;
};

const TestCase testCases[] = {
	{
		false,
		1,
		0,
		100,
	},
	{
		false,
		8,
		0,
		100,
	},
	{
		true,
		1,
		8,
		100,
	},
	{
		true,
		8,
		8,
		100,
	},
	{
		true,
		32,
		8,
		100,
	},
};

static unsigned testCaseIndex;

static uint16_t results[1024];

static void measure()
{
	auto& testCase = testCases[testCaseIndex];

	auto freq = System.getCpuFrequency();

	if(testCase.fast) {
		noInterrupts();
	}

	memset(results, 0, sizeof(results));

	uint32_t cycleCount = 0;

	ElapseTimer timer;

	for(unsigned i = 0; i < testCase.iterations; ++i) {
		volatile uint32_t before;
		asm volatile("rsr %0, ccount" : "=r"(before));

		if(testCase.fast) {
			system_adc_read_fast(results, testCase.sampleCount, testCase.adcClockDiv);
		} else {
			for(unsigned i = 0; i < testCase.sampleCount; ++i) {
				results[i] = system_adc_read();
			}
		}

		volatile uint32_t after;
		asm volatile("rsr %0, ccount" : "=r"(after));

		cycleCount += after - before;
	}

	auto elapsed = timer.elapsed();

	if(testCase.fast) {
		interrupts();
	}

	Serial.printf("#%u: %s %uMHz", testCaseIndex, testCase.fast ? "FAST" : "SLOW", freq);
	Serial.printf("\tAdcNum = %u \tIterations = %u", testCase.sampleCount, testCase.iterations);
	Serial.printf("\tCycles = %u \tAv. cycles = %u", cycleCount,
				  cycleCount / (testCase.iterations * testCase.sampleCount));
	Serial.printf("\tElapsed = %u \tAverage = %u\r\n", elapsed, elapsed / (testCase.iterations * testCase.sampleCount));

	Serial.print("ADC = [");
	for(unsigned i = 0; i < testCase.sampleCount; ++i) {
		if(i != 0) {
			Serial.print(", ");
		}
		Serial.print(results[i]);
	}
	Serial.println(']');

	unsigned delay = 100;
	if(freq == eCF_80MHz) {
		System.setCpuFrequency(eCF_160MHz);
	} else {
		System.setCpuFrequency(eCF_80MHz);
		++testCaseIndex;
		if(testCaseIndex >= ARRAY_SIZE(testCases)) {
			testCaseIndex = 0;
			delay = 5000; // Wait 5s then repeat
		}
	}

	measurementTimer.setCallback(SimpleTimerCallback(measure));
	measurementTimer.startMs(delay, false);
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true);

	WifiStation.enable(false);
	System.onReady(measure);
}
```

Oshio09, 2019-07-27:
Just to clarify, disabling WiFi and interrupts was a recommendation of the datasheet, this us just a test case, the real program checks if WIFI is enabled, disable WIFI and interrupts and enable them again after the ADC reading.
But I got curious and I ran the same test without disabling the features and got, basically, the same results. 
I knew that this isn't a proper topic to Sming, but I got a really good feedback lately and my previous experience with Espressif's BBS isn't a pleasant one. Any way, I glad for you sharing your thoughts...closing this issue.

closed 2019-07-27 by Oshio09

mikee47, 2019-07-27:
I'm guessing the noise will increase with interrupts and WiFi left enabled, but maybe there are other side effects. I'd be very interested to hear if you uncover anything further with this, be good to add it to the Sming documentation.

mikee47, 2019-07-27:
Found this FAQ https://www.espressif.com/en/support/explore/faq.

Salient points:

## What are the specifications for the internal ADC of ESP8266?

Channels: 1
Sampling rate:
* 100k samples/sec (Wi-Fi modem turned off)
* 1k samples/sec (Wi-Fi modem normally active)
* Resolution: 12-bit (10-bit read by system_adc_read())

## How accurate is the internal ADC?

When connected to routers, ESP8266 will enter modem_sleep in STA mode only, which will cause the changes of the internal chip currents and reference values, and thus result in abnormal ADC sampling.

If a high accuracy is required, please use system_adc_fast_read API. But RF circuitry should be shut down before measuring and Wi-Fi will be disconnected. For an relatively lower accuracy when readings’ difference of 1 or 2 can be tolerated, Wi-Fi can be configured to non_sleep mode.

For lower accuracy, the user may enter sleep mode. Power consumption is lower in this case.

## What should I use the internal ADC for?

The internal ADC can be used for temperature sensing or sensing approximate current drawn by external devices. Note that because the ADC readings are prone to noise, it should only be used for applications where high accuracy is not required. For example, thermal cut-off mechanisms, etc.