SmingHub/Sming #1784: Integrating VSCode debugger and Sming.

Oshio09, 2019-07-29:
Someone has successfully integrated VSCode visual debugger and Sming?

Oshio09, 2019-07-30:
After some research on the topic and I didn't find a definite answer to that. My goal is to add this information to Sming documentation.

slaff, 2019-07-30:
If `VSCode visual debugger` is compatible with GDB then all you need is to read that page: https://sming.readthedocs.io/en/latest/information/debugging.html

mikee47, 2019-07-30:
It's [in the docs](https://sming.readthedocs.io/en/latest/tools/vscode.html), I think?

slaff, 2019-07-30:
> If VSCode visual debugger is compatible with GDB then

I made a quick search and it seems that VSCode can work also with GDB debuggers. Take a look at that video for some hints: https://www.youtube.com/watch?v=agBOQ3SI4R0 .

Oshio09, 2019-07-30:
I appreciate the help, but I've already looked at those links. @mikee47  I've already looked this link although is a good start point, is a little outdated. The intelliSense is a little trickier, I have opened an issue in the VSCode c++ plugin extension ([link](https://github.com/microsoft/vscode-cpptools/issues/3969#issuecomment-514744709)), on of the maintainers gave me a little hack a tagged as a bug.
The debugging information in the Sming's docs is clear, the problem is how to integrate this on VSCode, they use a lot of json syntax in the configuration files that I'm not used to, I'm sure that shouldn't be a complicated stuff.

Oshio09, 2019-07-30:
@slaff  thanks! I will see the video tomorrow and give you an update, where I'm right now is 3:30 AM.

Oshio09, 2019-07-30:
I did more reading and I think the problem might be the fact I'm indicating the wrong file to the xtensa-gdb. Any of you could tell me which file the gdb should be reading?

I clear all previous building files including the Sming's `out` folder, then ran: 
`make ENABLE_GDB=1 SMING_ARCH=Esp8266`

`out` folder tree sructure:
```
.
├── build-type.mk
└── Esp8266
    └── debug
        ├── build
        │   ├── App
        │   │   └── App-54addfb1d462fc16256687ca4983c7cb
        │   │       ├── app
        │   │       │   ├── application.cpp.d
        │   │       │   ├── application.o
        │   │       │   ├── m_udp.cpp.d
        │   │       │   └── m_udp.o
        │   │       ├── Arch
        │   │       │   └── Esp8266
        │   │       │       └── Components
        │   │       │           ├── gdbstub
        │   │       │           │   ├── appcode
        │   │       │           │   │   ├── gdb_hooks.cpp.d
        │   │       │           │   │   ├── gdb_hooks.o
        │   │       │           │   │   └── gdbstub-entry.o
        │   │       │           │   ├── gdbhostio.cpp.d
        │   │       │           │   ├── gdbhostio.o
        │   │       │           │   ├── GdbPacket.cpp.d
        │   │       │           │   ├── GdbPacket.o
        │   │       │           │   ├── gdbstub.cpp.d
        │   │       │           │   ├── gdbstub.o
        │   │       │           │   ├── gdbsyscall.cpp.d
        │   │       │           │   ├── gdbsyscall.o
        │   │       │           │   ├── gdbuart.cpp.d
        │   │       │           │   └── gdbuart.o
        │   │       │           └── rboot
        │   │       │               ├── appcode
        │   │       │               │   ├── rboot-overrides.c.d
        │   │       │               │   └── rboot-overrides.o
        │   │       │               └── rboot
        │   │       │                   └── appcode
        │   │       │                       ├── rboot-api.c.d
        │   │       │                       ├── rboot-api.o
        │   │       │                       ├── rboot-bigflash.c.d
        │   │       │                       └── rboot-bigflash.o
        │   │       └── build.cfg
        │   ├── app_0.map
        │   ├── app_0.out
        │   └── bootrom.elf
        ├── config.mk
        ├── firmware
        │   ├── fwMeminfo.new
        │   ├── rboot.bin
        │   ├── rom0.bin
        │   ├── rom0.bin.cfg
        │   └── spiff_rom.bin
        └── lib
            ├── clib-App-54addfb1d462fc16256687ca4983c7cb.a
            └── libmain-rboot.a

17 directories, 36 files
```

Oshio09, 2019-07-30:
Just to make this information clear, the debuger is working when using the terminal, is just the integration with VSCode that doesn't work.

And here is the error I 'm getting: 
```
ERROR: Unable to start debugging. GDB exited unexpectedly.
The program '/home/user/Codes/UDP/out/Esp8266/debug/build/app_0.out' has exited with code 42 (0x0000002a).
```

mikee47, 2019-07-31:
That is the correct file. You can verify what the command line should be with ``make gdb V=1``.

Oshio09, 2019-08-06:
Hello guys, some updates.

I opened an [issue](https://github.com/microsoft/vscode-cpptools/issues/4011) at the cpp extension for vscode. It seems that I'm not the first to fail in this task. 

Recently the user **ngrplus**  claimed to have solved the problem(https://github.com/microsoft/vscode-cpptools/issues/1447#issuecomment-517906636), his solution was for arduino-esp8266 framework, after some tests I failed to accomplish meaningful results.

 I'd appreciated I one of you could gave some advice based on this new information.

mikee47, 2019-08-06:
This works for me:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Esp8266 GDB",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/out/Esp8266/debug/build/app_0.out",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": true,
            "MIMode": "gdb",
            "miDebuggerPath": "${env:ESP_HOME}/xtensa-lx106-elf/bin/xtensa-lx106-elf-gdb.exe",
            "miDebuggerArgs": "-x ${env:SMING_HOME}/Arch/Esp8266/Components/gdbstub/gdbcmds -b 115200",
            "miDebuggerServerAddress": "COM4",
        }
    ]
}```

mikee47, 2019-08-07:
I've not spent long with vscode, but my impression so far is that there are a lot of manual steps required to set up:

1. Download/install vscode
2. Install required extensions
3. Creating a new project
4. Adding all the required souce code paths (for all available architectures)
5. Setting up debugger configuration

Most of this can be automated. The [Quick Start](https://github.com/zgoda/sming-qstart/blob/master/sming-qstart.py) script is a good start, but I feel this sort of thing should be better integrated into Sming, specifically by adding make targets to:

* Create a new project. e.g. `cd $SMING_HOME; make create-project <path to project>`
* Setup/update a new vscode project, e.g. `make vscode-setup`. This can create the initial configuration files, or update source code paths, etc. as required.
* Setup or update a new eclipse project, e.g. `make eclipse-setup`

These would probably use python scripts to do the actual work. Documentation can be very minimal as we can just refer to the script for details of how to perform the steps manually, if required.

closed 2021-11-16 by slaff