SmingHub/Sming #1803: Python support in Sming

mikee47, 2019-08-04:
The way I see it we have 3 choices:

1. Continue to support both Python 2.7 and Python 3.4+
2. Deprecate Python 2 and require python 3
3. Don't bother with python 3

Thoughts?

P.S. https://python3statement.org/

slaff, 2019-08-05:
I would go for "Deprecate Python 2 and require python 3". After a year python 2 is reaching its end of life ( "Python 2.7 has been moved five years into the future, to 2020."  https://legacy.python.org/dev/peps/pep-0373/). It would probably take another 5-10 years until it disappears from all major OS distributions.

That said we must assure that the python code works with python 3.4+ and 2.7 compatibility will be nice to have, but not required.

verybadsoldier, 2019-08-09:
+1
It's time for Py2 to go for good.

Derkades, 2019-09-24:
Why would maintaining compatibility with Python 2 be important? You can run python3 and python2 side by side if you have applications that require python2 on the same system.

mikee47, 2024-04-04:
Python3 use appears stable, closing this issue as resolved.

closed 2024-04-04 by mikee47