SmingHub/Sming #1816: Build doesn't find header file

kmihaylov, 2019-08-11:
Hello. Just updated to the latest develop. After make-distclean and make some headers could not be found. I'm not aware how the build system works. May you check?

Thanks!

```Building /opt/Sming/Sming/out/Esp8266/debug/lib/clib-libyuarel.a
make[1]: Nothing to be done for 'build'.

Building /opt/Sming/Sming/out/Esp8266/debug/lib/clib-Sming-ccc08e87d72c17c6c82f1e2405aa0d08.a
In file included from /opt/Sming/Sming/Core/FileSystem.cpp:11:0:
/opt/Sming/Sming/Core/FileSystem.h:18:26: fatal error: spiffs_sming.h: No such file or directory
 #include "spiffs_sming.h"
                          ^
compilation terminated.
/opt/Sming/Sming/component-wrapper.mk:155: recipe for target 'Core/FileSystem.cpp.d' failed
make[1]: *** [Core/FileSystem.cpp.d] Error 1
/opt/Sming/Sming/project.mk:357```

mikee47, 2019-08-11:
@kmihaylov What version have you updated from?

mikee47, 2019-08-11:
@kmihaylov There's probably an empty directory `Arch/Esp8266/Components/spiffs` which you'll need to delete.

kmihaylov, 2019-08-11:
@mikee47 
Wow thanks! It worked. I had to remove also rboot/rboot from Components. Now I see some deprecated functions. A lot of new stuff!
Thank you!

closed 2019-08-11 by kmihaylov

mikee47, 2019-08-11:
Have a read of https://sming.readthedocs.io/en/latest/upgrading/3.8-4.0.html

mikee47, 2019-08-11:
Also https://sming.readthedocs.io/en/latest/_inc/Sming/building.html - explanation of your problem under 'known issues'. Should really put that somewhere more obvious :-(

kmihaylov, 2019-08-11:
Also I had to install ```sphinx-common```, Ubuntu 18.04.
These links would be helpful on the Sming's github front page. I checked another link (sming-framework.readthedocs.io), obviously I was mistaken.

mikee47, 2019-08-11:
I guess once this makes it out of develop the front page will get updated. What's `sphinx-common`?

kmihaylov, 2019-08-11:
A package needed to build Sming, I believe it is related to the new documentation system. sphinx-build (or something similar) was missing.

mikee47, 2019-08-11:
OK, that shouldn't be necessary unless you need to build the docs. locally - the online docs. now track the current develop build. If that was preventing you I'll need to find a fix.

kmihaylov, 2019-08-11:
I just did the following:
```
apt remove sphinx-common
apt autoremove
git clone [SMING_URL] /opt/Sming-test
export SMING_HOME=/opt/Sming-test
cd $SMING_HOME/Sming
make dist-clean
```

Here is the output:
```
Cleaning all submodules...
Cleaning all samples...
make[1]: Entering directory '/opt/Sming-test/docs'
../samples/HttpServer_FirmwareUpload/component.mk:11: warning: overriding recipe for target 'web-pack'
../samples/HttpServer_ConfigNetwork/component.mk:5: warning: ignoring old recipe for target 'web-pack'
../samples/HttpServer_FirmwareUpload/component.mk:14: warning: overriding recipe for target 'web-upload'
../samples/HttpServer_ConfigNetwork/component.mk:9: warning: ignoring old recipe for target 'web-upload'
/bin/bash: sphinx-build: command not found
Makefile:209: recipe for target 'clean' failed
make[1]: *** [clean] Error 127
make[1]: Leaving directory '/opt/Sming-test/docs'
Makefile:54: recipe for target 'docs-clean' failed
make: *** [docs-clean] Error 2
```

kmihaylov, 2019-08-11:
My fault, should be ```export SMING_HOME=/opt/Sming-test/Sming```.
Framework build is OK, dist-clean still fails.

kmihaylov, 2019-08-11:
Maybe rBootHttpUpdate -> RBootHttpUpdater can be added to 'Deprated list'?

mikee47, 2019-08-11:
> Maybe rBootHttpUpdate -> RBootHttpUpdater can be added to 'Deprated list'?

Yes, got some fixes to the docs, will include that.