SmingHub/Sming #1839: Add library

teicors, 2019-09-19:
Hi,
I would add a library in order to manage the DST for my timezone and I found a library https://github.com/JChristensen/Timezone that requires https://github.com/PaulStoffregen/Time.
I installed (using git clone) both in  /opt/sming/Sming/Libraries, I added to components.mk in my dir project 
`ARDUINO_LIBRARIES := ArduinoJson5 Time Timezone`.
I did a 
`make dist-clean`
But during the make I receive the following error:
```
AR /opt/sming/Sming/out/Esp8266/debug/lib/clib-LiquidCrystal.a

Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-Time.a
C+ /opt/sming/Sming/Libraries/Time/Time.cpp
C+ /opt/sming/Sming/Libraries/Time/DateStrings.cpp
AR /opt/sming/Sming/out/Esp8266/debug/lib/clib-Time.a

Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-Timezone.a
C+ /opt/sming/Sming/Libraries/Timezone/src/Timezone.cpp
In file included from /opt/sming/Sming/Libraries/Timezone/src/Timezone.cpp:12:0:
/opt/sming/Sming/Libraries/Timezone/src/Timezone.h:21:18: fatal error: Time.h: No such file or directory
 #include <Time.h>
                  ^
compilation terminated.
/opt/sming/Sming/component-wrapper.mk:155: recipe for target 'src/Timezone.o' failed
```
Unfortunately the git mentioned cannot be used anymore because the owner changed the code, but I got few months ago and they will be adapted and I have no idea about the version downloaded.
The only way is to rewrite the 
`<Time.h>`
to
`"../../Time.h"`
to get 
```
Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-Timezone.a
C+ /opt/sming/Sming/Libraries/Timezone/src/Timezone.cpp
AR /opt/sming/Sming/out/Esp8266/debug/lib/clib-Timezone.a

Building /home/angelo/Software/esp8266/project/Contatore_luce_nonos/out/Esp8266/debug/lib/clib-App-218dd5e4ade6882d55a1c549cad7553f.a
C+ /home/angelo/Software/esp8266/project/Contatore_luce_nonos/app/webserver.cpp
```
but is not correct. And also my code connot use 
`<Timezone.h>`
but
`#include "../../../../../../../../opt/sming/Sming/Libraries/Timezone/src/Timezone.h"`

mikee47, 2019-09-19:
What you need to do is create a separate [Component Repository](https://sming.readthedocs.io/en/latest/_inc/Sming/building.html#component-repositories) and place your libraries there. You can find a worked example of how to do this in the (SharedComponent)[https://github.com/SmingHub/Sming/tree/develop/tests/SharedComponent] test code.

mikee47, 2019-09-19:
I have Sming-ported libraries for time zones, daylight-saving calculation, etc. so I'll get them tidied up and submitted in the next day or two. This sort of thing should be available 'out of the box'.

closed 2019-09-27 by slaff