SmingHub/Sming #1841: build error: duplicate swap()

etmmahi, 2019-09-19:
When building an application with the foolowing libs in  components.mk:
```ARDUINO_LIBRARIES := ArduinoJson6 Adafruit_SSD1306```

then the following code fails:
```
#include <SmingCore.h>
#include <Libraries/Adafruit_SSD1306/Adafruit_SSD1306.h>
#include <ArduinoJson.h>
```
with:
```
In file included from /home/martijn/espdev/Sming/Sming/Libraries/Adafruit_SSD1306/Adafruit_SSD1306.h:49:0,
                 from /home/martijn/espdev/projects3/bug01/app/application.cpp:3:
/home/martijn/espdev/Sming/Sming/Libraries/Adafruit_GFX/Adafruit_GFX.h:11:20: error: expected unqualified-id before '{' token
 #define swap(a, b) { int16_t t = a; a = b; b = t; }
                    ^
In file included from /home/martijn/espdev/Sming/Sming/Libraries/ArduinoJson6/ArduinoJson/src/ArduinoJson/MsgPack/endianess.hpp:8:0,
                 from /home/martijn/espdev/Sming/Sming/Libraries/ArduinoJson6/include/../ArduinoJson/src/ArduinoJson/MsgPack/MsgPackDeserializer.hpp:11,
                 from /home/martijn/espdev/Sming/Sming/Libraries/ArduinoJson6/include/../ArduinoJson/src/ArduinoJson.hpp:36,
                 from /home/martijn/espdev/Sming/Sming/Libraries/ArduinoJson6/include/../ArduinoJson/src/ArduinoJson.h:9,
                 from /home/martijn/espdev/Sming/Sming/Libraries/ArduinoJson6/include/ArduinoJson.h:38,
                 from /home/martijn/espdev/projects3/bug01/app/application.cpp:4:
/home/martijn/espdev/Sming/Sming/Libraries/ArduinoJson6/ArduinoJson/src/ArduinoJson/Polyfills/utility.hpp:9:30: error: expected unqualified-id before '{' token
 inline void swap(T& a, T& b) {
                              ^
/home/martijn/espdev/Sming/Sming/component-wrapper.mk:155: recipe for target 'app/application.o' failed
```

It seems the swap() define in Adafruit_GFX.h is a (bad) duplicate.
Change the define into an inline function inside the Adafruit_GFX class fixes the problem.
```
 static inline void swap(int16_t &a, int16_t &b) {int16_t t = a; a = b; b = t;}
```

There may exist a better solution...

etmmahi, 2019-09-30:
The better solution (also applied in the original Adafruit_GFX library) is to move the macro into the cpp file and rename it to prevent name clashes. 
This solution is fount in the pull request

mikee47, 2019-09-30:
My preference would be to remove all the swap() macros and instead using std::swap. I'll work a PR up for that since it affects other bits.

etmmahi, 2019-09-30:
Nice, when ready I am happy to test

PR #1856 (2019-09-30): Remove `swap` macros and make std::swap available
Swap macros not necessary.
Fixes #1841, supersedes #1855.
closed 2019-09-30 by slaff