SmingHub/Sming #1844: RC releases for Sming 4.0

slaff, 2019-09-27:
We will start Release Candidate (RC) releases before we officially announce version 4.0. The plan for the moment is to have the following RCs
- [ ] RC1: latest code from develop. The release tag will be set in the `develop` branch. 
      Expected release date - 29th of September. 
- [ ] RC2: Leave support only for SDK >= 3.0.1. Due to the security issues found recently the older SDKs will not be supported officially. 
     Expected release date - 15th of October.
- [ ] RC3: Feature freeze and bug fixes only.

Expected release date for Sming 4.0.0 is beginning of November, 2019.

teicors, 2019-09-30:
Hi, just updated to latest version and in compiling my code, I receive this error:
/opt/sming/Sming/out/Esp8266/debug/lib/clib-ArduinoJson5.a(JsonObjectStream.o):(.rodata._ZTV16JsonObjectStream[vtable for JsonObjectStream]+0x38): undefined reference to `MemoryDataStream::seek(int)'

The json version is 5 into component.mk

slaff, 2019-09-30:
@mikee47 Can you take a look at that issue?

teicors, 2019-09-30:
after a make dtst-clean the configuration of the "project" is reset

/opt/sming/Sming/Arch/Esp8266/Components/esptool/esptool/esptool.py -p  -b 115200 write_flash -ff 40m -fm qio -fs 512KB 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x002000 out/Esp8266/debug/firmware/rom0.bin 0x100000 out/Esp8266/debug/firmware/spiff_rom.bin

How to catch the error:I

> make SMING_ARCH=Esp8266 SPI_MODE=dio SPI_SIZE=4M COM_SPEED_ESPTOOL=1000000

[compiling ok]

> make dist-clean

> make flash

the error appers and the only way to flash the esp is  a

> make SMING_ARCH=Esp8266 SPI_MODE=dio SPI_SIZE=4M COM_SPEED_ESPTOOL=1000000

mikee47, 2019-09-30:
@teicors Can you reproduce the problem by modifying one of the sample applications? That way I can reproduce the error.

mikee47, 2019-09-30:
> after a make dtst-clean the configuration of the "project" is reset

Yes, that's by design. You can perform a clean build without resetting config like this:

```
make components-clean clean
make
```

mikee47, 2019-09-30:
@teicors Something to check. Do `git status` and make sure there aren't any old files from the previous build - those should be deleted.

teicors, 2019-09-30:
@mikee47 
Sul branch develop
Your branch is up to date with 'origin/develop'.

I did 

> make components-clean clean
and the error disappears, even after a 

> make dist-clean
So I suppose something fixed after the components-clean; in my opinion the dist-clean should be stronger than components-clean or must include it.

Sorry for wasting your time.

teicors, 2019-09-30:
@mikee47 it seems no one error during both clean (I'm working only with Ubuntu):

> angelo@angelo-thinkpad:~/Software/esp8266/project/Contatore_luce_nonos$ make components-clean clean
> 
> Contatore_luce_nonos: Invoking 'components-clean clean' for Esp8266 (debug) architecture
> Cleaning all Component libraries
> Cleaning application...
> 

and
> angelo@angelo-thinkpad:~/Software/esp8266/project/Contatore_luce_nonos$ make dist-clean
> 
> Contatore_luce_nonos: Invoking 'dist-clean' for Esp8266 (debug) architecture
> Cleaning all submodules...
> Cleaning all samples...
> ../samples/HttpServer_FirmwareUpload/component.mk:11: attenzione: sovrascrittura del set di istruzioni per l'obiettivo "web-pack"
> ../samples/HttpServer_ConfigNetwork/component.mk:5: attenzione: ignorato il set di istruzioni obsoleto per l'obiettivo "web-pack"
> ../samples/HttpServer_FirmwareUpload/component.mk:14: attenzione: sovrascrittura del set di istruzioni per l'obiettivo "web-upload"
> ../samples/HttpServer_ConfigNetwork/component.mk:9: attenzione: ignorato il set di istruzioni obsoleto per l'obiettivo "web-upload"
>

mikee47, 2019-09-30:
The only difference I can see is that `dist-clean` *doesn't* clean your own project, only `make clean` does that. But I'm not sure that would explain the error you initially reported.

teicors, 2019-09-30:
@mikee47 the errors in the dist-clean phase era related to HttpServer_FirmwareUpload and HttpServer_ConfigNetwork in the sample dir: do you agree with me ?

mikee47, 2019-09-30:
@teicors Yes - you can ignore those particular warnings.

kwis2, 2019-10-11:
Windows install still not work. As in #1241 with windows 10 fresh install there is problem `Extracting 'C:\Users\xxxx\AppData\Local\Temp\chocolatey\mingw-get\mingw-get-0.6.2-mingw32-beta-20131004-1-bin.tar.xz' to 'C:\tools\MinGW'` hangs, and finally 

> ERROR: Exception calling "Start" with "0" argument(s): "Nie można odnaleźć określonego pliku"
> Environment Vars (like PATH) have changed. Close/reopen your shell to
>  see the changes (or in powershell/cmd.exe just type `refreshenv`).
> The install of sming was NOT successful.
> Error while running 'C:\ProgramData\chocolatey\lib\sming\tools\chocolateyInstall.ps1'.

Des it mean Windows is not supported with Sming?
```

mikee47, 2019-10-12:
@kwis2 I've tried the chocolatey install on a fresh Windows 10 Virtualbox installation, and whilst I don't get the same error as you there are definitely issues.

I've only ever used the manual approach outlined at https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/windows-manual.html. It says DEPRECATED but it really just needs updating (work in progress). A few points about this:

* I'd recommend using the regular 32-bit MinGW graphical installer from https://osdn.net/projects/mingw/downloads/68260/mingw-get-setup.exe. The one chocolatey attempts to install is MinGW-64 which is un-necessary as we're only producing 32-bit builds.
* If you don't have eclipse installed already, just use the latest version with C/C++ support. It doesn't have to be Luna.

mikee47, 2019-10-12:
I'd also recommend you install GIT and pull in the latest develop.

mikee47, 2019-10-12:
I've found the easiest way to install MinGW is via command line:

```
mingw-get install mingw32-base-bin mingw-developer-toolkit-bin mingw32-gcc-g++-bin mingw32-pthreads-w32-dev mingw32-libmingwex
```
This is sufficient to build for both Host and Esp8266.

closed 2019-11-04 by slaff