SmingHub/Sming #1865: SPIFFS size

teicors, 2019-10-02:
Hi,
during the compiling of my test code, I achieve the code "errno -10001" it means "no space left on devce", but looking at a very old image, I see a difference of the dimension of the spiffs space: my question is "where can I change the space for the spiffs_rom.bin" (if It's possible) ?

Angelo

> angelo@angelo-thinkpad:~/Software/esp8266/project/MeteoControl_nonos$ ls -lart out/firmware/
> totale 872
> -rw-rw-r-- 1 angelo angelo   2032 apr 12 23:20 rboot.bin
> -rw-rw-r-- 1 angelo angelo    600 apr 12 23:20 fwMeminfo.new
> -rw-rw-r-- 1 angelo angelo 348896 apr 12 23:20 rom0.bin
> drwxrwxr-x 2 angelo angelo   4096 apr 12 23:20 .
> -rw-rw-r-- 1 angelo angelo 524288 apr 12 23:20 spiff_rom.bin
> drwxrwxr-x 5 angelo angelo   4096 ott  2 14:45 ..
> angelo@angelo-thinkpad:~/Software/esp8266/project/MeteoControl_nonos$ ls -lart out/Esp8266/debug/firmware/
> totale 568
> drwxrwxr-x 5 angelo angelo   4096 ott  2 15:46 ..
> -rw-r--r-- 1 angelo angelo   1824 ott  2 15:46 rboot.bin
> -rw-r--r-- 1 angelo angelo    600 ott  2 15:46 fwMeminfo.new
> -rw-r--r-- 1 angelo angelo   1632 ott  2 15:46 rom0.bin.cfg
> -rw-r--r-- 1 angelo angelo 364080 ott  2 15:46 rom0.bin
> drwxr-xr-x 2 angelo angelo   4096 ott  2 15:46 .
> -rw-r--r-- 1 angelo angelo 196608 ott  2 15:46 spiff_rom.bin

slaff, 2019-10-02:
`SPIFF_SIZE` is what you are looking for. Example
```
cd <application>
make SPIFF_SIZE=1024 # Will create a spiff with size 1K
```

teicors, 2019-10-02:
To achieve the space needed I had to execute

> make SPIFF_SIZE=511024
[some output]
> make clean
[some output]
> make

after the first make with the space, the spiffs is not rebuild.

mikee47, 2019-10-02:
You can do `make spiffs-image-update`. Use `make help` to get a list of options.
Remember you can search documentation quite easily at https://sming.readthedocs.io/en/latest/ (box at top left).

closed 2019-10-02 by slaff