SmingHub/Sming #1887: How to install NON OS SDK 3.0.1 for Sming 4.0.0

juppin, 2019-10-15:
Hello @mikee47 and @slaff ,

thanks for your work on Sming.

I wanna test the new 4.0.0 rc, but i´m struggling at setting up the build environment...
How do i build or download the needed NON OS SDK 3.0.1?
The docs does only mention 1.5.4 and the old esp-open-sdk supports only 1.5.4.

Thanks in advance and hopefully Sming will last a long time.

mikee47, 2019-10-15:
Hi @juppin, thanks for you kind words. The SDK gets pulled in automatically as a submodule, and is the default for rc2. If you're getting specific errors, please post some more details and we'll try to help.

juppin, 2019-10-15:
Thank you very much... Didn´t thought it would be that easy. :-)
The last time i used Sming i had to setup the environment manually.

closed 2019-10-15 by juppin