SmingHub/Sming #1888: 4.0.0-rc2: Crash on WifiStation.getMAC in global constructors

verybadsoldier, 2019-10-16:
In my code I have a global object that queries the MAC address in its constructor with `WifiStation.getMAC`. This was working fine until 3.8 (and also 3.9 I think). Now in `4.0.0-rc2` it crashes instantly. This is a minimal example that reproduces the bug (tested on ESP-12F):
```
#include <SmingCore.h>

#define LED_PIN 2 // GPIO2

Timer procTimer;
bool state = true;

class Test {
	public:
	Test() {
		
    String s = WifiStation.getMAC();
    printf("\r\nMAC INIT: %s\r\n", s.c_str());
}
};
    
void blink()
{
	digitalWrite(LED_PIN, state);
	state = !state;
}

Test t;

void init()
{
	Serial.begin(115200); // 115200 by default
    Serial.systemDebugOutput(true); // Debug output to serial
    
    String s = WifiStation.getMAC();
    printf("nMAC: %s\r\n", s.c_str());
	pinMode(LED_PIN, OUTPUT);
	procTimer.initializeMs(1000, blink).start();
}
```

Running that makes the device reboot very frequently (around 10 Hz or something?!).

mikee47, 2019-10-16:
I think those 'garbage' characters might hold a clue. Can you switch the terminal baud rate (probably 74880 baud?) to see what they say?

verybadsoldier, 2019-10-16:
Ah right, sorry. Did that yesterday already (38400 is the key):
```
                     �E�E�Q�Q�Z],��ը����K���Q�Z�V�UP��PE��Te*ը����ota1 not set
ota2 not set
rf cal sector: 1019
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 3.1.0-dev(bd176e4) compiled @ Jul  3 2019 15:07:05
phy ver: 1145_0, pp ver: 10.2

Fatal exception 28(LoadProhibitedCause):
���*�P*V��ET����T���*��ZU+UZ��UT���TQT�QQ��jP*Y���*����ZT�(U���j*Q�TT�TQ��eQB���Q��u���ZT�(U���j*QV��ZV��V��T�V�P+��U����K��ZkVVU���U�(TT�U��
                                                                                                                                                  �U����EJ��Pե����u�(TT�U�UT
                    �E�EZ����P�Z�V
                                  ��P���QT�Q�Z�V�UP��PE��Te*ը����ota1 not set
ota2 not set
rf cal sector: 1019
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 3.1.0-dev(bd176e4) compiled @ Jul  3 2019 15:07:05
phy ver: 1145_0, pp ver: 10.2

Fatal exception 28(LoadProhibitedCause):
���*�P*V��ET����T���*�u�U+UZ��UT����TQT�QQ��jP*Y���*����ZT�(U���j*Q�TT�TQ��eQB���Q��u���ZT�(U���j*QV��ZV��V��T���P+��U����K��ZkVVU���U�(TT�U��
                                                                                                                                                  �U�����TT�U+UZ���U�TT�U�UT
```

mikee47, 2019-10-16:
It's the constructor for your `Test` class which is crashing things. Most likely, `WifiStation` hasn't been constructed yet. Comment out the `Test t` line and it's fine.

verybadsoldier, 2019-10-16:
Yes, I know. But exactly that is my error report (it was working before).

mikee47, 2019-10-16:
If the `getMAC()` method were declared `static` then your code would be fine. As it's not (and never has been) your `Test t;` static constructor a time bomb which has now detonated.

The reason it's now not working is due to #1802, which abstracted `StationClass` to provide platform independence. You need to fix your code, I would suggest adding an initialisation method (e.g. `begin()`) to your global class and calling it from `init()`.

verybadsoldier, 2019-10-16:
Ahh ok, right, thank you!

closed 2019-10-16 by verybadsoldier