SmingHub/Sming #1901: error: "WIFI_PWD" redefined [-Werror]

teicors, 2019-10-21:
Hi,
I updated sming to the latest version by git branch develop and I catch the error.
I did the following steps:

```
cd samples/HttpClient
make
make component-clean clean
make

```
and now I catch the error. 
the HttpClient is vanilla from git clone <Sming>. No one modification.
The same error on my projects.

mikee47, 2019-10-21:
See #1892. The fix is to ensure WIFI_SSID is defined. The error message could be clearer but in essence it's appropriate  I think.

The use of these variables is really only for basic samples and one-off applications. In deployable devices a more sophisticated mechanism for managing connection information is needed.

teicors, 2019-10-21:
Yes, I know. I put for my gadgets the ssid and password in an external file with the right value. The error is not related to a password empty, but to the duplicate (redefine) of password. In the example I expect an error like "SSID not defined" and not redefine of password.

slaff, 2019-10-21:
@mikee47 The following change can be done: In Sming/component.mk replace the setting of WIFI_PASS from:
```
ifdef WIFI_SSID
	APP_CFLAGS			+= -DWIFI_SSID=\"$(WIFI_SSID)\"
endif
ifneq ($(origin WIFI_PWD),undefined)
	APP_CFLAGS			+= -DWIFI_PWD=\"$(WIFI_PWD)\"
endif
```

to 

```

teicors, 2019-10-21:
Not exactly.. an undefined WIFI SSID is not catched.
Now I don't report any duplicate, neither the WIFI SSID voluntary missed as verify the solution.

mikee47, 2019-10-21:
@slaff That's a good idea, thanks.

@teicors 
> undefined WIFI SSID is not catched.

We can fix that for all the samples:

```
#ifndef WIFI_SSID
static_assert(false, "Please define WIFI_SSID and, if required, WIFI_PWD.");
#define WIFI_SSID "WIFI_SSID"
#endif
```

PR #1904 (2019-10-22): Fix 'WIFI_PWD' redefined error
Fixes #1901.
closed 2019-10-22 by slaff