SmingHub/Sming #1905: Issues with the tests on Host

slaff, 2019-10-22:
Quite often Travis CI is failing with the following problem (Full log can be seen here: https://travis-ci.org/SmingHub/Sming/jobs/601194512 )

```
<td><img src="back.gif" alt="[PARENTDIR]"></td>
<td colspan="5"><a href="/">Parent Directory</a></td>
</tr>
host_exit: returning 0
main: >> Normal Exit <<
host_flashmem_cleanup: Closed "out/Host/debug/firmware/flash.bin"
cleanup: Goodbye!
pure virtual method called
terminate called without an active exception
signal_handler: SIGABRT - usually caused by an abort() or assert()
/home/travis/build/SmingHub/Sming/Sming/Arch/Host/app.mk:46: recipe for target 'run' failed
```

I guess the issue is due to some race condition because if I restart the build multiple times at some point the error is not present. 

> pure virtual method called

@mikee47 Can that be the reason for the issue? Do you have idea why is that happening?

mikee47, 2019-10-23:
| pure virtual method called

The `pure virtual method called` is very likely a red herring and it's more likely access to a freed memory location. 

I've seen this one before, it was a threading issue where the CThread destructor was getting called twice. My first suspect is somewhere in `Arch/Host/Components/driver/hwtimer.cpp`. I'll see if I can pin it down.

slaff, 2019-10-29:
@mikee47 I guess you have noticed this but there is also an issue with the host compilation on Windows. Can you take a look at that issue too?

mikee47, 2019-10-29:
My feeling is that the appveyor issues are probably related to the duplicate dnsserver.h files, so should disappear.

I'd like to update the MinGW tools so it's using GCC 8.2. Installing from scratch is tedious, it's much faster just pulling in a zipped image which is 105MB - do you have a preferred location to put that?

slaff, 2019-10-30:
> I'd like to update the MinGW tools so it's using GCC 8.2. Installing from scratch is tedious, it's much faster just pulling in a zipped image which is 105MB - do you have a preferred location to put that?

The best will be to a separate repo with toolchains under SmingHub. Unfortunately I don't have the permissions to create a new repo under SmingHub.

@anakod @alonewolfx2 @hreintke Can one of you create a new repository named **toolchains** under SmingHub ?

mikee47, 2019-10-31:
> > pure virtual method called

Running Ubuntu in Virtualbox I initially couldn't reproduce this error, so reduced the thread count to 2 to match travis. Running the 'SharedComponent' test the fault appears within about 15 attempts, usually less.

I added some debug statements which pins down the location:

```
host_flashmem_cleanup: Closed "out/Host/debug/firmware/flash.bin"
cleanup: Goodbye!
CTimerThread::~CTimerThread: ENTER virtual CTimerThread::~CTimerThread() 0x5658ebc0
CTimerThread::~CTimerThread: LEAVE virtual CTimerThread::~CTimerThread() 0x5658ebc0
CThread::~CThread: ENTER virtual CThread::~CThread() 0x5658ebc0
CThread::join: ENTER void CThread::join() 0x5658ebc0
pure virtual method called
terminate called without an active exception

signal_handler: SIGABRT - usually caused by an abort() or assert()
```

The exception is raised when calling `pthread_join()`. I found this note:

https://tombarta.wordpress.com/2008/07/10/gcc-pure-virtual-method-called/

Which states, "You may be trying to call members on a destructing (and thus incomplete) object.".

I cannot see the logic in any of this, but the gist seems to be that it's unsafe to call `join()` in the virtual CThread destructor.

closed 2019-11-01 by slaff

anakod, 2019-11-26:
> > I'd like to update the MinGW tools so it's using GCC 8.2. Installing from scratch is tedious, it's much faster just pulling in a zipped image which is 105MB - do you have a preferred location to put that?
> 
> The best will be a separate repo with toolchains under SmingHub. Unfortunately I don't have the permissions to create a new repo under SmingHub.
> 
> @anakod @alonewolfx2 @hreintke Can one of you create a new repository named **toolchains** under SmingHub ?

Hello @slaff. Not sure why you haven't owner acces to the root section of the SmingHub. I'ts definitely mistake and I have fixed it :)

anakod, 2019-11-26:
By some reason I didn't see notification when you mentioned me here, Kul have contacted me today on Telegram messanger and said about this issue.

slaff, 2019-11-26:
>  I didn't see notification when you mentioned me here

Check your spam folder.  Make sure that you are also"watching" this project: https://github.com/SmingHub/Sming/watchers

> It's definitely a mistake and I have fixed it :)

Thanks a lot @anakod.

anakod, 2019-11-27:
Yes, looks like it's in Spam folder :(

PR #1930 (2019-11-01): Fix `pure virtual method called` error when running CI tests
See #1905 for discussion.

Unsafe to call methods from virtual CThread destructor, so add `terminate()` method to all derived classes and call explicitly _before_ destroying the object.

Fix keyboard thread race condition.

Closes #1905