SmingHub/Sming #1926: WifiStation stops connection attempts after startScan

verybadsoldier, 2019-10-31:
I am maintaining a Wifi RGB controller firmware which should try to reconnect to the Wifi upon loss of connection. But after a certain amount of tries it should also spin-up its internal AP to give the user a chance to reconfigure the Wifi to another SSID.

The logic in that firmware is:

1. Upon loss of connection -> keep trying to reconnect
2. After 10 tries -> Do a startScan and spin up the internal AP
3. The reconnect attempts should keep going now but the user has the chance to connect to the AP and change SSID

The problem is that the reconnect attempts stop in the moment the 10 tries are reached and the internal AP is started.
I tried to reproduce the problem in a modified `Basic_Wifi` sample (on `4.0.0-rc3`). It turned out that the problem is triggered by calling `WifiStation.startScan` in `onStationDisconnect`. As soon as that method is called the Wifi scan takes place and finishes but then the reconnect attempts stop.

Is this an expected behavior? I thought that it should be possible to perform a Wifi scan (and spin up the AP) without affecting the WifiStation reconnects. I guess there is a good chance that I have wrong expectations here but maybe someone can clear this up.

There are several scenarios in which the Wifi connection could be lost but this is my test code (basically added code to `connectFail`) which fails on authenticating in the first place but should keep trying:
```
#include <SmingCore.h>

// If you want, you can define WiFi settings globally in Eclipse Environment Variables
#ifndef WIFI_SSID
#define WIFI_SSID "" // Put you SSID and Password here
#define WIFI_PWD ""
#endif

// Will be called when WiFi station network scan was completed
void listNetworks(bool succeeded, BssList& list)
{
}

// Will be called when WiFi station was connected to AP
void connectOk(IpAddress ip, IpAddress mask, IpAddress gateway)
{
	Serial.print(_F("I'm CONNECTED to "));
	Serial.println(ip);
}

int tryCount = 0;
// Will be called when WiFi station was disconnected
void connectFail(const String& ssid, MacAddress bssid, WifiDisconnectReason reason)
{
    ++tryCount;

	// The different reason codes can be found in user_interface.h. in your SDK.
	Serial.print(_F("Disconnected from \""));
	Serial.print(ssid);
	Serial.print(_F("\", reason: "));
	Serial.println(WifiEvents.getDisconnectReasonDesc(reason));

	if (tryCount == 10) {
           // *** now I would like to enable AP and do a scan first
          // calling here startScan triggers the problem
	    WifiStation.startScan(&listNetworks);
	    //WifiAccessPoint.config("MyAP", nullptr, AUTH_OPEN);
	    //WifiAccessPoint.enable(true, false);
           // ***
	}
}

// Will be called when WiFi hardware and software initialization was finished
// And system initialization was completed
void ready()
{
	Serial.println(_F("READY!"));

	if(WifiAccessPoint.isEnabled()) {
		Serial.print(_F("AP. ip: "));
		Serial.print(WifiAccessPoint.getIP());
		Serial.print(_F(" mac: "));
		Serial.println(WifiAccessPoint.getMacAddress());
	}
}

void init()
{
	Serial.begin(SERIAL_BAUD_RATE);
	Serial.systemDebugOutput(true); // Allow debug print to serial
	Serial.println(_F("Sming. Let's do smart things!"));

	// Set system ready callback method
	System.onReady(ready);

	// Soft access point
	WifiAccessPoint.enable(true);
	WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);

	// Station - WiFi client
	WifiStation.enable(true);
	WifiStation.config(_F(WIFI_SSID), _F(WIFI_PWD));

	// Optional: Change IP addresses (and disable DHCP)
	WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));
	WifiStation.setIP(IpAddress(192, 168, 1, 171));

	// Print available access points
	WifiStation.startScan(listNetworks); // In Sming we can start network scan from init method without additional code

	// Set callback that should be triggered when we have assigned IP
	WifiEvents.onStationGotIP(connectOk);

	// Set callback that should be triggered if we are disconnected or connection attempt failed
	WifiEvents.onStationDisconnect(connectFail);
}
```

The output is this:
```
<several attemps here before>
Disconnected from "hsync", reason: Authentication failure
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
28947463 disconnect from ssid hsync, reason 202

Disconnected from "hsync", reason: Authentication failure
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
32789448 disconnect from ssid hsync, reason 202

Disconnected from "hsync", reason: Authentication failure
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
36634475 disconnect from ssid hsync, reason 202

Disconnected from "hsync", reason: Authentication failure
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
40477965 disconnect from ssid hsync, reason 202

Disconnected from "hsync", reason: Authentication failure
scandone
43315941 scan completed: 20 found
```
So, at the end the 10th try is reached and the Wifi scan is performed. Then the output stops cause no reconnects are tried anymore which is unexpected for me.

avr39-ripe, 2019-10-31:
Take a look on my way to handle this in https://github.com/SmingHub/Sming/tree/master/samples/Basic_WebSkeletonApp

It either connects to saved (if any) wifi network or fires own AP with web interface that allows you enter ssid/pwd (no scan!) and enable/disable STA mode. After connecting to wifi network in STA mode it shuts down AP.

mikee47, 2019-10-31:
Maybe you just need to call WifiStation.connect() again from your `listNetworks()` callback.

mikee47, 2019-10-31:
See also Appendix A.4 in the [NON-OS SDK reference](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf)

verybadsoldier, 2019-11-01:
Ok thanks, that are two good points. I am investigating...

verybadsoldier, 2019-11-03:
Ok, calling `WifiStation.connect` in the callback helped. I am experiencing some other strange behaviors but they might just be errors on my side.

So thanks again, I appreciate the support here alot.

I close this ticket as it probably is not considered a bug of Sming. Just something the user has to take care of.

closed 2019-11-03 by verybadsoldier