SmingHub/Sming #1932: compiler error strncpy -Wstringop-truncation]

frankdownunder, 2019-11-02:
Are we concerned about this compiler error? Probably low impact but it would be nice to get rid of it.

> In function ‘strncpy’,
    inlined from ‘spiffs_object_create’ at ../spiffs/src/spiffs_nucleus.c:947:3:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: warning: ‘__builtin_strncpy’ specified bound 32 equals destination size [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```
  oix_hdr.p_hdr.flags = 0xff & ~(SPIFFS_PH_FLAG_FINAL | SPIFFS_PH_FLAG_INDEX | SPIFFS_PH_FLAG_USED);
  oix_hdr.type = type;
  oix_hdr.size = SPIFFS_UNDEFINED_LEN; // keep ones so we can update later without wasting this page
  strncpy((char*)oix_hdr.name, (const char*)name, SPIFFS_OBJ_NAME_LEN);
```

mikee47, 2019-11-03:
Curious the compiler doesn't flag up the other one on line 1010, have fixed both.

slaff, 2019-11-04:
Should be fixed with PR #1935. Feel free to reopen the issue if the problem is not fixes in the latest `develop` version.

closed 2019-11-04 by slaff