SmingHub/Sming #1937: HttpClient

qdimka, 2019-11-04:
Hi, everyone.

I started using a Sming a couple of days ago. It is a great tool.
But I have some problems.
I need to send the sensor data to thinkspeak. 
I am using this [sample](https://github.com/SmingHub/Sming/tree/lts-develop/3.8.x/samples/HttpClient_ThingSpeak).

I modified next line:
```cpp
void sendData()
{
   /* ... */
   url.Query["api_key"] = "my_key";
   /* ... */
}
```

Nothing to happened after first call thinkSpeak.downloadString. 
`onDataSent` wasn`t called. 
After the next call, I got the message "Already freed".

What do I do wrong?

```
phy ver: 972, pp ver: 10.1

mode : sta(5c:cf:7f:84:53:da)
add if0
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with LETAIQD_2.4G, channel 6
dhcp client start...
ip:192.168.0.178,mask:255.255.255.0,gw:192.168.0.1
pm open,type:2 0
3fff01d0 already freed
3fff01d0 already freed
3fff01d0 already freed
3fff0028 already freed
3fff0018 already freed
3fff0018 already freed
3fff0018 already freed
3fff0018 already freed
3fff0018 already freed
3fff0018 already freed
3fff0018 already freed
3fff0018 already freed
```

mikee47, 2019-11-04:
Hi @qdimka, we'll look into this but as you've only just started using Sming I'd recommend moving to Sming 4.0.0 which has been released today.

mikee47, 2019-11-04:
I've located the reason for the 'already freed' message, I'll submit a PR to patch that (present in 4.0 as well). However, I've yet to discover why no response is received (success or failure).

mikee47, 2019-11-04:
Compiling the framework with DEBUG_VERBOSE_LEVEL=3 we get this:
```
26670300 Creating new HttpClientConnection
26670691 HttpClientConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
26670940 HttpClientConnection::connecting ...
26675155 +TCP connection
26677257 connect to: api.thingspeak.com
26694713 DNS record found: api.thingspeak.com = 34.228.105.156
26695835 TCP connect result: 0
33893166 TCP connection error: -8
```

The -8 error code is `ERR_ABRT` (connection aborted).

mikee47, 2019-11-04:
I'm assuming a valid connection key is required so this isn't entirely unexpected.

qdimka, 2019-11-04:
Thanks, @mikee47 

I've tested Sming 4.0.0 (from develop). 
"Already freed" was fixed, thanks.
But I have the same result with DEBUG_VERBOSE_LEVEL=3  

```
2128874 Removing stale connection: State: 4, Active: 0
52129431 ~TCP connection
52129536 Creating new HttpClientConnection
52133259 HttpClientConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
52140210 HttpClientConnection::connecting ...
52144377 +TCP connection
52146541 connect to: api.thingspeak.com
52159523 DNS record found: api.thingspeak.com = 34.228.105.156
52160537 TCP connect result: 0
59577871 TCP connection error: -8
```

I've tried other sites without https, but still get the same error.
Also, I've tried to add HTTP headers, but no result.

qdimka, 2019-11-04:
I've got the progress. I added a port to the url.
```cpp
Url url;
url.Host = "api.thingspeak.com";
url.Port = 80;
url.Path = "/update";
url.Query["key"] = "WHRMN4Y50D05IK3R";
url.Query["field1"] = String(sensorValue);

Serial.println(String(url));

auto* request = thingSpeak.createRequest(url);

    HttpHeaders headers;
    headers[HTTP_HEADER_USER_AGENT] = _F("HttpClient/Sming");

request
    ->setMethod(HTTP_GET)
    ->setHeaders(headers)
    ->onRequestComplete(onDataSent);

thingSpeak.send(request);
```

```
26129584 Creating new HttpClientConnection
26129964 HttpClientConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
26134542 HttpClientConnection::connecting ...
26138714 +TCP connection
26140872 connect to: api.thingspeak.com
26275819 DNS record found: api.thingspeak.com = 34.226.171.107
26276857 TCP connect result: 0
26468020 TCP connected
26468235 TCP connected
26468315 HttpClientConnection::onReadyToSendData: waitingQueue.count: 1
26468920 MemoryDataStream::realloc 0 -> 52
26471413 Storing 52 bytes in stream
26474934 MemoryDataStream::realloc 52 -> 210
26478614 Storing 30 bytes in stream
26481860 Storing 29 bytes in stream
26485068 Storing 19 bytes in stream
26488244 Storing 2 bytes in stream
26491377 TCP onReadyToSendData: 0
26494611 TCP Written: 132, Available: 132, isFinished: 1, PushCount: 1 [TcpBuf: 2648]
26501942 TcpClient stream finished
26505091 HttpClientConnection::onReadyToSendData: waitingQueue.count: 0
26511397 Nothing in the waiting queue
26775685 TCP sent: 132
26775917 HttpClientConnection::onReadyToSendData: waitingQueue.count: 0
26776055 Nothing in the waiting queue
26776574 TCP onReadyToSendData: 2
26794858 The headers are complete
26798939 staticOnMessageComplete: Execution queue: 0, http://api.thingspeak.com:80/update?key=WHRMN4Y50D05IK3R&field1=1
Success sent
Server response: '5'
26802297 TCP connected
26804101 HttpClientConnection::onReadyToSendData: waitingQueue.count: 0
26810427 Nothing in the waiting queue
26813816 TCP onReadyToSendData: 0
26817297 TCP received: 644 bytes
26819798 HttpClientConnection::onReadyToSendData: waitingQueue.count: 0
26826129 Nothing in the waiting queue
26829511 TCP onReadyToSendData: 1
27677152 HttpClientConnection::onReadyToSendData: waitingQueue.count: 0
27677430 Nothing in the waiting queue
```
That's enough for me. I think we can close this issue.

mikee47, 2019-11-04:
> `onDataSent` wasn`t called.

Yes, this is another bug - thank you for raising this issue. You can track the progress of this in #1939.

The reason the sample doesn't work is that the scheme is missing from the URL. Easily fixed by adding:

`url.Scheme = URI_SCHEME_HTTP;`

closed 2020-01-02 by slaff