SmingHub/Sming #1943: How to process HTTP requests asynchronously?

mikee47, 2019-11-04:
My particular scenario is simple:

- Request arrives to turn a light on
  - Send command to remote device for execution
  - Mark request 'pending'
- Response arrives from remote device
  - Response forwarded to requestor and request is completed

This is easy to handle using websockets, but I'm now obliged to use 3rd party API which uses REST commands.

When handling `HttpResouce::onRequestComplete` we have to fulfull the request. We can return an error code, but I cannot see any way to defer it.

If anyone already has a solution for this, please say!

slaff, 2019-11-04:
If I understood you correctly in your scenario you are running Http Server on the device.
If that is the case then you can do the following. Let's say that the request goes to path `/light/`. You can add a simpler callback for this path that look like this:
```C++
void onLight(HttpRequest& request, HttpResponse& response)
{
     Special stream = new SpecialStream();
    // TODO: send the command to the remote device.
    // TODO: store the special stream so that it is accessible to the function that will be called when the call to the remote device has finished

     response.code = HTTP_STATUS_OK;
     response.sendDataStream(stream, MIME_JSON);
}
```

The important part here is the SpecialStream. 
- available() should return -1 so that content is sent using chunked encoding
- has a member variable `isReady` = false by default. 
- isFinished returning true when isReady is true. False otherwise
- readMemoryBlock() returning 0 when not ready and finally returning JSON ok, or similar when the isReady is true. A pseudo code of that stream can look like this:
```C++
class SpecialStream: public IDataSourceStream {
public:
	void setReady(bool isReady) {
		this->isReady = isReady;
	}

	uint16_t readMemoryBlock(char* data, int bufSize) override
	{
		if (!isReady) {
			return 0;
		}

		// TODO: copy to data the OK response.
       // Notice: you can return also 0 and don't send any response body but that depends on the
      // the expectations of the http client ...
	return <size-of-ok-response>;
}

bool isFinished() override
{
	return isReady;
}

bool seek(int len) override
{
        // if the ok response is small just return true should be fine.
        // in the cases where the response might not fit in a TCP send queue buffer then
       // the logic should be changed to handle the seek better in the cases where isReady is true.
	return true;
}

private:
bool isReady = false;
}

```

mikee47, 2019-11-04:
Thank you! That was exactly what I was looking for.

aemseemann, 2019-11-04:
Thanks very much! Awesome timing, I am about to face a very similar problem. 
However, I was wondering how to trigger sending the response once it becomes available. I guess calling `onReadyToSendData` or `TcpClient::pushAsyncPart` would do the trick, but both of them are protected...

mikee47, 2019-11-04:
I'm guessing `onPoll()` would pick it up... not sure what the loop time is, we'll see.

mikee47, 2019-11-04:
#1761 looks related

aemseemann, 2019-11-05:
Ah, thanks for the hint! I didn't known about the poll feature. I would expect a 2-second interval (from the docs of `tcp_poll` stating that the poll timer ticks twice a second and  `TcpConnection::initialize` where `tcp_poll(tcp, staticOnPoll, 4)` is called with a 4-tick interval).

Maybe one could extend HttpServerConnection::send like this to avoid the delay:
```cpp
void HttpServerConnection::send()
{
	if (state == eHCS_Ready) {
		state = eHCS_StartSending;
		onReadyToSendData(eTCE_Received);
	} else {
		onReadyToSendData(eTCE_Poll);
	}
}
```