SmingHub/Sming #1955: 3.0.1 open sdk - where?

stokersoft, 2019-11-07:
The stable Sming v4.0.0 supports 3.0.1 open sdk. But the official pfalcon repository, does not contain 3.0.1. Where is the recommended repository for open sdk 3.0.1 to use for v4.0.0. Thanks.

mikee47, 2019-11-07:
Hi @stokersoft, see #1887

jensahylda, 2019-11-08:
Thaks a lot. Its probably very obvious if you know how, last time i setup environment myself as we should and that worked great.  But to me, if i dont have to download openSDK myself, then i do a "cd $SMING_HOME" and "make dist-clean" - but then it complains about ESP_HOME not set correctly... There must be somethig that is still needed, except just making. Thanks and sorry for not seeing it.

mikee47, 2019-11-08:
Sming 4.0 will require a little re-orientation as there are a lot of changes, not least of which is the documentation at https://sming.readthedocs.io/en/latest/. I heartily recommend you find some quiet time and settle down with a cup of coffee and have a read through. It's tablet-friendly as well!

mikee47, 2019-11-08:
And please feel free to make any suggestions for improvements. You can also find us at https://gitter.im/SmingHub/Sming

jensahylda, 2019-11-08:
Thanks for very fast answer, appreciate that. I already read your documentation, it leads me to this:
https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/linux.html

And then i do that, i replace the version number with 3.0.1, it fails because that is not available.

Then i ask here, and i you send me to this answer:
"The SDK gets pulled in automatically as a submodule, and is the default for rc2."

And when i ask you about, that you send me back to the docs that sends me to installing open sdk.. and so on, we are cycling :-) 

You got some information in your head, that i am not able to read from this. 

Its for sure my problem, no doubt about that, and sorry for that. But i hope you can guide me to just that one step that that  starts the process of "The SDK gets pulled in automatically as a submodule, and is the default for rc2."

Thanks - and sorry.

mikee47, 2019-11-08:
@jensahylda You can follow the instructions on that page as-is, there is no need to change the version number since we only need the toolchain.

I agree the instructions are confusing! We'll update that, so thank your for pointing this out.

jensahylda, 2019-11-08:
@mikee47 - thanks a lot. I fully understand that documentation might need corrections - so glad you came and made this project alive again.

Evgsurf, 2019-11-15:
Looks it is necessary to have some manual "how to install Sming 4"