SmingHub/Sming #1968: OTA Fail because of software watchdog reset

cometurrata, 2019-11-15:
I had the issue with Sming-3.8 + SDK 2.0.0
I also have the issue with Sming-4.0.0 + SDK 3

I can easily reproduce that but doing many OTA.
Seems that it depends of the wifi or something. 
I can reproduce it on many Boards and on many wifi network.

Does someone have an idea 

Thanks for the great job by the way.

```
0x40102788: spi_flash_write at ??:?
0x40280418: flashmem_write_internal at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/spi_flash/flashmem.c:243
0x40105d22: rcReachRetryLimit at ??:?
0x40105e87: rcReachRetryLimit at ??:?
0x40106376: wDev_ProcessFiq at ??:?
0x4028051c: flashmem_write at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/spi_flash/flashmem.c:243
0x4000050c: ?? ??:0
0x402370b7: esf_buf_alloc at ??:?
0x4000444e: ?? ??:0
0x40000000: ?? ??:0
0x4010246b: flash_gd25q32c_read_status at ??:?
0x4000410f: ?? ??:0
0x40004a3c: ?? ??:0
0x40102700: spi_flash_erase_sector at ??:?
0x40228da9: rboot_write_flash at /home/come/luko/hardware/Sming-4.0.0/Sming/Components/rboot/rboot/appcode/rboot-api.c:108
0x4010515e: pp_post at ??:?
0x40235637: pp_attach at ??:?
0x402672ee: RbootOutputStream::write(unsigned char const*, unsigned int) at /home/come/luko/hardware/Sming-4.0.0/Sming/Wiring/Print.h:39
0x40235686: pp_attach at ??:?
0x40235792: pp_attach at ??:?
0x4010515e: pp_post at ??:?
0x40234743: ppTxPkt at ??:?
0x4020afab: ieee80211_output_pbuf at ??:?
0x4010164d: ets_timer_disarm at ??:?
0x40262fff: HttpClientConnection::onBody(char const*, unsigned int) at /home/come/luko/hardware/Sming-4.0.0/Sming/Wiring/Countable.h:23
0x40281ced: HttpConnection::staticOnBody(http_parser*, char const*, unsigned int) at /home/come/luko/hardware/Sming-4.0.0/Sming/Core/Network/Http/HttpConnection.cpp:157
0x40223e9d: etharp_send_ip at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/netif/etharp.c:630
0x40223eb4: etharp_send_ip at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/netif/etharp.c:630
0x40273058: http_parser_execute at /home/come/luko/hardware/Sming-4.0.0/Sming/Components/http-parser/http_parser.c:1999 (discriminator 1)
0x40104003: lmacRecycleMPDU at ??:?
0x40224112: etharp_output_to_arp_index at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/netif/etharp.c:890
0x40103abf: lmacProcessTxSuccess at ??:?
0x401038b8: lmacProcessTXStartData at ??:?
0x40224441: etharp_output at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/netif/etharp.c:995
0x40223c65: ip_output_if_opt at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/ipv4/ip.c:780
0x4023acf0: chip_v6_unset_chanfreq at ??:?
0x40101a9e: pvPortCalloc at ??:?
0x4023c728: __stdio_exit_needed at ??:?
0x40223c9a: ip_output_if at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/ipv4/ip.c:631
0x4021f5e6: pbuf_alloc at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/pbuf.c:909 (discriminator 3)
0x40260498: HttpConnection::onTcpReceive(TcpClient&, char*, int) at /home/come/luko/hardware/Sming-4.0.0/Sming/Core/Network/Http/HttpRequest.h:74
0x40106376: wDev_ProcessFiq at ??:?
0x40281d53: std::_Function_handler<bool (TcpClient&, char*, int), Delegate<bool (TcpClient&, char*, int)>::Delegate<HttpConnection>(bool (HttpConnection::*)(TcpClient&, char*, int), HttpConnection*)::{lambda(TcpClient&, char*, int)#1}>::_M_invoke(std::_Any_data const&, TcpClient&, char*, int) at /opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.5/functional:2058
0x4021f48a: pbuf_free at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/pbuf.c:909 (discriminator 3)
0x4025f1f3: std::function<bool (TcpClient&, char*, int)>::operator()(TcpClient&, char*, int) const at /opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.5/functional:2471
 (inlined by) TcpClient::onReceive(pbuf*) at /home/come/luko/hardware/Sming-4.0.0/Sming/Core/Network/TcpClient.cpp:110
0x4025eb36: TcpConnection::internalOnReceive(pbuf*, signed char) at /home/come/luko/hardware/Sming-4.0.0/Sming/Core/Network/TcpConnection.cpp:574
0x40235637: pp_attach at ??:?
0x4025ebd8: TcpConnection::initialize(tcp_pcb*)::{lambda(void*, tcp_pcb*, pbuf*, signed char)#2}::_FUN(void*, tcp_pcb*, pbuf*, signed char) at /home/come/luko/hardware/Sming-4.0.0/Sming/Core/Network/TcpConnection.cpp:342
 (inlined by) _FUN at /home/come/luko/hardware/Sming-4.0.0/Sming/Core/Network/TcpConnection.cpp:344
0x40227d78: tcp_input at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/tcp_in.c:394 (discriminator 1)
0x4010515e: pp_post at ??:?
0x40234743: ppTxPkt at ??:?
0x402238dd: ip_input at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/ipv4/ip.c:559
0x40224521: ethernet_input at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/netif/etharp.c:1379
0x40104931: lmacTxFrame at ??:?
0x40104ceb: ppEnqueueRxq at ??:?
0x402184d4: ets_timer_handler_isr at ??:?
[... 61 lines cut ...]
3ffffe50:  3fff8690 66664f5f 00656369 0000104a  
3ffffe60:  3fff8894 e3190000 3fff8870 3fff8664  
3ffffe70:  3fff6328 3fff6328 3fff85e0 40281d53  
3ffffe80:  40101793 086bc420 3fff6688 4025f1f3  
3ffffe90:  00000000 600011f0 3fff8870 3fff8870  
3ffffea0:  00000000 3fff6328 3fff85e0 4025eb36  
3ffffeb0:  00000001 ffffffc6 40235637 00000001  
3ffffec0:  3fff2264 3fff8870 3fff6328 4025ebd8  
3ffffed0:  3fff2264 3fff2260 3fff8894 40227d78  
3ffffee0:  3fff88ac 00000000 00000010 00000010  
3ffffef0:  00000000 0000056e 00000020 4010515e  
3fffff00:  3ffe0000 40234743 3fff6028 3fff49f8  
3fffff10:  3ffee6a2 3fff4a00 3fff6328 402238dd  
3fffff20:  3ffec140 3fff6a48 3fff6a48 00000000  
3fffff30:  00000000 06a20579 40234137 3fff6ac8  
3fffff40:  3fff6328 3fff6a48 3ffee694 40224521  
3fffff50:  00000068 00000001 40104931 3ffec640  
3fffff60:  3fff6028 40104ceb 00000001 00000000  
3fffff70:  402184d4 40104ab8 00000000 00000000  
3fffff80:  40104af4 00000000 00000002 3fff6ac8  
3fffff90:  3fffdc80 00000000 3fff6328 4021cccf  
3fffffa0:  40000f49 3fffdab0 ffffff00 40000f49  

================================================================
```

mikee47, 2019-11-15:
Can you reproduce the fault running in the debugger? The backtrace is more reliable.

cometurrata, 2019-11-15:
I never used the debugger, I will give it a try

cometurrata, 2019-11-15:
I got:
```
Program received signal SIGPWR, Power fail/restart.
0x40257606 in debug_crash_callback (rst_info=rst_info@entry=0x3ffff990, stack=stack@entry=1073740608, stack_end=stack_end@entry=1073741744)
    at /home/come/luko/hardware/Sming-4.0.0/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp:98
98              gdbstub_break_internal(DBGFLAG_RESTART);
```

cometurrata, 2019-11-15:
Not a Sming Issue
I believe it is not a software issue, my LDO is too weak so voltage drop and get the flash to fail.

closed 2019-11-15 by cometurrata