SmingHub/Sming #1970: Eclipse 'debug as local application' fails

mikee47, 2019-11-18:
If I right-click on a project, configured to build in the Host Emulator, then select `Debug as` -> `Local C/C++ Application` it fails to start. Investigation shows that it picks up the `app_0.out` file from the ESP8266 build instead.

I then have create a debug configuration manually.

This isn't a great deal of hassle but it's not as smooth as I'd like. Any suggestions?

slaff, 2019-11-19:
In Eclipse you will have to learn to "refresh" the project from time to time. If you have already compiled a sample with SMING_ARCH=Host then you refresh (Right click on the project -> Refresh (F5) ). After that select `Debug as -> Local C/C++ Application`. If you have more than one binary in the project you will be presented with a list of binaries to choose from. Similar to the screenshot below:

[image]

After that you can additionally configure the debugger. At least for me I had to add a .gdbinit file with the following line. 

```
handle SIGUSR1 nostop noprint
```

This prevents the debugger from constantly pausing here:

```C++
static void signal_handler(int sig)
{
	if(sig == SIGUSR1) {
		mainThreadSemaphore.wait();
	}
}
```

mikee47, 2019-11-19:
Thanks slaff, I'll give it a go.

closed 2019-12-03 by mikee47