SmingHub/Sming #1971: Json length problem

kwis2, 2019-11-19:
I test Basic_WebSkeletonApp example and there is problem with JsonObjectStream. With about 400 bytes of data there is stream cut - it looks like key/value were NULL/NULL on the end of stream and js script hangs.

mikee47, 2019-11-19:
Can you provide a little more detail so we can reproduce the issue?

kwis2, 2019-11-19:
I have modified `onAjaxGetState`  with

```
 for(uint16_t i=0;i<counter;i++)
 	{
 		json["counter"+String(i)] = i;
 	}
```
and it worked fine until 37 and next loop it was `null null` terminated and js hangs and with opera preview there was state:

```
 counter0: 0
 counter1: 1
 ......
 counter34: 34
 counter35: 35
 counter36: 36
 counter37: 37
 counter38: 38
 null: null

```

mikee47, 2019-11-20:
JsonObjectStream uses an internal DynamicJsonDocument with a default capacity of 1024 bytes. The problem you describe occurs when the document is full and cannot allocate space for the final key/value pair.

You can increase the document size like this:

```
JsonObjectStream* stream = new JsonObjectStream(2048);
```

But if your requirements are too large then you'll need a more involved approach to seralising the data. There's an example of how to do this in the RingTone library, in `RtttlJsonListStream.cpp`.

kwis2, 2019-11-20:
I think I have to clear where the problem is.
Finite capacity of JsonObjectStream is not a problem. Problem is that unexpectedly there is `null: null` which kill js and how could we avoid such situations.
On the other hand it could be good idea to know if data we add to stream reach the www....
I mean with modified Basic_WebSkeletonApp example we expect for instance counter40: 40 and there is only counter38: 38 availiable.
JsonObjectStream(2048) changed that only counter78: 78 is the last

mikee47, 2019-11-20:
@kwis2 You may find this helpful https://arduinojson.org/v6/api/jsonobject/subscript/

kmihaylov, 2019-11-26:
I had similar problem (in my case the values were "0" (integer).

Long story short, I had to increase the size of the object defined by ```DynamicJsonDocument``` despite my string's length was less than the jsonDocument object.

Just increased it twice or so and the "0" values became what was expected.

Now I see that I even use ```DeserializationError``` without checking the result. Probably there is similar type for serialization.

kmihaylov, 2019-12-07:
@kwis2, may you provide example of the problem? I've prepared some simple skeleton to use.

https://github.com/kmihaylov/Sming_ArduinoJson6_DebugSkeleton

kwis2, 2020-02-18:
@kmihaylov  Sorry I missed your post. 
in your github I commit example. (I am not sure if it was succesful). Json failed after about 430 bytes ...